espressif/esp-tflite-micro #2: Person detection example run error

notplus, 2021-12-07:
Hi, I tried to run person detection on my ESP32 CAM, everything seems OK but monitor showed something like below. I just change `CONFIG_CAMERA_MODEL_AI_THINKER=y` in config file. The same example from [here](https://github.com/espressif/tensorflow) performs well.

```
I (0) cpu_start: App cpu up.
I (381) cpu_start: Pro cpu start user code
I (381) cpu_start: cpu freq: 240000000
I (381) cpu_start: Application information:
I (386) cpu_start: Project name:     person_detection
I (392) cpu_start: App version:      8ee55e2
I (397) cpu_start: Compile time:     Dec  7 2021 14:51:39
I (403) cpu_start: ELF file SHA256:  dc5e9cde05b91be7...
I (409) cpu_start: ESP-IDF:          v4.3.1-dirty
I (414) heap_init: Initializing. RAM available for dynamic allocation:
I (421) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (428) heap_init: At 3FFB5450 len 0002ABB0 (170 KiB): DRAM
I (434) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (440) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (447) heap_init: At 4008CDD0 len 00013230 (76 KiB): IRAM
I (453) spi_flash: detected chip: generic
I (457) spi_flash: flash io: qio
I (463) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (20) sccb: pin_sda 26 pin_scl 27

E (20) i2c: i2c_param_config(647): i2c clock choice is invalid, please check flag and frequency
I (20) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (1060) i2c: i2c_set_pin(828): scl and sda gpio numbers are the same
E (2060) i2c: i2c_set_pin(828): scl and sda gpio numbers are the same
```

BasileBerckmoes, 2021-12-21:
Having the same error with the ESP-EYE

vikramdattu, 2021-12-22:
@notplus @BasileBerckmoes 

Can you share the complete LOGs?

What IDF branch do you use?
Please use `release/v4.2`
Also, use the `esp32-camera` which is submodule of this project.
If you haven't cloned recursively, just do: `git submodule update --init`

With IDF branch: `release/v4.2`, I get following log on ESP-EYE:
```
I (0) cpu_start: App cpu up.
I (414) heap_init: Initializing. RAM available for dynamic allocation:
I (421) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (427) heap_init: At 3FFB4C20 len 0002B3E0 (172 KiB): DRAM
I (434) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (440) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (446) heap_init: At 4008C084 len 00013F7C (79 KiB): IRAM
I (452) cpu_start: Pro cpu start user code
I (471) spi_flash: detected chip: generic
I (471) spi_flash: flash io: qio
I (471) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (20) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (20) gpio: GPIO[14]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (30) sccb: pin_sda 18 pin_scl 23

I (120) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (120) gpio: GPIO[37]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (130) gpio: GPIO[38]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (140) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (150) gpio: GPIO[35]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (160) gpio: GPIO[14]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (170) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (180) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (190) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (200) gpio: GPIO[27]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (210) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (210) camera: Allocating 1 frame buffers (9 KB total)
I (220) camera: Allocating 9 KB frame buffer in OnBoard RAM
Camera Initialized

Image Captured

person score:101 no person score -101
Image Captured

person score:-20 no person score 20
Image Captured

```

BasileBerckmoes, 2021-12-22:
Im using the esp-idf v4.3.2, i will check if the issue is solved by downgrading to v4.2.

I (0) cpu_start: App cpu up.
I (896) spiram: SPI SRAM memory test OK
I (904) cpu_start: Pro cpu start user code
I (904) cpu_start: cpu freq: 240000000
I (904) cpu_start: Application information:
I (907) cpu_start: Project name:     person_detection
I (912) cpu_start: App version:      1256525-dirty
I (918) cpu_start: Compile time:     Dec 22 2021 11:45:18
I (924) cpu_start: ELF file SHA256:  0fb30aac97a5f0b2...
I (930) cpu_start: ESP-IDF:          v4.3.2-dirty
I (935) heap_init: Initializing. RAM available for dynamic allocation:
I (942) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (948) heap_init: At 3FFB5A58 len 0002A5A8 (169 KiB): DRAM
I (955) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (961) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (967) heap_init: At 40091F7C len 0000E084 (56 KiB): IRAM
I (974) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (982) spi_flash: detected chip: generic
I (986) spi_flash: flash io: qio
I (992) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (20) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (20) gpio: GPIO[14]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (30) sccb: pin_sda 18 pin_scl 23

E (30) i2c: i2c_param_config(647): i2c clock choice is invalid, please check flag and frequency
E (1050) i2c: i2c_set_pin(828): scl and sda gpio numbers are the same
E (2050) i2c: i2c_set_pin(828): scl and sda gpio numbers are the same
E (3050) i2c: i2c_set_pin(828): scl and sda gpio numbers are the same

BasileBerckmoes, 2021-12-22:
v4.2 did the trick! Thank you

vikramdattu, 2021-12-22:
@notplus is the issue similar for you?

notplus, 2021-12-26:
Solved with v4.2. Thank you!

closed 2021-12-26 by notplus

osugiw, 2023-03-08:
> @notplus @BasileBerckmoes
> 
> Can you share the complete LOGs?
> 
> What IDF branch do you use? Please use `release/v4.2` Also, use the `esp32-camera` which is submodule of this project. If you haven't cloned recursively, just do: `git submodule update --init`
> 
> With IDF branch: `release/v4.2`, I get following log on ESP-EYE:
> 
> ```
> I (0) cpu_start: App cpu up.
> I (414) heap_init: Initializing. RAM available for dynamic allocation:
> I (421) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
> I (427) heap_init: At 3FFB4C20 len 0002B3E0 (172 KiB): DRAM
> I (434) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
> I (440) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
> I (446) heap_init: At 4008C084 len 00013F7C (79 KiB): IRAM
> I (452) cpu_start: Pro cpu start user code
> I (471) spi_flash: detected chip: generic
> I (471) spi_flash: flash io: qio
> I (471) cpu_start: Starting scheduler on PRO CPU.
> I (0) cpu_start: Starting scheduler on APP CPU.
> I (20) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
> I (20) gpio: GPIO[14]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
> I (30) sccb: pin_sda 18 pin_scl 23
> 
> I (120) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
> I (120) gpio: GPIO[37]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
> I (130) gpio: GPIO[38]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
> I (140) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
> I (150) gpio: GPIO[35]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
> I (160) gpio: GPIO[14]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
> I (170) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
> I (180) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
> I (190) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
> I (200) gpio: GPIO[27]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
> I (210) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
> I (210) camera: Allocating 1 frame buffers (9 KB total)
> I (220) camera: Allocating 9 KB frame buffer in OnBoard RAM
> Camera Initialized
> 
> Image Captured
> 
> person score:101 no person score -101
> Image Captured
> 
> person score:-20 no person score 20
> Image Captured
> ```
> 
> No menuconfig changes for `ESP-EYE`.

How do you inference using camera? My program already initialize, but I do not know how to start the inference.
[image]