espressif/esp-tflite-micro #3: About Tensor Arena Size

notplus, 2021-12-07:
Hi, hello world example performs well on my ESP32 CAM, but when I change `kTensorArenaSize` to about 200k, it occurs error like ``/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: region `dram0_0_seg' overflowed by 41976 bytes``. It seems that the memory is not enough, but my chip has 500KB internal RAM and 4MB PSRAM. For my application, about 250k arena size is needed. So is that possible to let tensor arena size bigger?

vikramdattu, 2021-12-08:
Hi @notplus it is possible to do this.

Instead of going for dram (static buffer), you can allocate this buffer on heap (internal or external depending your requirement) to get more flexibility.

For reference: https://github.com/espressif/tflite-micro-esp-examples/blob/master/examples/person_detection/main/main_functions.cc#L68 

Thanks,
Vikram

notplus, 2021-12-08:
Hi, thank you for your reply. I tried the code you motioned, but it remain error while running.
Code:
```
constexpr int kTensorArenaSize = 250*1024;
uint8_t *tensor_arena;

// some code 

  if (tensor_arena == NULL) {
    printf("%s: free RAM size: %d, INTERNAL: %d, PSRAM: %d\n", "tag", heap_caps_get_free_size(MALLOC_CAP_8BIT), heap_caps_get_free_size(MALLOC_CAP_8BIT | MALLOC_CAP_INTERNAL), heap_caps_get_free_size(MALLOC_CAP_8BIT | MALLOC_CAP_SPIRAM));
    tensor_arena = (uint8_t *) heap_caps_malloc(kTensorArenaSize, MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT);
    // tensor_arena = (uint8_t *) heap_caps_malloc(kTensorArenaSize,MALLOC_CAP_8BIT | MALLOC_CAP_SPIRAM);
  }
  if (tensor_arena == NULL) {
    printf("Couldn't allocate memory of %d bytes\n", kTensorArenaSize);
    return;
  }

```

Output:
```
tag: free RAM size: 287300, INTERNAL: 287300, PSRAM: 0
Couldn't allocate memory of 256000 bytes
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
```

vikramdattu, 2021-12-08:
@notplus 

You have not enabled SPIRAM.

You can enable it via menuconfig as follows:
idf.py menuconfig > components > ESP32 specific > SPIRAM

and then rebuild the application with above code 

```
tensor_arena = (uint8_t *) heap_caps_malloc(kTensorArenaSize, MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT);
```

notplus, 2021-12-08:
Ok, but does this mean that there is no way to allocate 250k internal memory? Other platfom which only has 384 kb SRAM can allocate 250k arena size when run TFLM.

vikramdattu, 2021-12-08:
For static memory, ESP32 restrict it below 160KB.
But since, since you have `287300` free at the time,  `ideally` you should be able to allocate 250KB though via malloc.

Could you please also add prints in your code for largest free block? allocations obviously capped by this number.
`heap_caps_get_largest_free_block(MALLOC_CAP_8BIT | MALLOC_CAP_INTERNAL);`

For more details on how 520KB SRAM is organised, you may take a look at this nice blog: https://blog.espressif.com/esp32-programmers-memory-model-259444d89387

Please note that 192KB is reserved for IRAM and remaining (328KB) is used as a DRAM. (ROM functions+Data+BSS+Heap)

To analyse what's eating up memory, you may run `idf.py size` for basic and `idf.py size-components` for yet detailed size info of the app.

notplus, 2021-12-08:
Ok, I got it. Thank you for your nice reply. Maybe it's better to use esp-dl.  :laughing:

commit 064b8feb (2023-08-14): Merge branch 'refactor/use_board_support_packages' into 'master'

Install dependencies from component registry

Closes #3

See merge request app-frameworks/tflite-micro-esp-examples!91
closed 2021-12-08 by vikramdattu