espressif/esp-tflite-micro #4: person_detection 例程结果好像不正确

HoGC, 2022-01-07:
我使用ESP32S3-EYE可以运行例程，相机初始化也没有问题，但是log结果始终是随机的一正一负
```
person score:-34 no person score 34
Image Captured

person score:-10 no person score 10
Image Captured

person score:-15 no person score 15
Image Captured

person score:33 no person score -33
Image Captured

person score:-10 no person score 10
Image Captured
```

vikramdattu, 2022-01-07:
Hi @HoGC,
We have used the same model and similar setup to the one tflite uses. (https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/examples/person_detection)
The model behaviour is this way. Even on x86, the outputs are always on +ve and other exact -ve to that value.

In general,
if person_score > no_preson_score by good margin it is a person. 

Thanks.

vikramdattu, 2022-04-07:
This is fixed with following commit:
https://github.com/espressif/tflite-micro-esp-examples/commit/b905680d525161e46ba2b036a9d2b5a77e165154

Scores are now requantized and scaled to % of confidence.
e.g., previous `person score:115 no person score -115` is now changed to `person score:95%, no person score 5%`

commit 4490e29b (2023-04-26): Merge branch 'support/github_ci_matrix_build' into 'master'

Build examples with idf_ver and targets matrix

Closes #4

See merge request app-frameworks/tflite-micro-esp-examples!92
closed 2022-04-07 by vikramdattu