espressif/esp-tflite-micro #5: Loading Custom Model error: Bad Tensor

abhishek-nair-13, 2022-01-08:
I have been able to generate the default micro_speech model.

However when I put custom model generated from this repo: https://github.com/ARM-software/ML-examples/tree/master/tflu-kws-cortex-m

I get the below error after flashing the code onto ESP-EYE Dev Kit. 

`Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4
load:0x3fff0034,len:7444
load:0x40078000,len:13904
load:0x40080400,len:5296
0x40080400: _init at ??:?

entry 0x40080710
I (29) boot: ESP-IDF v4.2 2nd stage bootloader
I (29) boot: compile time 18:07:28
I (29) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) qio_mode: Enabling default flash chip QIO
I (44) boot.esp32: SPI Speed      : 80MHz
I (49) boot.esp32: SPI Mode       : QIO
I (53) boot.esp32: SPI Flash Size : 2MB
I (58) boot: Enabling RNG early entropy source...
I (63) boot: Partition Table:
I (67) boot: ## Label            Usage          Type ST Offset   Length
I (74) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (81) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (89) boot:  2 factory          factory app      00 00 00010000 00100000
I (96) boot: End of partition table
I (101) boot_comm: chip revision: 1, min. application chip revision: 0
I (108) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x16088 ( 90248) map
I (143) esp_image: segment 1: paddr=0x000260b0 vaddr=0x3ffb0000 size=0x020fc (  8444) load
I (146) esp_image: segment 2: paddr=0x000281b4 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/abhishek/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (151) esp_image: segment 3: paddr=0x000285c0 vaddr=0x40080404 size=0x07a58 ( 31320) load
I (170) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x27094 (159892) map
0x400d0020: _stext at ??:?

I (217) esp_image: segment 5: paddr=0x000570bc vaddr=0x40087e5c size=0x01ee0 (  7904) load
0x40087e5c: rtc_clk_bbpll_configure at /home/abhishek/esp/esp-idf/components/soc/src/esp32/rtc_clk.c:366

I (226) boot: Loaded app from partition at offset 0x10000
I (226) boot: Disabling RNG early entropy source...
I (227) cpu_start: Pro cpu up.
I (231) cpu_start: Application information:
I (235) cpu_start: Project name:     micro_speech
I (241) cpu_start: App version:      8ff0da9
I (246) cpu_start: Compile time:     Jan  8 2022 18:07:17
I (252) cpu_start: ELF file SHA256:  0081ee2fb39988e6...
I (258) cpu_start: ESP-IDF:          v4.2
I (263) cpu_start: Starting app cpu, entry point is 0x40081614
0x40081614: call_start_cpu1 at /home/abhishek/esp/esp-idf/components/esp32/cpu_start.c:287

I (255) cpu_start: App cpu up.
I (273) heap_init: Initializing. RAM available for dynamic allocation:
I (280) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (286) heap_init: At 3FFBB3B8 len 00024C48 (147 KiB): DRAM
I (292) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (299) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (305) heap_init: At 40089D3C len 000162C4 (88 KiB): IRAM
I (311) cpu_start: Pro cpu start user code
I (328) spi_flash: detected chip: generic
I (329) spi_flash: flash io: qio
W (329) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (340) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Bad input tensor parameters in model
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d3ccb  PS      : 0x00060930  A0      : 0x800d39ff  A1      : 0x3ffc1120  
0x400d3ccb: FeatureProvider::PopulateFeatureData(tflite::ErrorReporter*, int, int, int*) at /media/Data/tinyML/tflite-micro-esp-examples/examples/ML_KWS/build/../main/feature_provider.cc:37

A2      : 0x00000000  A3      : 0x3ffb1210  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x3ffc1164  A7      : 0x3ffb9010  A8      : 0x800f62e0  A9      : 0x3ffc0fa0  
A10     : 0x00000000  A11     : 0x3f402be4  A12     : 0x3ffc1110  A13     : 0x3ffc10f0  
A14     : 0x00000008  A15     : 0x00000025  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace:0x400d3cc8:0x3ffc1120 0x400d39fc:0x3ffc1160 0x400d321e:0x3ffc1190
0x400d3cc8: FeatureProvider::PopulateFeatureData(tflite::ErrorReporter*, int, int, int*) at /media/Data/tinyML/tflite-micro-esp-examples/examples/ML_KWS/build/../main/feature_provider.cc:36

0x400d39fc: loop at /media/Data/tinyML/tflite-micro-esp-examples/examples/ML_KWS/build/../main/main_functions.cc:142

0x400d321e: tf_main() at /media/Data/tinyML/tflite-micro-esp-examples/examples/ML_KWS/build/../main/main.cc:29 (discriminator 1)

ELF file SHA256: 0081ee2fb39988e6
`
ESP-IDF version used: 4.2

I am replacing the model.c file data with the custom model.

Can anyone please help me understand this error.

Thanks.

P.S. The default model performs poorly for identifying keyword on ESP-EYE Dev kit. It only identifies the word 'NO' with poor accuracy.

ljdsp, 2022-01-11:
I have the same error when load my Custom Model。

vikramdattu, 2022-01-11:
@abhishek-nair-13 @ljdsp 

Custom model may not work as is if input and output tensors are different than those in this example.
Please confirm in `micro_features/micro_model_settings.h` and `micro_features/micro_model_settings.c` file accordingly.

As far as low accuracy of original model is concerned, the model is for demonstration purpose. And also since ESP-EYE's mic sensitivity is low, you need to take mic closer while uttering the words.

closed 2022-11-21 by abhishek-r-nair