espressif/esp-tflite-micro #6: Edge impulse model problem between ANSI C and NN_OPTIMIZED.

MichaelDu9226, 2022-01-19:
Hi

I trained an image classification model through edge impulse, then I made some convert and customized based on this project.
In CLI_ONLY_INFERENCE mode, when config ANSI C, the result is correct. But when using NN_OPTIMIZED, the result is incorrect.

My custom work：
1. update new image 0-7, 0-3 is Qianzi(钳子)，4-7 is PCB
2. add some needed OpResolver https://github.com/MichaelDu9226/tflite-micro-esp-examples/blob/f1a42c308896cd588d3897e19c2b9fdab53c5b28/examples/person_detection/main/main_functions.cc#L94

 Is it an optimized operator problem?

This is my fork project, including models and images(0-7) used.
https://github.com/MichaelDu9226/tflite-micro-esp-examples/commit/f1a42c308896cd588d3897e19c2b9fdab53c5b28

vikramdattu, 2022-01-19:
@MichaelDu9226 thanks for the reporting!
Will try your example and revert back with the solution.

MichaelDu9226, 2022-01-27:
Hi, @vikramdattu Any updates on this issue? Looking forward to your solution.
Thanks

vikramdattu, 2022-01-27:
@MichaelDu9226 I am able to reproduce the issue and working on it.
Please use ANSI versions path for now. Will post the update here.

MichaelDu9226, 2022-01-27:
ok. Thanks for your reply.

vikramdattu, 2022-01-31:
Hi @advaitjain,
Can you please check the issue and if there is a known remedy for this?

As per my observation, for this particular case, tflite memory allocator ends up assigning same memory pointer for output tensor and scratch buffer.

MichaelDu9226, 2022-02-09:
@vikramdattu 
I have tested your pull request. The detect_image result is ok now.
Thanks.

closed 2022-03-22 by vikramdattu