espressif/esp-tflite-micro #7: Esp32 crashing when calling interpreter->Invoke()

arcticlimer, 2022-01-22:
Hello! I'm trying to use tensorflow-lite and my AI-THINKER ESP32-CAM is crashing whenever I call `interpreter->Invoke()`. I'm suspecting this has something to do with the available opcodes, since the crash happens only if I'm using AllOpsResolver. If I try to select the following operations
```cpp
  static tflite::MicroMutableOpResolver<6> micro_op_resolver;
  micro_op_resolver.AddReshape();
  micro_op_resolver.AddAveragePool2D();
  micro_op_resolver.AddConv2D();
  micro_op_resolver.AddDepthwiseConv2D();
  micro_op_resolver.AddMaxPool2D();
  micro_op_resolver.AddSoftmax();
```
It errors out with "Didn't find op for builtin opcode 'MUL' version '2'". The model that I'm using is a fine-tuned MobileNetV2 built with TensorFlow 2.

The logs from the monitor:
```
I (27) boot: ESP-IDF v4.3.2 2nd stage bootloader                                                                                                                                                                                  
I (27) boot: compile time 00:51:42                                                                                                                                                                                                
I (27) boot: chip revision: 3                                                                                                                                                                                                     
I (30) boot_comm: chip revision: 3, min. bootloader chip revision: 0                                                                                                                                                              
I (37) boot.esp32: SPI Speed      : 40MHz                                                                                                                                                                                         
I (42) boot.esp32: SPI Mode       : DIO                                                                                                                                                                                           
I (46) boot.esp32: SPI Flash Size : 4MB                                                                                                                                                                                           
I (51) boot: Enabling RNG early entropy source...                                                                                                                                                                                 
I (56) boot: Partition Table:                                                                                                                                                                                                     
I (60) boot: ## Label            Usage          Type ST Offset   Length                                                                                                                                                           
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000                                                                                                                                                         
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000                                                                                                                                                         
I (82) boot:  2 factory          factory app      00 00 00010000 0036b000                                                                                                                                                         
I (90) boot: End of partition table                                                                              
I (94) boot_comm: chip revision: 3, min. application chip revision: 0                                            
I (101) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2b0ae0h (2820832) map                           
I (1189) esp_image: segment 1: paddr=002c0b08 vaddr=3ffb0000 size=03b18h ( 15128) load                           
I (1195) esp_image: segment 2: paddr=002c4628 vaddr=40080000 size=0b9f0h ( 47600) load   
I (1216) esp_image: segment 3: paddr=002d0020 vaddr=400d0020 size=727fch (468988) map                                                                                                                                    
I (1396) esp_image: segment 4: paddr=00342824 vaddr=4008b9f0 size=05884h ( 22660) load                           
I (1406) esp_image: segment 5: paddr=003480b0 vaddr=50000000 size=00010h (    16) load                           
I (1415) boot: Loaded app from partition at offset 0x10000                                                       
I (1415) boot: Disabling RNG early entropy source...
I (1427) psram: This chip is ESP32-D0WD        
I (1429) spiram: Found 64MBit SPI RAM device                                                                     
I (1429) spiram: SPI RAM mode: flash 40m sram 40m                                                                
I (1433) spiram: PSRAM initialized, cache is in low/high (2-core) mode.                                          
I (1440) cpu_start: Pro cpu up.                                                                                                                                                                                                   
I (1444) cpu_start: Starting app cpu, entry point is 0x400814dc                                                  
0x400814dc: call_start_cpu1 at /home/vini/esp-idf-v4.3.2/components/esp_system/port/cpu_start.c:150                                                                                           
I (0) cpu_start: App cpu up.                                                                                     
I (2335) spiram: SPI SRAM memory test OK                                                                                                                                                                                          
I (2344) cpu_start: Pro cpu start user code                                                                      
I (2344) cpu_start: cpu freq: 240000000                                                                                                                                                                                           
I (2344) cpu_start: Application information:                                                                                                                                                                                      
I (2347) cpu_start: Project name:     esp32cam_pet_feeder                                                        
I (2353) cpu_start: App version:      6bb7d41-dirty                                                                                                                                                                               
I (2358) cpu_start: Compile time:     Jan 21 2022 00:51:55                                                       
I (2364) cpu_start: ELF file SHA256:  2ff88e1fc76ad1da...                                                                                                                                                                         
I (2370) cpu_start: ESP-IDF:          v4.3.2            
I (2376) heap_init: Initializing. RAM available for dynamic allocation:                                          
I (2383) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM                                                       
I (2389) heap_init: At 3FFB59F0 len 0002A610 (169 KiB): DRAM                                                     
I (2395) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM                                                    
I (2402) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM                                                   
I (2408) heap_init: At 40091274 len 0000ED8C (59 KiB): IRAM                                                      
I (2414) spiram: Adding pool of 4096K of external SPI memory to heap allocator                                   
I (2424) spi_flash: detected chip: generic
I (2427) spi_flash: flash io: dio          
I (2432) cpu_start: Starting scheduler on PRO CPU.
[... cut ...]
0x4012f4c3: esp_task_wdt_reset at /home/vini/esp-idf-v4.3.2/components/esp_common/src/task_wdt.c:336

0x4012f4cf: idle_hook_cb at /home/vini/esp-idf-v4.3.2/components/esp_common/src/task_wdt.c:88

0x400da0e5: esp_vApplicationIdleHook at /home/vini/esp-idf-v4.3.2/components/esp_common/src/freertos_hooks.c:51 (discriminator 1)

0x4008c1cd: prvIdleTask at /home/vini/esp-idf-v4.3.2/components/freertos/tasks.c:3846 (discriminator 1)

0x4008d8f1: vPortTaskWrapper at /home/vini/esp-idf-v4.3.2/components/freertos/port/xtensa/port.c:168
```

vikramdattu, 2022-01-27:
Hi @arcticlimer possible to share the bug-reproducible branch?

arcticlimer, 2022-01-31:
Yes. Probably in the next days I'll make a reproducible example and share it.

JVKran, 2022-02-07:
Hi there!

This appears to be the same issue as #10. The WDT triggers when the CPU is to busy to perform management tasks. I had the same issue. This can be solved by disabling the WDT in the menuconfig or to add a vTaskDelay(100) in the loop. 

Kind regards,

Jochem

vikramdattu, 2022-02-11:
Hi @arcticlimer, as @JVKran suggested, did you try increasing time for WDT (or disable it completely) or adding some small delay in the `loop`?

arcticlimer, 2022-02-13:
Hello, sorry for the slow response, I couldn't work in my ESP32 + tflite project in the last days. Indeed the problem was that my prediction was taking ~8 seconds and watchdog didn't like it.

arcticlimer, 2022-02-13:
As a suggestion, not sure if this can be done outside tflite-micro source code, maybe adding some kind of heartbeat calling code to watchdog to avoid this kind of error.

vikramdattu, 2022-02-13:
@arcticlimer putting delay in the code is surely needs touching common code.
As a fallback, you can increase WDT time to some large value or disable it altogether via menuconfig. You may consider adding this change to `sdkconfig.defaults`.

The crash not `really a crash/corruption` but gdb printing some debug info at the moment WDT expired. This can also be disabled via menuconfig.