espressif/esp-tflite-micro #9: Can not include idf head files

ljdsp, 2022-01-25:
in this project, I can not include idf head files, for example in main.cc:
#include "esp_http_client.h"
#include "esp_console.h"
#include "nvs_flash.h"
#include "protocol_examples_common.h"
the error info as follows:

[2/11] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/main.cc.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/main.cc.obj
ccache D:\Users\jolin\.espressif\tools\xtensa-esp32-elf\esp-2021r1-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe   -Iconfig -I../main -ID:/l00302745/esp/develop/idf/components/newlib/platform_include -ID:/l00302745/esp/develop/idf/components/freertos/include -ID:/l00302745/esp/develop/idf/components/freertos/port/xtensa/include -ID:/l00302745/esp/develop/idf/components/esp_hw_support/include -ID:/l00302745/esp/develop/idf/components/esp_hw_support/port/esp32/. -ID:/l00302745/esp/develop/idf/components/heap/include -ID:/l00302745/esp/develop/idf/components/log/include -ID:/l00302745/esp/develop/idf/components/lwip/include/apps -ID:/l00302745/esp/develop/idf/components/lwip/include/apps/sntp -ID:/l00302745/esp/develop/idf/components/lwip/lwip/src/include -ID:/l00302745/esp/develop/idf/components/lwip/port/esp32/include -ID:/l00302745/esp/develop/idf/components/lwip/port/esp32/include/arch -ID:/l00302745/esp/develop/idf/components/soc/include -ID:/l00302745/esp/develop/idf/components/soc/esp32/. -ID:/l00302745/esp/develop/idf/components/soc/esp32/include -ID:/l00302745/esp/develop/idf/components/hal/esp32/include -ID:/l00302745/esp/develop/idf/components/hal/include -ID:/l00302745/esp/develop/idf/components/esp_rom/include -ID:/l00302745/esp/develop/idf/components/esp_rom/esp32 -ID:/l00302745/esp/develop/idf/components/esp_rom/include/esp32 -ID:/l00302745/esp/develop/idf/components/esp_common/include -ID:/l00302745/esp/develop/idf/components/esp_system/include -ID:/l00302745/esp/develop/idf/components/esp32/include -ID:/l00302745/esp/develop/idf/components/driver/include -ID:/l00302745/esp/develop/idf/components/driver/esp32/include -ID:/l00302745/esp/develop/idf/components/esp_ringbuf/include -ID:/l00302745/esp/develop/idf/components/efuse/include -ID:/l00302745/esp/develop/idf/components/efuse/esp32/include -ID:/l00302745/esp/develop/idf/components/xtensa/include -ID:/l00302745/esp/develop/idf/components/xtensa/esp32/include -ID:/l00302745/esp/develop/idf/components/espcoredump/include -ID:/l00302745/esp/develop/idf/components/esp_timer/include -ID:/l00302745/esp/develop/idf/components/esp_ipc/include -ID:/l00302745/esp/develop/idf/components/esp_pm/include -ID:/l00302745/esp/develop/idf/components/vfs/include -ID:/l00302745/esp/develop/idf/components/esp_wifi/include -ID:/l00302745/esp/develop/idf/components/esp_wifi/esp32/include -ID:/l00302745/esp/develop/idf/components/esp_event/include -ID:/l00302745/esp/develop/idf/components/esp_netif/include -ID:/l00302745/esp/develop/idf/components/esp_eth/include -ID:/l00302745/esp/develop/idf/components/tcpip_adapter/include -ID:/l00302745/esp/develop/idf/components/app_trace/include -ID:/l00302745/develop/tflite-micro-esp-examples/components/tflite-lib -ID:/l00302745/develop/tflite-micro-esp-examples/components/tflite-lib/third_party/gemmlowp -ID:/l00302745/develop/tflite-micro-esp-examples/components/tflite-lib/third_party/flatbuffers/include -ID:/l00302745/develop/tflite-micro-esp-examples/components/tflite-lib/third_party/ruy -ID:/l00302745/develop/tflite-micro-esp-examples/components/tflite-lib/third_party/kissfft -ID:/l00302745/esp/develop/idf/components/spi_flash/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -O2 -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -std=gnu++11 -fno-exceptions -fno-rtti -D_GNU_SOURCE -DIDF_VER=\"v4.3.1\" -DESP_PLATFORM -DTF_LITE_STATIC_MEMORY -MD -MT esp-idf/main/CMakeFiles/__idf_main.dir/main.cc.obj -MF esp-idf\main\CMakeFiles\__idf_main.dir\main.cc.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/main.cc.obj -c ../main/main.cc
**../main/main.cc:26:10: fatal error: esp_http_client.h: No such file or directory**
 #include "esp_http_client.h"
          ^~~~~~~~~~~~~~~~~~~

vikramdattu, 2022-01-25:
@ljdsp 
Which example are you trying to build?
If you are creating your own application, you might have missed some steps.
Easier way is to copy one of the existing examples and make  changes to that.

ljdsp, 2022-01-25:
I want to use hettp、socket,、console feature in the example  "micro_speech", but the compile error said "../main/main.cc:25:10: fatal error: esp_http_client.h: No such file or directory".  I modify main.cc code as follows, 
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <sys/time.h>

#include "esp_log.h"
#include "esp_system.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "main_functions.h"
#include "esp_http_client.h"
#include "esp_console.h"
#include "nvs_flash.h"
#include "protocol_examples_common.h"

void tf_main(void) {
  setup();
  while (true) {
    loop();
  }
}

void ld_http_init(void)
{
    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    /* This helper function configures Wi-Fi or Ethernet, as selected in menuconfig.
     * Read "Establishing Wi-Fi or Ethernet Connection" section in
     * examples/protocols/README.md for more information about this function.
     */
    ESP_ERROR_CHECK(example_connect());
    ESP_LOGI(TAG, "Connected to AP, begin http example");
}

extern "C" void app_main() {
  ld_http_init();
  xTaskCreate((TaskFunction_t)&tf_main, "tensorflow", 8 * 1024, NULL, 8, NULL);
  vTaskDelete(NULL);
}

vikramdattu, 2022-01-25:
@ljdsp you must also list those components as REQUIRES in CMakeLists.txt file.
e.g., https://github.com/espressif/tflite-micro-esp-examples/blob/60ac067014509cd378e6933bf744c2c6610fd8a4/examples/micro_speech/main/CMakeLists.txt#L11

ljdsp, 2022-01-25:
ok. thanks

closed 2022-01-27 by vikramdattu