espressif/esp-tflite-micro #10: esp32 and esp32s3 person detection gets stuck and watch dog timer resets the board

mocleiri, 2022-01-31:
I have an m5 timer camera which has 8MB PSRAM and 4MB Flash and an OV3660 camera.

I add the CONFIG_OV3660_SUPPORT=y sdk configuration option.

When I connect to the board after flashing it repeatedly resets with a WDT timer error on the PRO cpu.

### ESP32 with SPRAM

I compiled examples/person_detection and flashed it to the board which then produced this output repeatedly.

```
rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7592
load:0x40078000,len:15572
load:0x40080400,len:4392
entry 0x4008069c
I (161) boot: ESP-IDF v4.3.2-216-gf1adc8efbe 2nd stage bootloader
I (161) boot: compile time 20:07:33
I (161) boot: chip revision: 3
I (166) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (173) qio_mode: Enabling default flash chip QIO
I (178) boot.esp32: SPI Speed      : 40MHz
I (183) boot.esp32: SPI Mode       : QIO
I (188) boot.esp32: SPI Flash Size : 4MB
W (192) boot.esp32: PRO CPU has been reset by WDT.
W (198) boot.esp32: WDT reset info: PRO CPU PC=0x40008547
W (204) boot.esp32: WDT reset info: APP CPU PC=0x40008547
I (210) boot: Enabling RNG early entropy source...
I (216) boot: Partition Table:
I (219) boot: ## Label            Usage          Type ST Offset   Length
I (227) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (234) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (242) boot:  2 factory          factory app      00 00 00010000 00180000
I (249) boot: End of partition table
I (254) boot_comm: chip revision: 3, min. application chip revision: 0
I (261) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=79afch (498428) map
I (423) esp_image: segment 1: paddr=00089b24 vaddr=3ffb0000 size=029c8h ( 10696) load
I (427) esp_image: segment 2: paddr=0008c4f4 vaddr=40080000 size=03b24h ( 15140) load
I (434) esp_image: segment 3: paddr=00090020 vaddr=400d0020 size=4ad38h (306488) map
I (532) esp_image: segment 4: paddr=000dad60 vaddr=40083b24 size=08484h ( 33924) load
I (544) esp_image: segment 5: paddr=000e31ec vaddr=50000000 size=00010h (    16) load
I (551) boot: Loaded app from partition at offset 0x10000
I (551) boot: Disabling RNG early entropy source...
I (565) cpu_start: Pro cpu up.
I (565) cpu_start: Starting app cpu, entry point is 0x40081284
I (0) cpu_start: App cpu up.
ets Jul 29 2019 12:21:46
```

I'm tying into this project as a submodule in my [tensorflow-micropython-examples](https://github.com/mocleiri/tensorflow-micropython-examples) project.

On my end we've observed the same behavior with the ansi optimized functions on esp32 and for the esp32s3 optimized functions.
https://github.com/mocleiri/tensorflow-micropython-examples/issues/9

Below are the traces we see when running in micropython.  We are compiling the esp32 using idf 4.3.2 and esp32s3 using idf 4.4 at commit: 142bb32c50fa9875b8b69fa539a2d59559460d72 (this is what micropython uses)

### ESP32 SPRAM

```
MicroPython e891475-dirty on 2022-01-27; ESP32 module (microlite) with ESP32
Type "help()" for more information.
>>> import run
interpreter_make_new: model size = 18712, tensor area = 8208
Starting
found - no at 0 seconds -> 1

W (171) boot.esp32: PRO CPU has been reset by WDT.
W (172) boot.esp32: WDT reset info: PRO CPU PC=0x40094c83
W (173) boot.esp32: WDT reset info: APP CPU PC=0x402453aa (waiti mode)
```

### ESP32S3 SPIRAM

```
18712
interpreter_make_new: model size = 18712, tensor area = 8208
Starting
found - no at 0 seconds -> 1

Guru Meditation Error: Core 1 panic'ed (InstructionFetchError). Exception was unhandled.

Core 1 register dump:
PC : 0x3fce9f00 PS : 0x00060430 A0 : 0x8200cbf2 A1 : 0x3fce48a0
A2 : 0x3fceaa64 A3 : 0x3fce3ed4 A4 : 0x00000003 A5 : 0x00000000
A6 : 0x00000000 A7 : 0x3d83f218 A8 : 0x8200c119 A9 : 0x3fce4880
A10 : 0x3fceaa04 A11 : 0x3d83f218 A12 : 0x3fce9f00 A13 : 0x00000000
A14 : 0x3d83e2c0 A15 : 0x3d83cc2c SAR : 0x0000000b EXCCAUSE: 0x00000002
EXCVADDR: 0x3fce9f00 LBEG : 0x42025065 LEND : 0x42025084 LCOUNT : 0x00000000

Backtrace:0x3fce9efd:0x3fce48a0 |<-CORRUPTED
```
###  ESP32S3 no PS RAM
```
18712
interpreter_make_new: model size = 18712, tensor area = 8208
Starting
found - no at 0 seconds -> 1

Guru Meditation Error: Core 1 panic'ed (InstructionFetchError). Exception was unhandled.

Core 1 register dump:
PC : 0x3fce6f98 PS : 0x00060830 A0 : 0x8200cbf2 A1 : 0x3fce4610
A2 : 0x3fce7bac A3 : 0x3fce3c44 A4 : 0x00000003 A5 : 0x00000000
A6 : 0x00000000 A7 : 0x3fcb20b8 A8 : 0x8200c119 A9 : 0x3fce45f0
A10 : 0x3fce7b4c A11 : 0x3fcb20b8 A12 : 0x3fce6f98 A13 : 0x00000000
A14 : 0x3fcb1160 A15 : 0x3fcafacc SAR : 0x0000000b EXCCAUSE: 0x00000002
EXCVADDR: 0x3fce6f98 LBEG : 0x42025065 LEND : 0x42025084 LCOUNT : 0x00000000

Backtrace:0x3fce6f95:0x3fce4610 |<-CORRUPTED
```

mocleiri, 2022-01-31:
I realized that on the micropython side we were having problems with micro_speech but I tested the person_detection instead.  I went back and built the micro_speech example and after reconfiguring the pins for my microphone it worked without a problem.

I need to retest person_detection and this time change the pins and then see what is happening.

mocleiri, 2022-01-31:
I changed the pins on person_detection to CONFIG_CAMERA_MODEL_M5STACK_PSRAM=y and reflashed but I still get the same issue where the board keeps restarting.

I'll debug on device for the micro_speech example on my end and report back with any findings.

RSC-Games, 2022-05-02:
I couldn't help but notice that you were using MicroPython. I had previously written a C user module that created bindings to FreeRTOS runtime stats, and whenever more than 7 tasks were running, I got a WDT reset. Once, however, I managed to uncover a DoubleException. With more digging, I found that both Core 0 and Core 1 were entering the panic handler, and no printout happened.
The cause was the fact that I didn't allocate enough RAM for the buffer storing the runtime stats, and I figured out this:
Core 0, inside of FreeRTOS, wrote out of bounds, and likely corrupted the heap or another task stack.
Core 1 was likely affected by this.
Then, because interrupts were off during the exception, no printout happened. And another exception caused Core 0 to re-enter the panic handler.
And since interrupts were off, no printout.
So, the issue could actually be something that simple.

Of course, since your backtrace is corrupted, you can't post one, but maybe if you used GDB and poked around in there, it would help. (Assuming you don't have a JTAG debugger.) Or, if you sent me a copy of the ELF file I could decode the exception and give you my findings.

vikramdattu, 2023-02-02:
@mocleiri closing this issue. Please create a new issue, if you still face issues with latest version.

closed 2023-02-02 by vikramdattu