espressif/esp-tflite-micro #12: ESP-NN vs ESP-DL

JVKran, 2022-02-07:
Hi there!

I'm currently in the process of evaluating which path to go for image classification with an ESP-EYE. I currently see three options.

1. Running the exported Edge Impulse project. This however, is very slow. It classifies the images at around once per second. This is working currently.
2. Running the exported tflite model created with the tensorflow lite modelmaker. This is quite a bit faster at around 500-300ms
    on an ESP32. This is also working.
3. Converting the model to something runnable by the ESP-Deep Learning (ESP-DL) framework. Haven't got this to work yet.

Now, I've actually got two (related) questions. I'm very happy with the performance increase ESP-NN provides. I do however, wonder what boost I can expect with the ESP32-S3. Somehwere in between 3 and 6 times as fast as the ESP32?

Furthermore, I'm wondering what performance difference you think there would be with using the ESP-DL. It's quite a hassle to convert a TFlite model to something acceptable by the ESP-DL. Even more using a camera since practically everything is hidden behind precompiled libraries... Using the ESP-DL is going to be a matter of costs and benefits. If the performance difference is so large, we might go with that. Otherwise, I'd love to stick around with ESP-NN.

Kind regards,

Jochem

vikramdattu, 2022-02-11:
Hi @JVKran,

> I do however, wonder what boost I can expect with the ESP32-S3. Somewhere in between 3 and 6 times as fast as the ESP32?

Depending upon the OPs your model uses, you can expect good performance boost over ESP32. Easily ~6x(If most layers are Conv and DepthwiseConv). (Keeping SPIRAM modes Quad/80MHz for ESP32S3 to match ESP32 config)

> Furthermore, I'm wondering what performance difference you think there would be with using the ESP-DL.

ESP-DL might bring you ~2x or more performance improvement over tflite+ESP-NN solution for ESP32-S3 depending upon the model.

Some important reasons why ESP-DL outweighs tflite+ESP-NN are:
1. Quantization spec:
a. tflite: ` real_value = (int8_value - zero_point) * scale`
b. esp-dl: `real_value = int_value * 2^{exponent}`. [spec](https://github.com/espressif/esp-dl/blob/master/docs/en/quantization_specification.md)
ESP-DL reduces the rescaling operation to negligible cycles whereas, tflite has [complex operation](https://github.com/espressif/tflite-micro-esp-examples/blob/e22baad17f127f6003ad21107064670397c35a54/components/tflite-lib/tensorflow/lite/kernels/internal/reference/integer_ops/conv.h#L114).

2. Compiled model
- `esp-dl` uses compiled model structure, as opposed to tflite's embedding complete model. This gives `esp-dl` benefits of placement of each weight matrix into external/internal RAM and also the control over alignment of each. Please take a look into [tutorial](https://github.com/espressif/esp-dl/tree/master/tutorial) and try the [convert_tool](https://github.com/espressif/esp-dl/tree/master/tools/convert_tool)
- Edge Impulse's model compilation feature can be used to achieve this. Will give you control over placement of each weight into internal/external RAM.

> It's quite a hassle to convert a TFlite model to something acceptable by the ESP-DL. Even more using a camera since practically everything is hidden behind precompiled libraries... Using the ESP-DL [...]

I agree, this is partly because some OPs are not commonly supported by ESP-DL vs TfLite and partly because it is new platform to learn.
If performance is satisfactory considering ~6x over ESP32 for tflite+ESP-NN solution, It should be okay to stick to it.
If performance is critical still, I would consider ESP-DL for further performance and flexibility of memory allocations.

Thanks,
Vikram

JVKran, 2022-04-06:
Hi @vikramdattu,

Thanks for your thorough explanation. I went for simply using ESP-NN as it's so increcibly convenient. Thanks for this awesome component!

Kind regards,

Jochem

closed 2022-04-06 by JVKran