espressif/esp-tflite-micro #13: Couldn't allocate memory

ZlodeiBaal, 2022-02-10:
Hi!
I tried to run person detection on my ESP32 CAM. Program is starting on device, but then "Couldn't allocate memory" is happening.
What i have tryed:
1. Start in CLI Inferencing - same error
2. set CONFIG_CAMERA_MODEL_AI_THINKER=y according to my camera model - same error
3. Downgrade to v4.2 accirding to this issue - https://github.com/espressif/tflite-micro-esp-examples/issues/2  - same error

As target I use:
`idf.py set-target esp32`

log:

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7448
load:0x40078000,len:13904
load:0x40080400,len:5296
0x40080400: _init at ??:?

entry 0x40080710
I (27) boot: ESP-IDF v4.2.2 2nd stage bootloader
I (27) boot: compile time 11:53:19
I (27) boot: chip revision: 1
I (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (37) qio_mode: Enabling default flash chip QIO
I (42) boot.esp32: SPI Speed      : 80MHz
I (47) boot.esp32: SPI Mode       : QIO
I (51) boot.esp32: SPI Flash Size : 4MB
I (56) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (80) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  2 factory          factory app      00 00 00010000 00180000
I (95) boot: End of partition table
I (99) boot_comm: chip revision: 1, min. application chip revision: 0
I (106) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x67cf8 (425208) map
I (240) esp_image: segment 1: paddr=0x00077d20 vaddr=0x3ffb0000 size=0x032c4 ( 12996) load
I (245) esp_image: segment 2: paddr=0x0007afec vaddr=0x40080000 size=0x0502c ( 20524) load
0x40080000: _WindowOverflow4 at D:/Programs/esp-idf-v4.2.2/components/freertos/xtensa/xtensa_vectors.S:1730

I (254) esp_image: segment 3: paddr=0x00080020 vaddr=0x400d0020 size=0x546d8 (345816) map
0x400d0020: _stext at ??:?

I (357) esp_image: segment 4: paddr=0x000d4700 vaddr=0x4008502c size=0x06900 ( 26880) load
0x4008502c: xRingbufferCreate at D:/Programs/esp-idf-v4.2.2/components/esp_ringbuf/ringbuf.c:846 (discriminator 1)

I (374) boot: Loaded app from partition at offset 0x10000
I (374) boot: Disabling RNG early entropy source...
I (374) cpu_start: Pro cpu up.
I (378) cpu_start: Application information:
I (383) cpu_start: Project name:     person_detection
I (389) cpu_start: App version:      1
I (393) cpu_start: Compile time:     Feb 10 2022 11:52:57
I (399) cpu_start: ELF file SHA256:  001690ba44f08755...
I (405) cpu_start: ESP-IDF:          v4.2.2
I (410) cpu_start: Starting app cpu, entry point is 0x4008192c
0x4008192c: call_start_cpu1 at D:/Programs/esp-idf-v4.2.2/components/esp32/cpu_start.c:287

I (403) cpu_start: App cpu up.
I (420) heap_init: Initializing. RAM available for dynamic allocation:
I (427) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (434) heap_init: At 3FFB53B0 len 0002AC50 (171 KiB): DRAM
I (440) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (446) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (453) heap_init: At 4008B92C len 000146D4 (81 KiB): IRAM
I (459) cpu_start: Pro cpu start user code
I (477) spi_flash: detected chip: generic
I (477) spi_flash: flash io: qio
I (477) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Couldn't allocate memory of 189440 bytes
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d4f4a  PS      : 0x00060530  A0      : 0x800d495d  A1      : 0x3ffbb270
0x400d4f4a: loop at e:\openproject\esp32\tflite-micro-esp-examples-master\examples\person_detection\build/../main/main_functions.cc:113

A2      : 0x3ffb3a20  A3      : 0x00000000  A4      : 0x0002e400  A5      : 0x3ffb3a10
A6      : 0xb33fffff  A7      : 0x00000000  A8      : 0x800d4b68  A9      : 0x3ffbb1e0
A10     : 0x3ffb1250  A11     : 0x3ffb8f10  A12     : 0x00000060  A13     : 0x00000000
A14     : 0x00000804  A15     : 0x00000008  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000010  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace:0x400d4f47:0x3ffbb270 0x400d495a:0x3ffbb290
0x400d4f47: loop at e:\openproject\esp32\tflite-micro-esp-examples-master\examples\person_detection\build/../main/main_functions.cc:113

0x400d495a: tf_main() at e:\openproject\esp32\tflite-micro-esp-examples-master\examples\person_detection\build/../main/main.cc:36 (discriminator 1)

ELF file SHA256: 001690ba44f08755

Rebooting...
```

vikramdattu, 2022-02-11:
Hi @ZlodeiBaal looks like you are running out of memory.

In cli only mode and for ESP32, you can reduce `constexpr int kTensorArenaSize = ` define to `135 * 1024` and try with internal memory.

For Camera and/or ESP32-S3 with ESP-NN optimizations enabled, please make sure that external RAM is enabled for your example and try again.

vikramdattu, 2022-04-25:
@ZlodeiBaal is the issue still relevant for you?

ZlodeiBaal, 2022-04-25:
@vikramdattu No, thanks. I have found another way and run my NN through micropython.

closed 2022-04-26 by vikramdattu