espressif/esp-tflite-micro #16: Running a regression model on Esp32

Raunak-Singh-Inventor, 2022-04-24:
I am trying to run a Regression model on the edukit which takes a array of 7 numbers and (is supposed to) output a array of 5 numbers.

 I can run the model using TensorFlow python on my Windows. But when I run it on the Esp32, I don't get a meaningful output.

[image]

I expected to see an output of: `[100, 100, 100, 100, 100]`.

Here is my code: https://github.com/Raunak-Singh-Inventor/Stockly/blob/master/edukit_deployment/main/tflite/main_functions.cpp

vikramdattu, 2022-04-25:
Hi @Raunak-Singh-Inventor , I checked your model and it is float model.
both Input and Output are float. Whereas while treating the input, you are using int8: https://github.com/Raunak-Singh-Inventor/Stockly/blob/0e14495d044aa1cab64e60a74598e62477cfbefe/edukit_deployment/main/tflite/main_functions.cpp#L85

Raunak-Singh-Inventor, 2022-04-26:
Thanks for the solution changing the input to float instead of int8: `model_input_buffer = model_input->data.f;` worked.

closed 2022-04-26 by Raunak-Singh-Inventor