espressif/esp-tflite-micro #17: Custom micro speech model slice size, count, stride and duration

JVKran, 2022-04-28:
Hi there!

Me again, but now with speech! I'm having a hard time with determining the values in micro_model_settings for a TensorFlow Lite Micro model generated with Edge Impulse. It's the tutorial project which can be found [here](https://studio.edgeimpulse.com/public/14225/latest).

When I paste the model into ```model.cc```, update the model size and category-count & labels and last but not least, set the slice size and count to correspond to ```model_input->dims->data[1]``` everything runs correctly. The recognization of 'helloworld' fails, however, and I can't find any sources on determining the values for the slice size, count, stride and duration.

Can anyone point me in the right direction?

Kind regards,

Jochem

JVKran, 2022-04-28:
For completeness, I've also opened up [an issue over at the forum of Edge Impulse](https://forum.edgeimpulse.com/t/slice-size-count-stride-and-duration/4321/5). For now, this seems like an implementation specific issue/question.

vikramdattu, 2022-04-29:
Hi @JVKran I checked into the code from cpp sdk, there already is a framework put by edge impulse to interpret the values. 
The `trigger` function present in the classifier of `edge_impulse_sdk` has this logic.

Also, look for the `run_classifier_continuous` function from `ei_run_classifier.h` to gain insight on how the raw audio input is treated and fed to the model.

Hope this is useful.