espressif/esp-tflite-micro #20: fatal error: esp_timer.h: No such file or directory

tygerlord, 2022-07-04:
Hello, got this 

idf.py build fail with message

esp/tflite-micro-esp-examples/components/tflite-lib/tensorflow/lite/micro/kernels/esp_nn/add.cc:31:10: fatal error: esp_timer.h: No such file or directory
   31 | #include <esp_timer.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
[121/208] Building CXX object esp-idf/tflite-lib/CMakeFiles/__idf_tflite-lib.dir/tensorflow/lite/micro/kernels/zeros_like.cc.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

Help welcome

vikramdattu, 2022-07-05:
@tygerlord What is your IDF version?
Please add `esp_timer` to following line and check if this works?
https://github.com/espressif/tflite-micro-esp-examples/blob/4b9a1732e2f0901ec33ab8337aad5469691d4e98/components/tflite-lib/CMakeLists.txt#L58

The line will change to ``` REQUIRES "esp_timer" "esp-nn") ```

tygerlord, 2022-07-05:
I use ESP-IDF v5.0-dev-3654-gc2ccc383da
(I've tried with previous v4.4.1 same erros if I remember, but perhaps I've misconfigured it)

Adding 'REQUIRE "esp_timer"' seems doing the job , unfortunatly got another error:
 
/tflite-micro-esp-examples/components/screen/controller_driver/ili9341/ili9341.c:19:10: fatal error: driver/gpio.h: No  such file or directory
   19 | #include "driver/gpio.h"

I will search.

Thank you very much @vikramdattu

closed 2022-07-05 by tygerlord

davidallenmann, 2022-07-22:
Using IDF v5.0 and get the same error compiling hello_world.
After adding require "esp_timer" I get an error indicating that the esp32-camera is not supported by v5 

```
CMake Warning at /Users/xx/w/tflite-micro-esp-examples/components/esp32-camera/CMakeLists.txt:78 (message):
  ESP-IDF version detected: '5.0'.
```

tygerlord, 2022-07-23:
Thank you @davidallenmann for your comment.

I've noticed too that bus get in trouble. In fact  for all examples I've tried to compile I get missing files like gpio.h, or led.h or something related to bus...

I've no managed to build any exmple ;-(

vikramdattu, 2022-07-25:
@tygerlord please use IDF branch `release/v4.4`.
Under development branches `master` and `release/v5.0` branch are not supported yet.