espressif/esp-tflite-micro #24: Hello World Build and set target failed for ESP32-S3

priyanshu-makhiyaviya, 2022-09-16:
@espressif/tflite-micro-esp-examples

System information

    Host OS Platform and Distribution (e.g., Linux Ubuntu): Ubuntu 18.04
    Target platform: ESP32-S3
    IDF version release/v4.4
[esspressif_tfliteMicro_helloword_esp32-s3_.txt](https://github.com/espressif/tflite-micro-esp-examples/files/9581706/esspressif_tfliteMicro_helloword_esp32-s3_.txt)

Describe the problem

I am trying to set up the tflite micro esp HelloWorld example for ESP32-S3.
when I tried to set the target using the command "idf.py set-target esp32s3", CMake failed.
 
_[esspressif_tfliteMicro_helloword_esp32-s3_.txt](https://github.com/espressif/tflite-micro-esp-examples/files/9581698/esspressif_tfliteMicro_helloword_esp32-s3_.txt)_

Please let me know how can i build and flash micro esp HelloWorld example for ESP32-S3.

Steps Followed:

Clone the TensorFlow repository:
https://github.com/espressif/tflite-micro-esp-examples.git

Changed directory to example/hellowolrd
cd /examples/hello_world

Set the IDF_TARGET
idf.py set-target esp32s3

Attached is the log file for reference.
Any suggestions for this issue?

vikramdattu, 2022-09-16:
@priyanshu-makhiyaviya Unfortunately, this is not re-producible at my end. Must be some CMake + OS specific issue.

Can you please remove below line and try?

https://github.com/espressif/tflite-micro-esp-examples/blob/2a93aa3106f181768d75f64bcac629f344a2ca22/components/tflite-lib/CMakeLists.txt#L81

priyanshu-makhiyaviya, 2022-09-16:
Thank you @vikramdattu , The build was successful after removing the above-mentioned line from CmakeList

After building the flash has not worked properly. It has stopped at "waiting for download"

	_--- idf_monitor on /dev/ttyACM0 115200 ---
	--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
	ESP-ROM:esp32s3-20210327
	Build:Mar 27 2021
	rst:0x15 (USB_UART_CHIP_RESET),boot:0x0 (DOWNLOAD(USB/UART0))
	Saved PC:0x40041a76
	waiting for download_

The full log file is attached for reference.
[2_esspressif_tfliteMicro_helloword_esp32-s3_.txt](https://github.com/espressif/tflite-micro-esp-examples/files/9583544/2_esspressif_tfliteMicro_helloword_esp32-s3_.txt)

Any suggestions for this issue?

vikramdattu, 2022-09-19:
@priyanshu-makhiyaviya You've left the device in download mode. Simple reset should bring it in execution mode.

priyanshu-makhiyaviya, 2022-09-19:
Hi @vikramdattu ,
Thank you for your support. It worked after resetting and flashing again.

There is one more thing that I really need to understand is,
- What is the significance of removing that line from CMakeList.txt?
- What is the meaning of that line? 

Please help me to understand this.

Thank you.

vikramdattu, 2022-09-19:
@priyanshu-makhiyaviya `-DTF_LITE_STATIC_MEMORY` this flag is legacy option, which is to chose static memory pool over runtime allocations. 
The option however, does not have any impact as of today. To be on the safer side (future changes could have use for the flag), I will add it in a way, it doesn't  break on some CMake version.

Can you please share the CMake version you are using?

priyanshu-makhiyaviya, 2022-09-19:
Hi @vikramdattu ,
I am using  CMake version 3.10.2.

Thank you.

closed 2023-03-31 by vikramdattu