espressif/esp-tflite-micro #25: Running custom classifier

william-hazem, 2022-11-11:
Hi Everyone!

I'm building my own classifier but I'm getting stuck at validation process using esp32 and esp-idf.

I want to make my images static but I don't now how convert then to a raw format as theses images [person_detection](https://github.com/espressif/tflite-micro-esp-examples/tree/master/examples/person_detection/static_images/sample_images). (eg. convert .jpg to raw data that can be embedded on esp flash).

vikramdattu, 2022-11-14:
Hi @william-hazem The code you are using is already optimized and is taking benefits of esp-nn under the hood.

What is the size of the model you're using? and what is the inference time you get? 
Please make sure to turn on QIO flash option and make sure the CPU is clocking at 240MHz if you're not.

william-hazem, 2022-11-15:
Hi @vikramdattu, thanks for answering.

My model has 14832 Bytes and tensors arena consumes around 155 kBytes. Currently i'm getting an inference between 4 and 5 secs. My model uses float data, and fews layers...

My esp is running with following:
* CPU 240MHz
* Flash Mode: QIO
* Flash SPI speed 40MHz
* ESP-NN: Optimized versions

vikramdattu, 2022-11-15:
@william-hazem can you please change the SPI speed to 80MHz and check? Also, can you please use internal memory for Arena if that's a possibility?

Optionally, via menuconfig, you may want to set Task Watchdog timeout to 5 seconds.

william-hazem, 2022-11-21:
Changing SPI speed  to 80MHz didn't improve speed. But I set  **compiler level to -O2 ("Performance")** and then It runned better, the inference time dropped and now is running each inference near to 2400 ms instead of ~4400 ms.

I'm actually using this operators, and convolution2D are being very expensive for inference time and using a lot of memory.
```java
static tflite::MicroMutableOpResolver<7> resolver;
resolver.AddFullyConnected();
resolver.AddQuantize();
resolver.AddDequantize();
resolver.AddAveragePool2D();
resolver.AddConv2D();
resolver.AddLogistic();
resolver.AddMean();
```
I think I need to make convolutional layers light, and keep compiling my idf project using -O2 flag to gain Performance.

vikramdattu, 2022-11-21:
Hi, @william-hazem thanks for sharing the experiment of using the optimization flag. 

Can you please analyze/share the breakup of the contribution of each layer type? You can find a reference here: https://github.com/espressif/tflite-micro-esp-examples/blob/master/examples/person_detection/main/main_functions.cc#L175 

Also, you may want to check the share of each layer.
Please note that ESP-NN optimizations are not as effective on ESP32 as on ESP32-S3 which has AI instructions.

SaketNer, 2023-12-31:
Hi, @william-hazem have you tried pruning the model and reducing the model size by quantisation to int8. In my case it helped improve the inference speed.

william-hazem, 2024-05-12:
I achieved better performance by adding some pooling layers to reduce the model size and changing the convolution kernel to reduce computational cost. The model was already quantized using int8, but I noticed a slight latency reduction when using it

Thanks by supporting @vikramdattu @SaketNer

closed 2024-05-12 by william-hazem