espressif/esp-tflite-micro #26: Cannot configure project person detection.

nikhil-robinson, 2022-11-21:
Hi there i am getting this output while opening the menuconfig.

{}[SDK Configuration Editor]
-- Found Git: /usr/bin/git (found version "2.37.1 (Apple Git-137.1)")

[SDK Configuration Editor]
-- Component directory /Users/bibox-ios/Desktop/tflite-micro-esp-examples/components/esp-nn does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/bibox-ios/Desktop/tflite-micro-esp-examples/components/esp32-camera does not contain a CMakeLists.txt file. No component will be added

[SDK Configuration Editor]
-- The C compiler identification is GNU 8.4.0

[SDK Configuration Editor]
-- The CXX compiler identification is GNU 8.4.0

[SDK Configuration Editor]
-- The ASM compiler identification is GNU

[SDK Configuration Editor]
-- Found assembler: /Users/bibox-ios/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc

[SDK Configuration Editor]
-- Detecting C compiler ABI info

[SDK Configuration Editor]
-- Detecting C compiler ABI info - done

[SDK Configuration Editor]
-- Check for working C compiler: /Users/bibox-ios/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info

[SDK Configuration Editor]
-- Detecting CXX compiler ABI info - done

[SDK Configuration Editor]
-- Check for working CXX compiler: /Users/bibox-ios/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ - skipped

[SDK Configuration Editor]
-- Detecting CXX compile features

[SDK Configuration Editor]
-- Detecting CXX compile features - done

[SDK Configuration Editor]
-- Building ESP-IDF components for target esp32s3

[SDK Configuration Editor]
---------------------------ERROR--------------------------

CMake Error at /Users/bibox-ios/esp/esp-idf/tools/cmake/build.cmake:201 (message):
Failed to resolve component 'esp-nn'.
Call Stack (most recent call first):
/Users/bibox-ios/esp/esp-idf/tools/cmake/build.cmake:232 (__build_resolve_and_add_req)
/Users/bibox-ios/esp/esp-idf/tools/cmake/build.cmake:233 (__build_expand_requirements)
/Users/bibox-ios/esp/esp-idf/tools/cmake/build.cmake:507 (__build_expand_requirements)
/Users/bibox-ios/esp/esp-idf/tools/cmake/project.cmake:384 (idf_build_process)
CMakeLists.txt:4 (project)

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
-- Configuring incomplete, errors occurred!
See also "/Users/bibox-ios/Desktop/tflite-micro-esp-examples/examples/person_detection/build/CMakeFiles/CMakeOutput.log".

[SDK Configuration Editor]
---------------------------ERROR--------------------------

cmake failed with exit code 1

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
Executing action: confserver
Running cmake in directory /Users/bibox-ios/Desktop/tflite-micro-esp-examples/examples/person_detection/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /Users/bibox-ios/Desktop/tflite-micro-esp-examples/examples/person_detection"...

[SDK Configuration Editor]
---------------------------ERROR--------------------------

SDK Configuration editor confserver process exited with code: 2
-----------------------END OF ERROR-----------------------

vikramdattu, 2022-11-29:
Hi, @nikhil-robinson please do a recursive submodule update in the cloned repo and try again.

`git submodule update --init --recursive`

vikramdattu, 2023-01-06:
@nikhil-robinson is your issue resolved now?

vikramdattu, 2026-11-09:
Closing, as there has been no activity here for a long time. esp-tflite-micro has changed a lot since (examples, ESP-IDF support and the upstream TFLite Micro sync). If you still see this with the latest release, please open a new issue with the ESP-IDF version, the board and the full log.

closed 2026-11-09 by vikramdattu