espressif/esp-tflite-micro #29: cmake Error building examples: cannot find common.cc

Tomp0801, 2023-01-19:
I think there is a mistake in the CMakeLists.txt of tflite-lib.
When I build the micro_speech example, this cmake error happens:

```cmake
CMake Error at C:/esp-idf/esp-idf-v4.4.3/tools/cmake/component.cmake:471 (add_library):
  Cannot find source file:

    F:/Repositories/tflite-micro-esp-examples/components/tflite-lib/tensorflow/lite/c/common.cc

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
  .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc
Call Stack (most recent call first):
  F:/Repositories/tflite-micro-esp-examples/components/tflite-lib/CMakeLists.txt:70 (idf_component_register)
```

common.cc does not exist at that location, however when I change line 52 in CMakeLists.txt of tflite-lib to `"${tflite_dir}/core/c/common.cc"` ( /core/c/ directory, instead of /c/ ), compilation works.

vikramdattu, 2023-01-20:
@Tomp0801 there have been changes in directory structure in latest `tflite-micro`. CMakeList.txt as a result need these changes.

I have pushed latest sync, this should solve the issue for all.

Tomp0801, 2023-01-25:
Building works now, thanks

closed 2023-01-25 by Tomp0801