espressif/esp-tflite-micro #31: internal compiler error in depthwiseconv_float.h when building examples

joeb42, 2023-03-01:
When building any of the examples I run into the same compiler error: `tflite-micro-esp-examples/components/tflite-lib/tensorflow/lite/kernels/internal/reference/depthwiseconv_float.h:95:1: error: insn does not satisfy its constraints:
   95 | }
      | ^
(insn 1399 40 396 19 (set (reg/v:SF 20 f1 [orig:66 total ] [66])
        (mem/u/c:SF (symbol_ref/u:SI ("*.LC27") [flags 0x2]) [0  S4 A32])) "tflite-micro-esp-examples/components/tflite-lib/tensorflow/lite/kernels/internal/reference/depthwiseconv_float.h":64:19 49 {movsf_internal}
     (expr_list:REG_EQUAL (const_double:SF 0.0 [0x0.0p+0])
        (nil)))
during RTL pass: postreload
tflite-micro-esp-examples/components/tflite-lib/tensorflow/lite/kernels/internal/reference/depthwiseconv_float.h:95:1: internal compiler error: in extract_constrain_insn, at recog.cc:2692
`.  This appears to be a very similar issue to those raised in https://github.com/tensorflow/tflite-micro/issues/634 and https://github.com/jcmvbkbc/gcc-xtensa/issues/14. The workaround for the aforementioned seems to be to compile at the O3 optimisation level, but since esp idf does not support O3 optimisation I am stuck. The error reproduces for me at both Os and O2 optimisation levels.

vikramdattu, 2023-03-02:
Hello @joeb42 can you please provide some more info on how to reproduce the issue:
1. With which example do you see this issue? 
2. Do you see the same in existing examples?

I have never encountered this issue. Some more info would be useful to debug this.

joeb42, 2023-03-02:
I'm building for esp32 chipset and first came across the issue when compiling the person detection, but persists for the other examples as well.

vikramdattu, 2023-03-02:
@joeb42 can you please also share the IDF version you are using?

joeb42, 2023-03-02:
`idf.py --version` yields `ESP-IDF v5.1-dev-3663-g8486a2c3ad`

vikramdattu, 2023-03-02:
@joeb42 thanks for the confirmation. Can you please use release branches and check the same? Examples are not tested on master branch, since it keeps evolving. I recommend you to please use the release branches.
You may chose to go with `release/v5.0`.

joeb42, 2023-03-03:
Thank you! Examples are compiling now, not sure how I managed to get a dev branch of idf

closed 2023-03-03 by vikramdattu