espressif/esp-tflite-micro #34: Error building person-detection

edge7, 2023-03-11:
Hi,
when trying to build I get:

```
resolver.cc: In function 'TfLiteStatus tflite::GetRegistrationFromOpCode(const tflite::OperatorCode*, const tflite::MicroOpResolver&, const TfLiteRegistration_V1**)':
/home/edge7/Desktop/projects/ing_edurso/blog/tflite_camera_article/esp32_tflite_person_detection/esp32_person_detection/components/tflite-lib/tensorflow/lite/micro/micro_op_resolver.cc:35:39: error: cannot convert 'const TfLiteRegistration*' to 'const TfLiteRegistration_V1*' in assignment
   35 |     *registration = op_resolver.FindOp(builtin_code);
      |                     ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
      |                                       |
      |                                       const TfLiteRegistration*
/home/edge7/Desktop/projects/ing_edurso/blog/tflite_camera_article/esp32_tflite_person_detection/esp32_person_detection/components/tflite-lib/tensorflow/lite/micro/micro_op_resolver.cc:46:39: error: cannot convert 'const TfLiteRegistration*' to 'const TfLiteRegistration_V1*' in assignment
   46 |     *registration = op_resolver.FindOp(name);
```

Can I ask which  tflite commit and esp-nn have you tested?

Thanks

vikramdattu, 2023-03-21:
Hello @edge7 it looks like there is some setup issues with your project. Are you able to build the existing examples? Existing person detection example should build and run just fine without any efforts.

vikramdattu, 2023-05-03:
Closing this as no activity on the issue and it's most likely a setup issue.

closed 2023-05-03 by vikramdattu