espressif/esp-tflite-micro #35: Micro Speech fails compiling with target esp32-s3

BitSalat, 2023-03-14:
Try to compile micro_speech example using Espressif IDE and IDF master. Compilation fails:

`C:/Users/xyz/Espressif/tflite-micro-esp-examples/components/tflite-lib/tensorflow/lite/kernels/internal/reference/depthwiseconv_float.h:95:1: error: insn does not satisfy its constraints:
   95 | }
      | ^
(insn 1399 40 396 19 (set (reg/v:SF 20 f1 [orig:66 total ] [66])
        (mem/u/c:SF (symbol_ref/u:SI ("*.LC27") [flags 0x2]) [0  S4 A32])) "C:/Users/xyz/Espressif/tflite-micro-esp-examples/components/tflite-lib/tensorflow/lite/kernels/internal/reference/depthwiseconv_float.h":64:19 49 {movsf_internal}
     (expr_list:REG_EQUAL (const_double:SF 0.0 [0x0.0p+0])
        (nil)))
during RTL pass: postreload
`

and:

`C:/Users/xyz/Espressif/tflite-micro-esp-examples/components/tflite-lib/tensorflow/lite/kernels/internal/reference/depthwiseconv_float.h:95:1: internal compiler error: in extract_constrain_insn, at recog.cc:2692
libbacktrace could not find executable to open
`

vikramdattu, 2023-03-14:
Please give it a try with stable IDF release branches.
`release/v5.0` or `release/v4.4` etc.

BitSalat, 2023-03-14:
You'r right, just compiled with idf 5.0 release and it worked smoothley. Thanx for your input!

BTW I am looking for the best way to train the micro speech tflite micro model for extending number of wake words.
Would you recommend colab using the TF notebooks or is there a more convenient way? (Edge Impulse isn't as it does not support Espressif esp32-S3)

AlexanderPaniutin, 2023-03-14:
Just faced the same problem with `ESP-IDF v5.1-dev-3972-g1559b6309f`
Moving it to `v5.0` tag resolves the problem.
Thanks!

vikramdattu, 2023-03-21:
> You'r right, just compiled with idf 5.0 release and it worked smoothley. Thanx for your input!
>
> BTW I am looking for the best way to train the micro speech tflite micro model for extending number of [...]

You should still be able to use edge impulse by getting the model trained over there and save non-compiled version. You can then use it with ESP32-S3.
There might be some work you will need as it is not supported right away but should not be much complicated I guess.

Found a great tutorial for your purpose: https://github.com/ShawnHymel/tflite-speech-recognition

You may also check for tf hub (https://tfhub.dev)

closed 2023-03-21 by vikramdattu