espressif/esp-tflite-micro #36: Can esp32 deploy models other than instances, such as some models recognized by OCR, and how to do it.

hqt567, 2023-03-17:
Can esp32 deploy models other than instances, such as some models recognized by OCR, and how to do it.

vikramdattu, 2023-03-20:
Hi @hqt567 can you please elaborate on the issue? You may refer specifically to the person detect example which does person detect task.
Other tasks will need a different model trained specifically to perform the task, converting which to tflite-micro format should run on ESP32.

hqt567, 2023-04-19:
Thank you for your reply @vikramdattu I'm a beginner in this field so I don't know how to train the model and how to deploy it to run on esp32. So I want to seek some help, is there any specific way to help me learn and complete these tasks.

closed 2026-11-09 by vikramdattu