espressif/esp-tflite-micro #39: ESP-IDF Build Project

gpsamt, 2023-04-18:
I'm using the examples https://github.com/espressif/tflite-micro-esp-examples/tree/master/examples and the 3 examples have the same error in Visual code with the extension of esp-idf.

C:/Users/USER/esp/esp-idf/tools/cmake/build.cmake:574 (__build_expand_requirements)
  C:/Users/USER/esp/esp-idf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:4 (project)

-- Configuring incomplete, errors occurred!
See also "D:/ProjetoVSC/tflite-micro-esp-examples-master/examples/person_detection/build/CMakeFiles/CMakeOutput.log".

 *  O processo de terminal "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B d:\ProjetoVSC\tflite-micro-esp-examples-master\examples\person_detection\build -S d:\ProjetoVSC\tflite-micro-esp-examples-master\examples\person_detection" was terminated with exit code: 1.

The error is the same in all 3 examples.

vikramdattu, 2023-04-18:
Can you please share some more logs? 
Please make sure if the `tflite-micro-esp-examples` clone you have done is recursive. If not you may simply run `git submodule update` under the cloned directory.

gpsamt, 2023-04-18:
Thanks a lot for the help. It worked out. I ran the following commands:

git clone --recurse-submodules https://github.com/espressif/tflite-micro-esp-examples.git

cd tflite-micro-esp-examples 

It worked. Now understand how you make the inference.

vikramdattu, 2023-04-19:
Closing this issue as resolved

closed 2023-04-19 by vikramdattu