espressif/esp-tflite-micro #46: micro speech model training

smhsn, 2023-05-25:
Hello @vikramdattu, 
I'd like to know which model architecture (repository) you used for training your yes/no model (model.cc). 
I'd like to customize the model for some other words. 

Thank you in advance.

vikramdattu, 2023-05-25:
Hi @smhsn thanks for the interest!

I have used model which is already trained by `google/tflite-micro` organization. You can find python notebook with steps how to train this particular model [here](https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/examples/micro_speech/train).

smhsn, 2023-05-25:
@vikramdattu thank you very much for the quick reply. 
I have used the same architecture to train a model but it fails to detect anything except unknown.
Also, in the main_functions.cc why there is a check for DepthwiseConv2D but not for Conv2D (the tiny_conv model from google/tflite-micro uses only Conv2D layer and not DepthwiseConv2D)?
Do we need to change any settings in the micro speech code to get it running for a new model?

`  static tflite::MicroMutableOpResolver<4> micro_op_resolver;
  if (micro_op_resolver.AddDepthwiseConv2D() != kTfLiteOk) {
    return;
  }
  if (micro_op_resolver.AddFullyConnected() != kTfLiteOk) {
    return;
  }
  if (micro_op_resolver.AddSoftmax() != kTfLiteOk) {
    return;
  }
  if (micro_op_resolver.AddReshape() != kTfLiteOk) {
    return;
  }`

vikramdattu, 2023-05-25:
@smhsn 

> I have used the same architecture to train a model but it fails to detect anything except unknown.

Can you please make sure the i2s stream you record is proper?

>  In the main_functions.cc why there is a check for DepthwiseConv2D but not for Conv2D (the tiny_conv model from `google/tflite-micro` uses only Conv2D layer and not DepthwiseConv2D)?
Do we need to change any settings in the micro speech code to get it running for a new model?

The existing model is using those 4 OPs (including `DepthwiseConv2D`) and hence those checks. If the newly trained model is using `Conv2D` please make that change on those lines!

It would look like:

```
if (micro_op_resolver.AddConv2D() != kTfLiteOk) {
    return;
  }
```

smhsn, 2023-05-25:
Thank you very much, It is working now after I replaced DepthwiseConv2D with Conv2D, I appreciate your help. 
By the way I could not find the existing model architecture which uses DepthwiseConv2D in the google/tflite-micro repo. What is the model called?

closed 2023-06-13 by vikramdattu