espressif/esp-tflite-micro #48: Failed to load the protocol_examples_common.h in micro_speech project

techlearnerr, 2023-05-29:
I have imported protocol_examples_common.h header and getting the following error

```
micro_speech/main/command_responder.cc:23:10:←[m←[K ←[01;31m←[Kfatal error: ←[m←[Kprotocol_examples_common.h: No such file or directory
   23 | #include ←[01;31m←[K"protocol_examples_common.h"←[m←[K
      |          ←[01;31m←[K^~~~~~~~~~~~~~~~~~~~~~~~~~~~←[m←[K
compilation terminated
```

I have changed the CMakeLists.text file to the following and still get the same error
```
cmake_minimum_required(VERSION 3.16)
set(EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/examples/common_components/protocol_examples_common  ../../components/)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(micro_speech)
```

Target platform: ESP32-S3
IDF version release/v5.0

vikramdattu, 2023-05-30:
Can you please also add `protocol_examples_common` to the `PRIV_REQUIRES` [here](https://github.com/espressif/tflite-micro-esp-examples/blob/65db24a989941489300578bef9de86d53f3fa764/examples/micro_speech/main/CMakeLists.txt#LL11C35-L11C35) and check again?

techlearnerr, 2023-05-30:
Thanks for your response. 
I add the protocol_examples_common to the PRIV_REQUIRES and get the following error.

```
-- Building ESP-IDF components for target esp32s3
Processing 1 dependencies:
[1/1] idf (5.0.1)
CMake Error at D:/.espressif/frameworks/esp-idf-v5.0.1/tools/cmake/build.cmake:245 (message):
  Failed to resolve component 'protocol_examples_common'.
Call Stack (most recent call first):
  D:/.espressif/frameworks/esp-idf-v5.0.1/tools/cmake/build.cmake:287 (__build_resolve_and_add_req)
  D:/.espressif/frameworks/esp-idf-v5.0.1/tools/cmake/build.cmake:574 (__build_expand_requirements)
  D:/.espressif/frameworks/esp-idf-v5.0.1/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:5 (project)

-- Configuring incomplete, errors occurred!
See also "D:/.espressif/frameworks/11/examples/micro_speech/build/CMakeFiles/CMakeOutput.log".
←[31mFAILED: ←[0mbuild.ninja
D:\.espressif\tools\cmake\3.24.0\bin\cmake.exe --regenerate-during-build -SD:\.espressif\frameworks\11\examples\micro_speech -BD:\.espressif\frameworks\11\examples\micro_speech\build
ninja: error: rebuilding 'build.ninja': subcommand failed
←[0;33mHINT: The component protocol_examples_common could not be found. This could be because: component name was misspelled, the component was not added to the build, the component has been moved to the IDF component manager or has been removed and refactored into some other component.
Please look out for component in 'https://components.espressif.com' and add using 'idf.py add-dependency' command.
Refer to the migration guide for more details about moved components.
Refer to the build-system guide for more details about how components are found and included in the build.←[0m
ninja failed with exit code 1, output of the command is in the d:\.espressif\frameworks\11\examples\micro_speech\build\log\idf_py_stderr_output_80880 and d:\.espressif\frameworks\11\examples\micro_speech\build\log\idf_py_stdout_output_80880
```

vikramdattu, 2026-11-09:
Closing, as there has been no activity here for a long time. esp-tflite-micro has changed a lot since (examples, ESP-IDF support and the upstream TFLite Micro sync). If you still see this with the latest release, please open a new issue with the ESP-IDF version, the board and the full log.

closed 2026-11-09 by vikramdattu