espressif/esp-tflite-micro #51: hello_world with ESP-BOX-LITE (esp32-s3)

ozanoner, 2023-06-13:
The _hello_world_ example doesn't seem working on [ESP-BOX-LITE (esp32-s3)](https://github.com/espressif/esp-box/blob/master/docs/hardware_overview/esp32_s3_box_lite/hardware_overview_for_lite.md) as supposed. the result of the inference is always the same for all inputs. The example is a bit modified to show the **y_quantized** value

y_value: 1.0420598*2^0
y_quantized: 127

```
$ idf.py --version
ESP-IDF v4.4.4-1-g6727a4f775

```

```

--- idf_monitor on /dev/ttyACM0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048df9
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x17cc
load:0x403c9700,len:0xe78
load:0x403cc700,len:0x310c
entry 0x403c9980
I (24) boot: ESP-IDF v4.4.4-1-g6727a4f775 2nd stage bootloader
I (25) boot: compile time 00:55:36
I (25) boot: chip revision: v0.1
I (28) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (35) qio_mode: Enabling default flash chip QIO
I (40) boot.esp32s3: Boot SPI Speed : 80MHz
I (45) boot.esp32s3: SPI Mode       : QIO
I (49) boot.esp32s3: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0b030h ( 45104) map
I (120) esp_image: segment 1: paddr=0001b058 vaddr=3fc905a0 size=0280ch ( 10252) load
I (123) esp_image: segment 2: paddr=0001d86c vaddr=40374000 size=027ach ( 10156) load
I (132) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1ba5ch (113244) map
I (155) esp_image: segment 4: paddr=0003ba84 vaddr=403767ac size=09df0h ( 40432) load
I (168) boot: Loaded app from partition at offset 0x10000
I (169) boot: Disabling RNG early entropy source...
I (180) cpu_start: Pro cpu up.
I (180) cpu_start: Starting app cpu, entry point is 0x40375160
0x40375160: call_start_cpu1 at /home/esp32b2/esp/esp-idf/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (194) cpu_start: Pro cpu start user code
I (194) cpu_start: cpu freq: 160000000
I (194) cpu_start: Application information:
I (197) cpu_start: Project name:     hello_world
I (202) cpu_start: App version:      5f3c906
I (207) cpu_start: Compile time:     Jun 13 2023 01:02:05
I (213) cpu_start: ELF file SHA256:  0583855cdc69b5fe...
I (219) cpu_start: ESP-IDF:          v4.4.4-1-g6727a4f775
I (225) heap_init: Initializing. RAM available for dynamic allocation:
I (232) heap_init: At 3FC940A8 len 00055668 (341 KiB): D/IRAM
I (239) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (245) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (252) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (258) spi_flash: detected chip: gd
I (262) spi_flash: flash io: qio
W (266) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (280) sleep: Configure to isolate all GPIO pins in sleep state
I (286) sleep: Enable automatic switching of GPIO sleep configuration
I (293) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
y_quantized: 127

x_value: 1.0*2^-127, y_value: 1.0420598*2^0

y_quantized: 127

x_value: 1.2566366*2^-2, y_value: 1.0420598*2^0

y_quantized: 127

x_value: 1.2566366*2^-1, y_value: 1.0420598*2^0

y_quantized: 127

x_value: 1.8849551*2^-1, y_value: 1.0420598*2^0

y_quantized: 127

x_value: 1.2566366*2^0, y_value: 1.0420598*2^0
```

vikramdattu, 2023-06-13:
Hi @ozanoner does disabling optimisations work as expected for you? If it does, please apply a patch to esp-nn mentioned [here](https://github.com/espressif/esp-nn/issues/5#issuecomment-1589070024) and share you observation if the issue is fixed.

ozanoner, 2023-06-13:
Thank you! I confirm that the example works with NN optimisations disabled ( (Top) → ESP-NN → Optimization for nn functions )  and [fix_mismatch.patch](https://github.com/espressif/esp-nn/files/11734010/fix_mismatch.patch) applied.

closed 2023-06-13 by ozanoner