espressif/esp-tflite-micro #54: Watchdog triggered when testing 'micro_speech' sample on ESP-IDF v5.1

misterb0407, 2023-06-15:
Steps:
1. cd to root of project files (not examples/micro_speech) so that 'component folder is recognized by system
2. Build the sample under 'micro_speech' examples by using docker ESP-IDF version 5.2
`$docker run --rm -v $PWD:/project -w /project -it espressif/idf:release-v5.1`
Then inside the docker prompt:
`$ idf.py build`

2. Then flash and monitor, you would see the following in the log:
`E (434939) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (434939) task_wdt:  - IDLE (CPU 0)
E (434939) task_wdt: Tasks currently running:
E (434939) task_wdt: CPU 0: tensorflow
E (434939) task_wdt: CPU 1: IDLE
E (434939) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x400E614F:0x3FFB0D50 0x40082905:0x3FFB0D80 0x400D6122:0x3FFC0200 0x400D5D8D:0x3FFC0240 0x400D5916:0x3FFC0270

Heard unknown (205) @434600ms
Heard unknown (205) @438300ms
Heard unknown (224) @439900ms
Heard unknown (213) @441500ms
Heard unknown (232) @443100ms
Heard unknown (215) @444700ms
Heard unknown (203) @448100ms`

This was not observed when building the same code with ESP-IDF version `release-v4.4`

vikramdattu, 2023-06-15:
@misterb0407 is the target `esp32` in both the setups?

misterb0407, 2023-06-15:
Hi @vikramdattu , yes , the target was esp32 for both setup. As a work around I just disable watchdog for now from the `idf.py menuconfig`

vikramdattu, 2023-06-15:
@misterb0407 okay, thanks! I will certainly check for newer IDF branch and update you on the same.

vikramdattu, 2023-06-15:
@misterb0407 I saw your issue on IDF: https://github.com/espressif/esp-idf/issues/11678 
Does having that particular change for I2S setup help with this example?

misterb0407, 2023-06-15:
Hi @vikramdattu , it is hard to say as the example here uses legacy I2S driver but the https://github.com/espressif/esp-idf/issues/11678 uses the new I2S driver. Do you have any config (in this legacy driver) you would like me to try?

vikramdattu, 2026-11-09:
Closing, as there has been no activity here for a long time. esp-tflite-micro has changed a lot since (examples, ESP-IDF support and the upstream TFLite Micro sync). If you still see this with the latest release, please open a new issue with the ESP-IDF version, the board and the full log.

closed 2026-11-09 by vikramdattu