espressif/esp-tflite-micro #55: Example 'micro_speech' hung when I2S config bits_per_sample changed to 24 from 16

misterb0407, 2023-06-16:
Since I am using ESP-EYE which uses MSM261S4030H0R microphone, and according to datasheets it has 24-bit depth in a 32 bit wordk. So in `i2s_init()` I change the : 

```
i2s_config_t i2s_config = {
    ...
    .bits_per_sample = (i2s_bits_per_sample_t)24, // was 16
}
```
But when loading it to the board, the console remain like this forever, regardless we speak over phone or not:

```
I (336) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (430) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (430) I2S: I2S1, MCLK output by GPIO0
I (490) TF_LITE_AUDIO_PROVIDER: Audio Recording started
```

By right we should expect print out to reflect "yes" or "no" being detected.

vikramdattu, 2023-06-19:
@misterb0407 from the data sheet: `The serial data is in slave mode I2S format, which has 24‐bit depth in a 32 bit word`.
That means, you need to set `bits_per_sample` to 32 bits and ignore MSB 8 bits.
I could find an example retrieving data in this format [here](https://blog.squix.org/2019/08/esp32-esp-eye-browser-based-spectrum-analyzer.html).

But since, the feature extractor from the example expects the data in 16 bits, is this something you want to do?

misterb0407, 2023-06-23:
Hi @vikramdattu , my main objective is to get this micro_example works on my ESP-EYE board with 'decent' accuracy, Currently it is just so poor that I don't believe it is because of the model alone. 

Anyway thanks for the link, I tried similar setting, but no improvement observed.

vikramdattu, 2026-11-09:
Closing, as there has been no activity here for a long time. esp-tflite-micro has changed a lot since (examples, ESP-IDF support and the upstream TFLite Micro sync). If you still see this with the latest release, please open a new issue with the ESP-IDF version, the board and the full log.

closed 2026-11-09 by vikramdattu