espressif/esp-tflite-micro #60: 模型替换之后无法被运行

cannnnnnnnnnnn, 2023-08-09:
编译hello_world例程运行都没有问题，但是一旦将模型数据替换为[官方例程](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/examples/hello_world/train.py)的模型数据就导致输出的数据全为零
请问一下能不能将模型训练的脚本文件添加至项目文件夹中

cannnnnnnnnnnn, 2023-08-09:
[image]

vikramdattu, 2023-08-09:
Hello @cannnnnnnnnnnn the model which I have embedded is a model which was present in tflite-micro repo since the beginning of the development, (which was previously part of TensorFlow/) repo.

How did you quantise the model? The script above shares steps till generation of model in float32 format. The example assumes the model is `int8` quantised and hence input and output to it is int8 converted back to float.
If you didn't convert the float32 model to int8, do consider providing input and interpreting the output as float32 and not int8.  You may find the script [here.](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/examples/hello_world/quantization/ptq.py) Hope this helps.

cannnnnnnnnnnn, 2023-08-10:
感谢您的解答，我根据您给的解决思路已经将模型成功在设备上跑通了
[image]

closed 2023-08-10 by cannnnnnnnnnnn