espressif/esp-tflite-micro #64: Use on custom ESP-IDF project

Steven35700, 2023-09-26:
I would like to integrate TFLite micro into my ESP-IDF project. I managed to get the examples to work.
I started to clone this directory and include it in my own project by adding it on my idf_component.yml.
But I have some problems of versionning when I build : 

```
Executing "ninja all"...
[0/1] Re-running CMake...-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32s3
CMake Error at D:/Espressif_5_1/frameworks/esp-idf-v5.1/tools/cmake/build.cmake:540 (message):
  Traceback (most recent call last):

    File "<frozen runpy>", line 198, in _run_module_as_main
    File "<frozen runpy>", line 88, in _run_code
    File "D:\Espressif_5_1\python_env\idf5.1_py3.11_env\Lib\site-packages\idf_component_manager\prepare_components\__main__.py", line 6, in <module>
      main()
    File "D:\Espressif_5_1\python_env\idf5.1_py3.11_env\Lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 124, in main
      args.func(args)
    File "D:\Espressif_5_1\python_env\idf5.1_py3.11_env\Lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 31, in prepare_dep_dirs
      ).prepare_dep_dirs(
        ^^^^^^^^^^^^^^^^^
    File "D:\Espressif_5_1\python_env\idf5.1_py3.11_env\Lib\site-packages\idf_component_manager\core.py", line 67, in wrapper
      return func(self, *args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "D:\Espressif_5_1\python_env\idf5.1_py3.11_env\Lib\site-packages\idf_component_manager\core.py", line 529, in prepare_dep_dirs
      downloaded_component_paths, downloaded_component_version_dict = download_project_dependencies(
                                                                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "D:\Espressif_5_1\python_env\idf5.1_py3.11_env\Lib\site-packages\idf_component_manager\dependencies.py", line 146, in download_project_dependencies
      if is_solve_required(project_requirements, solution):
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "D:\Espressif_5_1\python_env\idf5.1_py3.11_env\Lib\site-packages\idf_component_manager\dependencies.py", line 99, in is_solve_required
      component.name, spec='=={}'.format(component.version.semver))
                                         ^^^^^^^^^^^^^^^^^^^^^^^^
    File "D:\Espressif_5_1\python_env\idf5.1_py3.11_env\Lib\site-packages\idf_component_tools\manifest\manifest.py", line 275, in semver
      raise TypeError('Version is not semantic')

  TypeError: Version is not semantic

Call Stack (most recent call first):
  D:/Espressif_5_1/frameworks/esp-idf-v5.1/tools/cmake/project.cmake:464 (idf_build_process)
  CMakeLists.txt:22 (project)
```

I would like to know if this bug is known and if I can easily solve it. 
Also, I was wondering if it was possible to directly use the tensorflow lite for microcontroller github directory and integrate it into my project ? And would the esp-NN functions will be used anyway ?

vikramdattu, 2023-09-27:
@Steven35700 I have not seen this error. The error is from IDF component manager and it is failing to resolve dependencies. I would suggest you to update component manager, remove `dependencies.lock` file which got generated in the example directory and see if the error goes away.

```
rm -rf dependencies.lock build/
python -m pip --upgrade idf-component-manager
idf.py build
```

> Also, I was wondering if it was possible to directly use the tensorflow lite for microcontroller github directory and integrate it into my project ? And would the esp-NN functions will be used anyway ?
Yes, it is possible and esp-nn will be pulled transparently via component manager as this is specified in dependencies. (https://github.com/espressif/tflite-micro-esp-examples/blob/2c89a70157c974491d33f19ed94254ecc5e58df8/components/esp-tflite-micro/idf_component.yml#L8-L10)

GeoffroyGit, 2023-12-01:
Hi @Steven35700 
I had the same issue after I do a full clean and try to build.
Here is how I solved it (this is inspired by @vikramdattu 's answer):
```
$ pip freeze | grep idf-component-manager
idf-component-manager==1.3.2
$ pip install --upgrade pip
$ pip install --upgrade idf-component-manager
$ pip freeze | grep idf-component-manager
idf-component-manager==1.4.1
```
So I had the issue when I had `idf-component-manager` version 1.3.2 but now that I have version 1.4.1 the issue is solved.

closed 2026-11-09 by vikramdattu