espressif/esp-tflite-micro #65: tensorflow/lite/array.h: No such file or directory

MATTYGILO, 2023-10-05:
https://github.com/tensorflow/tflite-micro/issues/2078

I am getting this error:

tflm/tensorflow/lite/kernels/kernel_util.cpp:28:10: fatal error: tensorflow/lite/array.h: No such file or directory
 #include "tensorflow/lite/array.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

and I know it is because TF_LITE_STATIC_MEMORY seem not to be being defined but here's my cmake file:

include (${CMAKE_CURRENT_LIST_DIR}/micropython_esp.cmake)

get_filename_component(TENSORFLOW_DIR ${PROJECT_DIR}/../../../tensorflow ABSOLUTE)

add_library(microlite INTERFACE)

set(TF_ESP_DIR "${CMAKE_CURRENT_LIST_DIR}/tflm")
set(TF_LITE_DIR "${TF_ESP_DIR}/tensorflow/lite")
set(TF_MICRO_DIR "${TF_LITE_DIR}/micro")
set(TF_MICROLITE_LOG
        ${TF_MICRO_DIR}/debug_log.cpp
        ${TF_MICRO_DIR}/micro_time.cpp
        )

# lite c

file(GLOB TF_LITE_C_SRCS
        "${TF_LITE_DIR}/c/*.cpp"
        "${TF_LITE_DIR}/c/*.c")

# lite core/api
file(GLOB TF_LITE_API_SRCS
        "${TF_LITE_DIR}/core/api/*.cc"
        "${TF_LITE_DIR}/core/api/*.c"
        "${TF_LITE_DIR}/core/c/*.cc"
        "${TF_LITE_DIR}/core/c/*.c")

# lite experimental
file(GLOB TF_LITE_MICROFRONTEND_SRCS
        "${TF_LITE_DIR}/experimental/microfrontend/lib/*.c"
        "${TF_LITE_DIR}/experimental/microfrontend/lib/*.cc"
        "${TF_LITE_DIR}/experimental/microfrontend/lib/*.cpp")

# lite kernels
file(GLOB TF_LITE_KERNELS_SRCS
        "${TF_LITE_DIR}/kernels/*.c"
        "${TF_LITE_DIR}/kernels/*.cpp"
        "${TF_LITE_DIR}/kernels/*.cc"
        "${TF_LITE_DIR}/kernels/internal/*.c"
        "${TF_LITE_DIR}/kernels/internal/*.cpp"
        "${TF_LITE_DIR}/kernels/internal/*.cc"
        "${TF_LITE_DIR}/kernels/internal/reference/*.c"
        "${TF_LITE_DIR}/kernels/internal/reference/*.cpp"
        "${TF_LITE_DIR}/kernels/internal/reference/*.cc"
        )

# lite schema
file(GLOB TF_LITE_SCHEMA_SRCS
        "${TF_LITE_DIR}/schema/*.c"
        "${TF_LITE_DIR}/schema/*.cc"
        "${TF_LITE_DIR}/schema/*.cpp")

# micro

file(GLOB TF_MICRO_SRCS
        "${TF_MICRO_DIR}/*.c"
        "${TF_MICRO_DIR}/*.cc"
        "${TF_MICRO_DIR}/*.cpp")

# logs are platform specific and added seperately

#list(REMOVE_ITEM TF_MICRO_SRCS ${TF_MICRO_DIR}/debug_log.cpp)
#list(REMOVE_ITEM TF_MICRO_SRCS ${TF_MICRO_DIR}/micro_time.cpp)

# arena allocator
file(GLOB TF_MICRO_ARENA_ALLOCATOR_SRCS
        "${TF_MICRO_DIR}/arena_allocator/*.cpp"
        "${TF_MICRO_DIR}/arena_allocator/*.cc"
        "${TF_MICRO_DIR}/arena_allocator/*.c")

# micro kernels

file(GLOB TF_MICRO_KERNELS_SRCS
        "${TF_MICRO_DIR}/kernels/*.c"
        "${TF_MICRO_DIR}/kernels/*.cc"
        "${TF_MICRO_DIR}/kernels/*.cpp")

# micro memory_planner

file(GLOB TF_MICRO_MEMORY_PLANNER_SRCS
        "${TF_MICRO_DIR}/memory_planner/*.cpp"
        "${TF_MICRO_DIR}/memory_planner/*.cc"
        "${TF_MICRO_DIR}/memory_planner/*.c")

# tflite_bridge

file(GLOB TF_MICRO_TFLITE_BRIDGE_SRCS
        "${TF_MICRO_DIR}/tflite_bridge/*.cpp"
        "${TF_MICRO_DIR}/tflite_bridge/*.cc"
        "${TF_MICRO_DIR}/tflite_bridge/*.c")

set (BOARD_ADDITIONAL_SRCS "")

set(tfmicro_kernels_dir ${TF_MICRO_DIR}/kernels)

# remove sources which will be provided by esp_nn
list(REMOVE_ITEM TF_MICRO_KERNELS_SRCS
        "${tfmicro_kernels_dir}/add.cc"
        "${tfmicro_kernels_dir}/conv.cc"
        "${tfmicro_kernels_dir}/depthwise_conv.cc"
        "${tfmicro_kernels_dir}/fully_connected.cc"
        "${tfmicro_kernels_dir}/mul.cc"
        "${tfmicro_kernels_dir}/pooling.cc"
        "${tfmicro_kernels_dir}/softmax.cc"
        )

# tflm wrappers for ESP_NN
FILE(GLOB ESP_NN_WRAPPERS
        "${tfmicro_kernels_dir}/esp_nn/*.cc")

#   microlite micropython module sources
set (MICROLITE_PYTHON_SRCS
        ${CMAKE_CURRENT_LIST_DIR}/tensorflow-microlite.c
        ${CMAKE_CURRENT_LIST_DIR}/micropython-error-reporter.cpp
        )

list(APPEND MICROLITE_PYTHON_SRCS
        ${CMAKE_CURRENT_LIST_DIR}/openmv-libtf.cpp
        )

target_sources(microlite INTERFACE
        # micro_python sources for tflite
        ${MICROLITE_PYTHON_SRCS}

        # tf lite sources
        ${TF_LITE_C_SRCS}
        ${TF_LITE_API_SRCS}
        ${TF_LITE_MICROFRONTEND_SRCS}
        ${TF_LITE_KERNELS_SRCS}
        ${TF_LITE_SCHEMA_SRCS}

        # tf micro sources
        ${TF_MICRO_SRCS}
        ${TF_MICRO_ARENA_ALLOCATOR_SRCS}
        ${TF_MICRO_KERNELS_SRCS}
        ${TF_MICRO_MEMORY_PLANNER_SRCS}
        ${TF_MICRO_TFLITE_BRIDGE_SRCS}

        ${TF_MICROLITE_LOG}
        ${ESP_NN_SRCS} # include esp-nn sources for Espressif chipsets
        ${ESP_NN_WRAPPERS} # add tflm wrappers for ESP_NN
        )

target_sources(microlite INTERFACE
        ${CMAKE_CURRENT_LIST_DIR}/python_ops_resolver.cc
        )

target_include_directories(microlite INTERFACE
        ${TF_ESP_DIR}
        ${TF_ESP_DIR}/third_party/kissfft
        ${TF_ESP_DIR}/third_party/kissfft/tools
        ${TF_ESP_DIR}/third_party/flatbuffers/include
        ${TF_ESP_DIR}/third_party/gemmlowp
        ${TF_ESP_DIR}/third_party/ruy
        ${ESP_NN_INC})

target_compile_definitions(microlite INTERFACE
        MODULE_MICROLITE_ENABLED=1
        TF_LITE_STATIC_MEMORY=1
        TF_LITE_MCU_DEBUG_LOG
        NDEBUG)
target_compile_definitions(microlite INTERFACE
        ESP_NN=1 # enables esp_nn optimizations if those sources are added
        CONFIG_NN_OPTIMIZED=1 # use Optimized vs ansi code from ESP-NN
        )

target_compile_options(microlite INTERFACE
        # Required for ESP32-S3
        -mlongcalls
        -fno-unroll-loops
        -Wno-unused-function

        -Wno-error
        -Wno-error=float-conversion
        -Wno-error=nonnull
        -Wno-error=double-promotion
        -Wno-error=pointer-arith
        -Wno-error=unused-const-variable
        -Wno-error=sign-compare
        -fno-rtti
        -fno-exceptions
        -O3
        -Wno-error=maybe-uninitialized)

target_link_libraries(usermod INTERFACE microlite)
Which has TF_LITE_STATIC_MEMORY defined here:

target_compile_definitions(microlite INTERFACE
        MODULE_MICROLITE_ENABLED=1
        TF_LITE_STATIC_MEMORY=1
        TF_LITE_MCU_DEBUG_LOG
        NDEBUG)
Why then when I am building does it think it is not defined?

vikramdattu, 2023-10-06:
@MATTYGILO The reason is, micro_python has pre-processing stage and it doesn't include the flags set from the cmake files you mentioned here.

I have already have a fix for this. I hope, you already have this in your code base: https://github.com/mocleiri/tensorflow-micropython-examples/commit/127ff3d183f019f5802a1e02c6f22be072b2addc

MATTYGILO, 2023-10-06:
@vikramdattu Thanks so much that solved it. I am so appreciative of your help!!

closed 2023-10-06 by MATTYGILO