espressif/esp-tflite-micro #67: Inconsistent Model Predictions

MATTYGILO, 2023-10-06:
@vikramdattu Thank you helping me earlier, I started modifying the firmware as I wanted to be able to fix an error I am receiving.

### Repos in use
https://github.com/espressif/tflite-micro-esp-examples
https://github.com/espressif/esp-nn

### Micropython firmware based off
https://github.com/mocleiri/tensorflow-micropython-examples

### Summarised
I have a model which is made up of simple inverted res net layers with no recurrence or memory. It is binary classification with sigmoid activation. However I am getting inconsistent predictions. For example:

```
0.5960784
0.454902
0.4470588
0.4470588
0.4470588
0.4470588
```

As you can see the predictions start at a value then change and then stabilise. They are using exactly the same data (Absolutely Certain). I have tried many other tflite models and the same problem occurs.

I have done tests where I have loaded two of the same models at the same time and they both follow the same pattern above. When I reset the device and start the predictions again they follow the exact same pattern.

### Temporary Fix (Don't understand why):
I have found by wiping the header of the arena and reinitialising the model I get consistent predictions. However, to wipe the arena and reinitialise it is too slow for me. 

I was wondering whether you had any thoughts on what could be the root cause of the problem?

#### Gut feeling:
I think that something is going wrong in the header of the arena, where input/Intermediate/output tensors are being mixed up. Feels like some recursion is going on from invocation to invocation where data is being mixed up in the header and that is why it stabilises when it finds data that leads to the same data x-> x

#### Note:
I have have about 3 pages of in depth trial and error logs that I can provide if you need more info

All help would be very appreciated I’ve been stuck on this for about 5 days.

vikramdattu, 2023-10-06:
@MATTYGILO 

1. What happens if you just re-populate the input? I guess the input Tensor gets re-used.
2. Is it possible for me to try the setup? I didn't looked into the possibility of running https://github.com/mocleiri/tensorflow-micropython-examples on host machine. Is it possible to do so? I am interested to check the result on host platform as well. Also, it will be easier to debug the issue, if it can be re-produced on host.

MATTYGILO, 2023-10-06:
@vikramdattu My firmware isn't in a clean enough state as it has many other things implemented, I think that [https://github.com/mocleiri/tensorflow-micropython-examples](https://github.com/mocleiri/tensorflow-micropython-examples) may have the same problem. Is it possible to use their firmware for your testing?

1. In my testing I am re populating the input tensor each time and it still causes the same bug
2. Currently using an esp32s3, sadly I don't know how to run on host

If there is anything else I can do to aid you do say

MATTYGILO, 2023-10-06:
@vikramdattu If I cleaned up [tensorflow-micropython-examples](https://github.com/mocleiri/tensorflow-micropython-examples), tailored it to specifically esp and made it easier for people to use. Would you be happy to help contribute/maintain?

MATTYGILO, 2023-10-07:
So I modified my firmware so that it would output these values
```
// Output whether we are using tflite static, TF_LITE_STATIC_MEMORY
#ifdef TF_LITE_STATIC_MEMORY
mp_printf(MP_PYTHON_PRINTER, "Using TF_LITE_STATIC_MEMORY\n");
#endif

// Output whether we are using esp nn
#ifdef ESP_NN
mp_printf(MP_PYTHON_PRINTER, "Using ESP_NN\n");
#endif

// Output whether the neural network is optimised
#ifdef CONFIG_NN_OPTIMIZED
mp_printf(MP_PYTHON_PRINTER, "Using CONFIG_NN_OPTIMIZED\n");
#endif
```
I am getting this output:

```
Using TF_LITE_STATIC_MEMORY
Using ESP_NN
Using CONFIG_NN_OPTIMIZED
```

MATTYGILO, 2023-10-07:
@vikramdattu Quick question are these model size limits in tflite micro? Or could there be any problems with using a large model to do with the arena?

vikramdattu, 2023-10-07:
@MATTYGILO in theory there isn't any limit on model size of tflite.
It would only be limited by memory availability of the chipset. But again, that should not be the reason for erroneous outputs you see.

MATTYGILO, 2023-10-07:
@vikramdattu Thanks for the clarification, just trying to think of what the problem may be.

MATTYGILO, 2023-10-09:
I fixed my problem it was an incredibly silly error seeming the depths I went to fix it. Turned out I was entering input index data incorrectly and out of range of my specific model.

Anyway whilst doing this debugging I released a version of tflite micro for esp device specifically micropython. Hope it is of use to everyone, I will continue to improve it and instructions etc: [https://github.com/MATTYGILO/tflite-micropython-esp](https://github.com/MATTYGILO/tflite-micropython-esp)

MATTYGILO, 2023-10-09:
@vikramdattu It would be great for some feedback/contributions for this repo if you can. [https://github.com/MATTYGILO/tflite-micropython-esp](https://github.com/MATTYGILO/tflite-micropython-esp)

Many Thanks
Matt

closed 2023-10-09 by MATTYGILO

vikramdattu, 2023-10-09:
Hi, @MATTYGILO good to hear that you fixed the issue.

I sure will contribute to the repo wherever I can. Thanks for the great repo. 👍