espressif/esp-tflite-micro #69: Error when bytes_written in 'micro_speechs' example

adityale1711, 2023-11-01:
I'm trying to run micro_speech example but I have trouble with this code:
`int bytes_written = rb_write(g_audio_capture_buffer, (uint8_t *)g_i2s_read_buffer, bytes_read, pdMS_TO_TICKS(100));`

when I build and flash to ESP32-S3 I get this error message:

> Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
> 
> Core  0 register dump:
> PC      : 0x40056f60  PS      : 0x00060430  A0      : 0x8200a035  A1      : 0x3fcdc2f0
> A2      : 0x00000000  A3      : 0x3fcad9f0  A4      : 0x0000c800  A5      : 0x00000000
> A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3fcdc2a0
> A10     : 0x3fc94018  A11     : 0x3fc94018  A12     : 0x00060420  A13     : 0x00060423
> A14     : 0x00000001  A15     : 0x0000cdcd  SAR     : 0x00000002  EXCCAUSE: 0x0000001d
> EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000c7f
> 
> 
> Backtrace: 0x40056f5d:0x3fcdc2f0 |<-CORRUPTED
> 
> 
> 
> 
> ELF file SHA256: 55ecc57c2662042f
> 
> Rebooting...

I have increased/decreased the value of variable kAudioCaptureBufferSize and i2s_bytes_to_read but I still got that error

vikramdattu, 2023-11-01:
Hi @adityale1711 I have checked with esp-eye (esp32 based board) and esp-s3-eye (esp32-s3 based) boards and IDF `release/v5.1` and it is working flawless.
1. Did you have any custom changes, which might be preventing `InitAudioRecording` to not happen?
2. Which is the dev board and IDF version you're working with? 
3. Could you please share complete logs, right from reboot of the device?

adityale1711, 2023-11-01:
1. I skipped the GetAudioSamples function in future_provider.cc and only used the GetAudioSamples1 function, should I have to use both?
2. I'm using DevKit-C and IDF v5.0
3. Is this a complete log you need :

> ESP-ROM:esp32s3-20210327
> Build:Mar 27 2021
> rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fce3810,len:0x165c
> load:0x403c9700,len:0xbe0
> load:0x403cc700,len:0x2d9c
> entry 0x403c9900
> I (25) boot: ESP-IDF v5.0.2-dirty 2nd stage bootloader
> I (25) boot: compile time 14:16:06
> I (25) boot: chip revision: v0.1
> I (27) boot.esp32s3: Boot SPI Speed : 80MHz
> I (32) boot.esp32s3: SPI Mode       : DIO
> I (37) boot.esp32s3: SPI Flash Size : 16MB
> I (41) boot: Enabling RNG early entropy source...
> I (47) boot: Partition Table:
> I (50) boot: ## Label            Usage          Type ST Offset   Length
> I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
> I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
> I (73) boot:  2 factory          factory app      00 00 00010000 00200000
> I (80) boot:  3 storage          Unknown data     01 82 00210000 00100000
> I (88) boot: End of partition table
> I (92) esp_image: segment 0: paddr=00010020 vaddr=3c070020 size=d69e0h (879072) map
> I (258) esp_image: segment 1: paddr=000e6a08 vaddr=3fc91900 size=024a8h (  9384) load
> I (260) esp_image: segment 2: paddr=000e8eb8 vaddr=40374000 size=07160h ( 29024) load
> I (270) esp_image: segment 3: paddr=000f0020 vaddr=42000020 size=6ff6ch (458604) map
> I (354) esp_image: segment 4: paddr=0015ff94 vaddr=4037b160 size=066cch ( 26316) load
> I (366) boot: Loaded app from partition at offset 0x10000
> I (366) boot: Disabling RNG early entropy source...
> I (379) cpu_start: Pro cpu up.
> I (379) cpu_start: Starting app cpu, entry point is 0x4037529c
> 0x4037529c: call_start_cpu1 at C:/Espressif/frameworks/esp-idf-v5.0.2/components/esp_system/port/cpu_start.c:141
> 
> I (0) cpu_start: App cpu up.
> I (393) cpu_start: Pro cpu start user code
> I (393) cpu_start: cpu freq: 160000000 Hz
> I (393) cpu_start: Application information:
> I (396) cpu_start: Project name:     rpw-tflite-v3
> I (402) cpu_start: App version:      1
> I (406) cpu_start: Compile time:     Oct 31 2023 14:15:44
> I (412) cpu_start: ELF file SHA256:  55ecc57c2662042f...
> I (418) cpu_start: ESP-IDF:          v5.0.2-dirty
> I (424) cpu_start: Min chip rev:     v0.0
> I (428) cpu_start: Max chip rev:     v0.99
> I (433) cpu_start: Chip rev:         v0.1
> I (438) heap_init: Initializing. RAM available for dynamic allocation:
> I (445) heap_init: At 3FCD4840 len 00014ED0 (83 KiB): DRAM
> I (451) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
> I (458) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
> I (464) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
> I (471) spi_flash: detected chip: gd
> I (475) spi_flash: flash io: dio
> W (479) i2s(legacy): legacy i2s driver is deprecated, please migrate to use driver/i2s_std.h, driver/i2s_pdm.h or driver/i2s_tdm.h
> I (493) cpu_start: Starting scheduler on PRO CPU.
> I (0) cpu_start: Starting scheduler on APP CPU.
> I (20) FEATURE_PROVIDER: InitializeMicroFeatures successful
> Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
> 
> Core  0 register dump:
> PC      : 0x40056f60  PS      : 0x00060430  A0      : 0x8200a035  A1      : 0x3fcdc2f0
> A2      : 0x00000000  A3      : 0x3fcad9f0  A4      : 0x0000c800  A5      : 0x00000000
> A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3fcdc2a0
> A10     : 0x3fc94018  A11     : 0x3fc94018  A12     : 0x00060420  A13     : 0x00060423
> A14     : 0x00000001  A15     : 0x0000cdcd  SAR     : 0x00000002  EXCCAUSE: 0x0000001d
> EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000c7f
> 
> 
> Backtrace: 0x40056f5d:0x3fcdc2f0 |<-CORRUPTED
> 
> 
> 
> 
> ELF file SHA256: 55ecc57c2662042f
> 
> Rebooting...

vikramdattu, 2023-11-01:
@adityale1711 thanks for the additional context.
The function `GetAudioSamples1` is added to aid in debugging which reads 16K samples in one go!. If you want to use it for any reason, can you please apply attached change here and let me know if this works for you? (I have also enabled the use of GetAudioSamples1, so you do not need to do that additionally.)
[test_change.patch](https://github.com/espressif/esp-tflite-micro/files/13225034/test_change.patch)

adityale1711, 2023-11-02:
I've tried with your test_change.patch but I still got the same error

vikramdattu, 2023-11-02:
@adityale1711 sorry to know that. Since the issue is not re-producible at my end, may I ask you to share the minimal modified example from you which I can check?
Thanks.

adityale1711, 2023-11-02:
Here's my modified:

1. I'm using my model for pest detection
2. kTensorArenaSize is 100 * 1024
3. 

> // Get information about the memory area to use for the model's input.
> model_input = interpreter->input(0);
> if ((model_input->dims->size != 4) || (model_input->dims->data[0] != 1) || (model_input->dims->data[1] != kFeatureElementCount) || (model_input->type != kTfLiteInt8)){
>         MicroPrintf("Bad input tensor parameters in model");
>         return;
>     }

4. `extern const uint8_t infested_start[] asm("_binary_infested_wav_start");`
`extern const uint8_t not_infested_start[] asm("_binary_not_infested_wav_start");`

5. I'm using sample rate 8 KHz

> *how_many_new_slice = kFeatureSliceCount;
> int16_t *audio_samples = nullptr;
> int audio_samples_size = 0;
> 
> static int cnt = 0;
> audio_samples = (int16_t *)(not_infested_start + 44);
> 
> switch (cnt++ % 2){
> case 0:
>     audio_samples = (int16_t *)(infested_start + 44);
>     break;
> case 1:
>     audio_samples = (int16_t *)(not_infested_start + 44);
>     break;
> 
> default:
>     break;
> }
> 
> audio_samples_size = 8000;
> 
> GetAudioSamples(&audio_samples_size, &audio_samples);

6. Here's I trying to increase and decrease this value
`const int32_t kAudioCaptureBufferSize = 400 * 1024;`
`const int32_t i2s_bytes_to_read = 100 * 1024;`

7. And modified code like in test_change.patch

vikramdattu, 2023-11-02:
Hi @adityale1711 the buffer sizes you are setting are huge and are not allocatable on the SRAM.
Please reduce these to the one already set and it should work fine.

```
const int32_t kAudioCaptureBufferSize = 40000;
const int32_t i2s_bytes_to_read = 3200;
```

adityale1711, 2023-11-03:
I tried not calling the GetAudioSamples function but I got another error like this:

> Output type UINT32 (16) not supported.
> Type INT32 (2) not supported.
> Node ADD (number 16) failed to invoke with status 1
> Feature generator model invocation failed

> My audio is 16-bit per sample and the Sample Rate is 8000 Hz

It's the same error when I try to run micro_speech without modifying the code.

vikramdattu, 2023-11-03:
@adityale1711 there must be some issues with your workspace/build.
- Please check if you are on the latest code base. Do a `git pull --rebase`
- Do check for any local changes using `git status` and `git log`.
- Do a clean build using`idf.py fullclean` and then build the example again.

I am closing this issue, as the `incorrect bytes write` issue is not valid anymore. Please feel free to open a new issue for this new problem.

closed 2023-11-03 by vikramdattu