espressif/esp-tflite-micro #72: Example person_detection does not build.

MichelBerg, 2024-02-01:
I am trying the esp-tflite-micro examples. The hello world example worked as intended. But while building the person_detection example i run into the following error even though i followed all the steps (https://github.com/espressif/esp-tflite-micro/tree/master/examples/person_detection)

I am using esp-idf release/v4.4 and my board is a ESP32-S3-EYE.

```
bergm@bergm-VirtualBox:~/esp_idf_v4.4/esp-who-v4.4/examples/hello_world/person_detection_tflm$ idf.py build

Executing action: all (aliases: build)

Running ninja in directory /home/bergm/esp_idf_v4.4/esp-who-v4.4/examples/hello_world/person_detection_tflm/build

Executing "ninja all"...

[1/14] Performing build step for 'bootloader'

[1/1] cd /home/bergm/esp_idf_v4.4/esp-who-v4.4/examples/hello_world/person_detection_tflm/build/bootloader/esp-idf/esptool_py && /home/bergm/.espressif/python_env/idf4.4_py3.10_env/bin/python /home/bergm/esp_idf_v4.4/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /home/bergm/esp_idf_v4.4/esp-who-v4.4/examples/hello_world/person_detection_tflm/build/bootloader/bootloader.bin

Bootloader binary size 0x5820 bytes. 0x27e0 bytes (31%) free.

[2/12] Building C object esp-idf/espre...p32_s3_eye.dir/esp32_s3_eye_idf4.c.obj

FAILED: esp-idf/espressif__esp32_s3_eye/CMakeFiles/__idf_espressif__esp32_s3_eye.dir/esp32_s3_eye_idf4.c.obj 

/home/bergm/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc -DLV_CONF_INCLUDE_SIMPLE -I/home/bergm/esp_idf_v4.4/esp-who-v4.4/examples/hello_world/person_detection_tflm/build/config -I/home/bergm/esp_idf_v4.4/esp-who-v4.4/examples/hello_world/person_detection_tflm/managed_components/espressif__esp32_s3_eye/include -I/home/bergm/esp_idf_v4.4/esp-who-v4.4/examples/hello_world/person_detection_tflm/managed_components/espressif__esp32_s3_eye/priv_include -I/home/bergm/esp_idf_v4.4/esp-idf/components/newlib/platform_include -I/home/bergm/esp_idf_v4.4/esp-idf/components/freertos/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/freertos/include/esp_additions/freertos -I/home/bergm/esp_idf_v4.4/esp-idf/components/freertos/port/xtensa/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/freertos/include/esp_additions -I/home/bergm/esp_idf_v4.4/esp-idf/components/esp_hw_support/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/esp_hw_support/include/soc -I/home/bergm/esp_idf_v4.4/esp-idf/components/esp_hw_support/include/soc/esp32s3 -I/home/bergm/esp_idf_v4.4/esp-idf/components/esp_hw_support/port/esp32s3/. -I/home/bergm/esp_idf_v4.4/esp-idf/components/esp_hw_support/port/esp32s3/private_include -I/home/bergm/esp_idf_v4.4/esp-idf/components/heap/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/log/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/lwip/include/apps -I/home/bergm/esp_idf_v4.4/esp-idf/components/lwip/include/apps/sntp -I/home/bergm/esp_idf_v4.4/esp-idf/components/lwip/lwip/src/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/lwip/port/esp32/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/lwip/port/esp32/include/arch -I/home/bergm/esp_idf_v4.4/esp-idf/components/soc/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/soc/esp32s3/. -I/home/bergm/esp_idf_v4.4/esp-idf/components/soc/esp32s3/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/hal/esp32s3/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/hal/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/hal/platform_port/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/esp_rom/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/esp_rom/include/esp32s3 -I/home/bergm/esp_idf_v4.4/esp-idf/components/esp_rom/esp32s3 -I/home/bergm/esp_idf_v4.4/esp-idf/components/esp_common/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/esp_system/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/esp_system/port/soc -I/home/bergm/esp_idf_v4.4/esp-idf/components/esp_system/port/public_compat -I/home/bergm/esp_idf_v4.4/esp-idf/components/xtensa/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/xtensa/esp32s3/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/driver/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/driver/esp32s3/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/esp_pm/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/esp_ringbuf/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/efuse/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/efuse/esp32s3/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/vfs/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/esp_wifi/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/esp_event/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/esp_netif/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/esp_eth/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/tcpip_adapter/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/esp_phy/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/esp_phy/esp32s3/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/esp_ipc/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/app_trace/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/esp_timer/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/spiffs/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/spi_flash/include -I/home/bergm/esp_idf_v4.4/esp-who-v4.4/examples/hello_world/person_detection_tflm/managed_components/espressif__button/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/esp_adc_cal/include -I/home/bergm/esp_idf_v4.4/esp-who-v4.4/examples/hello_world/person_detection_tflm/managed_components/espressif__esp32-camera/driver/include -I/home/bergm/esp_idf_v4.4/esp-who-v4.4/examples/hello_world/person_detection_tflm/managed_components/espressif__esp32-camera/conversions/include -I/home/bergm/esp_idf_v4.4/esp-who-v4.4/examples/hello_world/person_detection_tflm/managed_components/espressif__esp_codec_dev/include -I/home/bergm/esp_idf_v4.4/esp-who-v4.4/examples/hello_world/person_detection_tflm/managed_components/espressif__esp_codec_dev/interface -I/home/bergm/esp_idf_v4.4/esp-who-v4.4/examples/hello_world/person_detection_tflm/managed_components/espressif__esp_codec_dev/device/include -I/home/bergm/esp_idf_v4.4/esp-who-v4.4/examples/hello_world/person_detection_tflm/managed_components/espressif__esp_lvgl_port/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/esp_lcd/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/esp_lcd/interface -I/home/bergm/esp_idf_v4.4/esp-who-v4.4/examples/hello_world/person_detection_tflm/managed_components/lvgl__lvgl -I/home/bergm/esp_idf_v4.4/esp-who-v4.4/examples/hello_world/person_detection_tflm/managed_components/lvgl__lvgl/src -I/home/bergm/esp_idf_v4.4/esp-who-v4.4/examples/hello_world/person_detection_tflm/managed_components -I/home/bergm/esp_idf_v4.4/esp-who-v4.4/examples/hello_world/person_detection_tflm/managed_components/lvgl__lvgl/examples -I/home/bergm/esp_idf_v4.4/esp-who-v4.4/examples/hello_world/person_detection_tflm/managed_components/lvgl__lvgl/demos -I/home/bergm/esp_idf_v4.4/esp-idf/components/fatfs/diskio -I/home/bergm/esp_idf_v4.4/esp-idf/components/fatfs/vfs -I/home/bergm/esp_idf_v4.4/esp-idf/components/fatfs/src -I/home/bergm/esp_idf_v4.4/esp-idf/components/wear_levelling/include -I/home/bergm/esp_idf_v4.4/esp-idf/components/sdmmc/include -mlongcalls  -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -O2 -fmacro-prefix-map=/home/bergm/esp_idf_v4.4/esp-who-v4.4/examples/hello_world/person_detection_tflm=. -fmacro-prefix-map=/home/bergm/esp_idf_v4.4/esp-idf=IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.4.6-440-g27ec26d2d3\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -DBUTTON_VER_MAJOR=3 -DBUTTON_VER_MINOR=2 -DBUTTON_VER_PATCH=0 -MD -MT esp-idf/espressif__esp32_s3_eye/CMakeFiles/__idf_espressif__esp32_s3_eye.dir/esp32_s3_eye_idf4.c.obj -MF esp-idf/espressif__esp32_s3_eye/CMakeFiles/__idf_espressif__esp32_s3_eye.dir/esp32_s3_eye_idf4.c.obj.d -o esp-idf/espressif__esp32_s3_eye/CMakeFiles/__idf_espressif__esp32_s3_eye.dir/esp32_s3_eye_idf4.c.obj -c /home/bergm/esp_idf_v4.4/esp-who-v4.4/examples/hello_world/person_detection_tflm/managed_components/espressif__esp32_s3_eye/esp32_s3_eye_idf4.c

[... cut ...]
/home/bergm/esp_idf_v4.4/esp-who-v4.4/examples/hello_world/person_detection_tflm/managed_components/espressif__esp32_s3_eye/esp32_s3_eye.c:378:42: error: unknown type name 'lv_disp_rot_t'; did you mean 'lv_disp_t'?

 void bsp_display_rotate(lv_disp_t *disp, lv_disp_rot_t rotation)

                                          ^~~~~~~~~~~~~

                                          lv_disp_t

[4/12] Linking C static library esp-idf/lvgl__lvgl/liblvgl__lvgl.a

ninja: build stopped: subcommand failed.

ninja failed with exit code 1

```

vikramdattu, 2024-02-01:
Hello @MichelBerg the builds are always tested with CI.
To make sure, I tried the build on my machine and it went perfect. Can you please do a clean build once? (Please remove `dependencies.lock` and `managed_components/` dirs as well along with `build/`, `sdkconfig` and `sdkconfig.old`)

MichelBerg, 2024-02-02:
Thanks for the fast response! 
I deleted the files as you said, but I still got the same error.
My environment is just a new VM using Ubuntu. Hello World example works but speech and person_detection do not. I still couldn't figure out why.

MichelBerg, 2024-02-13:
I was now able to build the project (error was caused by lvgl), but deployment won't work on all of my boards.
I tired to flash the CLI example to an ESP32-S3, which totaly worked.

But flashing on a ESP32-S3-EYE does not work. The console shows the following output:

...
```

Connecting...

Chip is ESP32-S3 (QFN56) (revision v0.1)

Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)

Crystal is 40MHz

MAC: 34:85:18:8c:57:64

Uploading stub...

Running stub...

Stub running...

Changing baud rate to 460800

Changed.

Configuring flash size...

Auto-detected Flash size: 8MB

Flash will be erased from 0x00000000 to 0x00005fff...

Flash will be erased from 0x00010000 to 0x000bafff...

Flash will be erased from 0x00008000 to 0x00008fff...

Compressed 21904 bytes to 13841...

Writing at 0x00000000... (100 %)

A fatal error occurred: Packet content transfer stopped (received 2 bytes)

CMake Error at run_serial_tool.cmake:58 (message):

  

  /home/bergm/.espressif/python_env/idf5.0_py3.10_env/bin/python;;/home/bergm/esp_idf_v5.0/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3

  failed
```
...

I have no clue why it does not work... I mean it is the same Chip?

vikramdattu, 2024-02-13:
@MichelBerg please try following two options:
1. Put the device into download mode manually: press `boot` button, press and release rst/en button while pressing the boot button. release. Does the board flash now?
After flashing you need to press rst/en to exit the download mode.
2. Use `--no-stub` option to the `esptool.py` command. The complete command is printed on console after `idf.py build`. copy that command and add `--no-stub` before `write_flash`
Does this work?

MichelBerg, 2024-02-13:
The second option made the program flashing. I added it to the idf.py menuconfig --> Serial falsher config --> [*] Disable download stub. (As you suggested, ty very much! 👍🏼)

After that the program crashed due to a "false" PSRAM config for the ESP32-S3-Eye (Octal instead of Quad-Mode)

While running the CLI Inference, I am not able to send commands over the console.

With the Display support, the code is running at 14 FPS and 100% CPU.

SaketNer, 2024-02-13:
Hey, i faced the same issue, for now, you can get rid of the issue by going to Menuconfig and setting "select BSP"  in application configuration to None. Doing this will not build the bsp package and hence "lv_disp_rot_t" issue will not come.

MichelBerg, 2024-02-13:
@SaketNer this might also be interesting https://github.com/espressif/esp-bsp/issues/292. I am working with BSP and faced some errors for another project. This might be helpful for you.

tore-espressif, 2024-02-20:
Hi guys, sorry about the problems with BSPs, with faced some issues with latest LVGL 9. It should be all fixed now.

> After that the program crashed due to a "false" PSRAM config for the ESP32-S3-Eye (Octal instead of Quad-Mode)

Where did you get you s3-eye from? Mass production S3-EYE is populated with Octal PSRAM

MichelBerg, 2024-02-20:
Yeah i guess its from the sdkconfig. My S3-EYE has the octal PSRAM. But in the config it was set to quad.
Thanks for your support guys. I think we can close this now.

closed 2024-02-20 by MichelBerg