espressif/esp-tflite-micro #73: Sound recognition?

gamename, 2024-02-22:
Hi,

Can this be used for generic sound recognition? Specifically, I'm looking for a way to identify cat sounds.  I already have the model, but I'm not sure if this is the correct library.

Suggestions appreciated.  

Thank you.
-T

vikramdattu, 2024-02-22:
Hi @gamename you can very well do that. Hope you already have quantised the model to suit tflite_micro format.

You can make existing micro_speech example to suit your needs and it should be good to go!
Once, that's done start by changing the model. Convert the model in cc format and put the contents in model.cc + model.h. 

Quick command to do this:
```
xxd -I cat_voice_model.tflite > cat_voice_model.cc
```
Other changes can go to [main_functions.cc](https://github.com/espressif/esp-tflite-micro/blob/master/examples/micro_speech/main/main_functions.cc). Feel free to explore.

gamename, 2024-02-22:
> Hi @gamename you can very well do that. Hope you already have quantised the model to suit tflite_micro format.
>
> You can make existing micro_speech example to suit your needs and it should be good to [...]

Thanks!

closed 2024-02-26 by gamename