espressif/esp-tflite-micro #74: Sound detection - Is this possible?

gamename, 2024-02-26:
Hi,

I'm using an esp32-s3-eye v2.2. It has 8MB each of flash and PSRAM.  Is it possible to use `yamnet.tflite` on an esp32-s3-eye v2.2 for sound identification? The `yamnet.tflite` file is about 3.9M in size.

The chip has an sd card slot, so I can use it to load the model file (i.e. no need to convert it to a `.cc` file with `xxd`.

Thoughts?

vikramdattu, 2024-02-27:
@gamename going by the size of the model, it is a quantised model I believe. If not, I would suggest you to quantise it to int8 weights. That'll reduce the size of the model to about 1/4th. Have you tried it with `.cc` first? Converting to `.cc` doesn't really increase the size of the model when the file get's embedded into the application, if you are concerned about that. The .cc file looks larger but the array size it puts the model in is still smaller.

About SD card, unfortunately, I have not really tried this approach. It definitely is a worth of a try IMO.  When loading from the SD card however, it makes sense to not convert to .cc as you suggest.

Let me know how it goes. If you need further help or want me to try, do let me know.

gamename, 2024-02-28:
@vikramdattu 

Right now, it is not working. Everything loads from the sd card, but the code is not able to discern cat sounds. Not sure if it is the model or the esp32-s3 code. I'm stalled. 

If you have the time, could you have a look please?

All the code is [here](https://github.com/cat-doorbell-v3). There is a repo for creating the model and another which is essentially a copy of the `micro_speech` example with cat-specific changes.

Thanks, and no worries if you do not have the bandwidth to look at it.

-T

gamename, 2024-02-29:
@vikramdattu 

What process did you use to build the `yes_micro_features_data.cc` file?  I'm not referring to the `xxd` conversion.  I'm referring to everything up to that?

The reason I ask is the C array in `yes_micro_features_data.cc` is tiny. I would like to replicate that size for my cat meow identification too.  

Thanks
-T

vikramdattu, 2024-03-01:
Hi @gamename this is test data and I had taken it long back from googles's tflite-micro. 
Currently, the feature generation happens via a different model in [this file](https://github.com/espressif/esp-tflite-micro/blob/master/examples/micro_speech/main/micro_features_generator.cc) and the features are then fed to detection model.

The tools [here](https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/examples/micro_speech#readme) can help you train your own model, evaluate it and convert it.

gamename, 2024-03-01:
> Hi @gamename this is test data and I had taken it long back from googles's tflite-micro. Currently, the feature generation happens via a different model in [this [...]

Thank you, sir.

gamename, 2024-03-03:
Hi @vikramdattu 

I'm getting this error. It looks like there are missing directories (`_site/tensorflow/include/**`).  Any idea how to fix it? Is there something I should run first?

```
bazel build //tensorflow/lite/micro/examples/micro_speech:audio_preprocessor
Starting local Bazel server and connecting to it...
ERROR: Traceback (most recent call last):
	File "/private/var/tmp/_bazel_tennis/e77d9007a951dc4f16a5de816efa9ff2/external/tensorflow_cc_deps/BUILD", line 7, column 16, in <toplevel>
		hdrs = glob(["_site/tensorflow/include/**"], allow_empty=False, exclude_directories=1),
Error in glob: glob pattern '_site/tensorflow/include/**' didn't match anything, but allow_empty is set to False (the default value of allow_empty can be set with --incompatible_disallow_empty_glob).
ERROR: /private/var/tmp/_bazel_tennis/e77d9007a951dc4f16a5de816efa9ff2/external/tensorflow_cc_deps/BUILD: no such target '@@tensorflow_cc_deps//:cc_library': target 'cc_library' not declared in package '' defined by /private/var/tmp/_bazel_tennis/e77d9007a951dc4f16a5de816efa9ff2/external/tensorflow_cc_deps/BUILD (Tip: use `query "@tensorflow_cc_deps//:*"` to see all the targets in that package)
ERROR: /Users/tennis/esp/tflite-micro/python/tflite_micro/signal/BUILD:255:23: no such target '@@tensorflow_cc_deps//:cc_library': target 'cc_library' not declared in package '' defined by /private/var/tmp/_bazel_tennis/e77d9007a951dc4f16a5de816efa9ff2/external/tensorflow_cc_deps/BUILD (Tip: use `query "@tensorflow_cc_deps//:*"` to see all the targets in that package) and referenced by '//python/tflite_micro/signal:ops/_stacker_op.so'
ERROR: Analysis of target '//tensorflow/lite/micro/examples/micro_speech:audio_preprocessor' failed; build aborted: Analysis failed
INFO: Elapsed time: 5.571s, Critical Path: 0.06s
INFO: 1 process: 1 internal.
ERROR: Build did NOT complete successfully
FAILED:
    Fetching repository @@tflm_pip_deps_numpy; starting
    Fetching repository @@local_config_xcode; Building xcode-locator
    Fetching repository @@tflm_pip_deps_tensorflow_macos; starting
```

gamename, 2024-03-08:
@vikramdattu 

Is your pre-processor model taken from [here](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/examples/micro_speech/models/audio_preprocessor_int8.tflite)?

Thanks,
-T

vikramdattu, 2024-03-09:
@gamename that's right, the model is taken from that particular location.

gamename, 2024-03-09:
> @gamename that's right, the model is taken from that particular location.

Perfect. Thanks.

gamename, 2024-03-10:
@vikramdattu 

for `micro_speech` example, what is the purpose of having `yes_micro_features_data.cc/h` and `no_micro_features_data.cc/h` in the directory? Are they there for reference? They don't seem to be used - or am I missing something?

vikramdattu, 2024-03-11:
@gamename you are correct. Those were there from old days added for testing and are not used currently. You may ignore those.

gamename, 2024-03-11:
> @gamename you are correct. Those were there from old days added for testing and are not used currently. You may ignore those.

Thanks!

gamename, 2024-03-11:
@vikramdattu 

This concerns building the actual model. I am using a script [here](https://github.com/gamename/tflite-micro-fork/blob/main/tensorflow/lite/micro/examples/micro_speech/train/meow_train.py) that is just a compilation of the steps outlined [here](https://github.com/gamename/tflite-micro-fork/blob/main/tensorflow/lite/micro/examples/micro_speech/train/train_micro_speech_model.ipynb). 

Here is what my input dir with samples looks like:

```bash
tree ./samples
./samples
├── _background_noise_
│   ├── README.md
│   ├── doing_the_dishes.wav
│   ├── dude_miaowing.wav
│   ├── exercise_bike.wav
│   ├── pink_noise.wav
│   ├── running_tap.wav
│   └── white_noise.wav
└── meow
    ├── cat0001.wav
    ├── cat0002.wav
    ...
```
(there are 77 total cat `.wav` files)

I'm confused about what needs to be in there.  Do I need to add a `silence` and `unknown` subdir (with contents) as well?

Thanks
-T

gamename, 2024-03-13:
@vikramdattu 

Another question. :)

Looking at [this construct](https://github.com/espressif/esp-tflite-micro/blob/61af88b7b30fda2078a7b52b2c1b600899a73e2e/examples/micro_speech/main/micro_model_settings.h#L31):
```// Variables for the model's output categories.
constexpr int kCategoryCount = 4;
constexpr const char* kCategoryLabels[kCategoryCount] = {
    "silence",
    "unknown",
    "yes",
    "no",
};```

...how do you know what the order of the labels ("silence", "unknown", etc) should be?  How is that set?

vikramdattu, 2024-03-14:
> [External: This email originated outside Espressif]

Hello TennisSmith,

That completely depends on the model trained.
It cannot be inferred from the model what categories are. Only the number of categories can be known from output tensor size.

Thanks.,
Vikram

gamename, 2024-03-14:
> That completely depends on the model trained. It cannot be inferred from the model what categories are. Only the number of categories can be known from output tensor size. 

That's not quite what I am asking. :)

My question is this: How do I know the *order* of the labels as they are used in python after the model has been created?

vikramdattu, 2026-11-09:
Closing, as there has been no activity here for a long time. esp-tflite-micro has changed a lot since (examples, ESP-IDF support and the upstream TFLite Micro sync). If you still see this with the latest release, please open a new issue with the ESP-IDF version, the board and the full log.

closed 2026-11-09 by vikramdattu