espressif/esp-tflite-micro #77: Problems  running hello_world

Hsannriahi, 2024-03-26:
hello, we are group of two peoples , we are working on implmentation of neural network on esp32 .besides , we are not familiar yet with this .
we have tried to run the hello_world model but we had some problems could u help us to fix it ?
[image]

vikramdattu, 2024-03-26:
Can you please explain the error you face? Check if you have internet connectivity as `esp-nn` will be resolved runtime.

Hsannriahi, 2024-03-26:
we was following the instructions in Readme to run the main.cc
we went to the project file and we ran it using the command idf.py build then execution stop (see-picture below)
im sure that the environment esp-idf is working perfectly 
before running the project i just added the dependencies  using
[image]
 idf.py add-dependency "esp-tflite-micro"

vikramdattu, 2024-03-27:
@Hsannriahi you do not need to explicitly add the `esp-tflite-micro` dependency as it is already taken care in examples [idf_component.yml](https://github.com/espressif/esp-tflite-micro/blob/master/examples/hello_world/main/idf_component.yml).
Which in turn should pull `esp-nn` automatically and build should go through.
Can you please share the `dependencies.lock` file which is generated in `example/hello_world` directory?

closed 2026-11-09 by vikramdattu