espressif/esp-tflite-micro #78: Problems with `micro_speech` example for esp32-s3-eye

gamename, 2024-04-04:
Hi,

I tried to use the `micro_speech` example on an esp32-s3-eye (v2.2).  I set it to be type `esp32s3` and it compiled just fine.  

However, the chip does not "hear" speech at all.  See the log below.  It simply stops at `main_task: Calling app_main()`. 

Is there something more I need to configure?

Thanks,
-T

```
Executing action: all (aliases: build)
Running ninja in directory /Users/tennis/src/esp32-s3-eye-micro-speech-example/build
Executing "ninja all"...
[0/1] Re-running CMake...
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
Processing 3 dependencies:
[3/3] idf (5.2.0)
-- Project sdkconfig file /Users/tennis/src/esp32-s3-eye-micro-speech-example/sdkconfig
Loading defaults file /Users/tennis/src/esp32-s3-eye-micro-speech-example/sdkconfig.defaults...
/Users/tennis/src/esp32-s3-eye-micro-speech-example/sdkconfig.defaults:16 CONFIG_FLASHMODE_QIO was replaced with CONFIG_ESPTOOLPY_FLASHMODE_QIO
/Users/tennis/src/esp32-s3-eye-micro-speech-example/sdkconfig.defaults:18 CONFIG_INT_WDT was replaced with CONFIG_ESP_INT_WDT
/Users/tennis/src/esp32-s3-eye-micro-speech-example/sdkconfig.defaults:19 CONFIG_TASK_WDT was replaced with CONFIG_ESP_TASK_WDT_INIT
/var/folders/x2/4lbk1sy91jv_j1811qfzb02h0000gn/T/kconfgen_tmpjs1v0wts:18 line was updated to CONFIG_ESP_INT_WDT=n
/var/folders/x2/4lbk1sy91jv_j1811qfzb02h0000gn/T/kconfgen_tmpjs1v0wts:19 line was updated to CONFIG_ESP_TASK_WDT_INIT=n
-- Compiler supported targets: xtensa-esp-elf
-- App "micro_speech" version: 1
-- Adding linker script /Users/tennis/src/esp32-s3-eye-micro-speech-example/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /Users/tennis/esp/v5.2/esp-idf/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script /Users/tennis/esp/v5.2/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script /Users/tennis/esp/v5.2/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script /Users/tennis/esp/v5.2/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script /Users/tennis/esp/v5.2/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script /Users/tennis/esp/v5.2/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script /Users/tennis/esp/v5.2/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_bootloader_format esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__esp-nn espressif__esp-tflite-micro esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash nvs_sec_provider openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport touch_element ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /Users/tennis/esp/v5.2/esp-idf/components/app_trace /Users/tennis/esp/v5.2/esp-idf/components/app_update /Users/tennis/esp/v5.2/esp-idf/components/bootloader /Users/tennis/esp/v5.2/esp-idf/components/bootloader_support /Users/tennis/esp/v5.2/esp-idf/components/bt /Users/tennis/esp/v5.2/esp-idf/components/cmock /Users/tennis/esp/v5.2/esp-idf/components/console /Users/tennis/esp/v5.2/esp-idf/components/cxx /Users/tennis/esp/v5.2/esp-idf/components/driver /Users/tennis/esp/v5.2/esp-idf/components/efuse /Users/tennis/esp/v5.2/esp-idf/components/esp-tls /Users/tennis/esp/v5.2/esp-idf/components/esp_adc /Users/tennis/esp/v5.2/esp-idf/components/esp_app_format /Users/tennis/esp/v5.2/esp-idf/components/esp_bootloader_format /Users/tennis/esp/v5.2/esp-idf/components/esp_coex /Users/tennis/esp/v5.2/esp-idf/components/esp_common /Users/tennis/esp/v5.2/esp-idf/components/esp_eth /Users/tennis/esp/v5.2/esp-idf/components/esp_event /Users/tennis/esp/v5.2/esp-idf/components/esp_gdbstub /Users/tennis/esp/v5.2/esp-idf/components/esp_hid /Users/tennis/esp/v5.2/esp-idf/components/esp_http_client /Users/tennis/esp/v5.2/esp-idf/components/esp_http_server /Users/tennis/esp/v5.2/esp-idf/components/esp_https_ota /Users/tennis/esp/v5.2/esp-idf/components/esp_https_server /Users/tennis/esp/v5.2/esp-idf/components/esp_hw_support /Users/tennis/esp/v5.2/esp-idf/components/esp_lcd /Users/tennis/esp/v5.2/esp-idf/components/esp_local_ctrl /Users/tennis/esp/v5.2/esp-idf/components/esp_mm /Users/tennis/esp/v5.2/esp-idf/components/esp_netif /Users/tennis/esp/v5.2/esp-idf/components/esp_netif_stack /Users/tennis/esp/v5.2/esp-idf/components/esp_partition /Users/tennis/esp/v5.2/esp-idf/components/esp_phy /Users/tennis/esp/v5.2/esp-idf/components/esp_pm /Users/tennis/esp/v5.2/esp-idf/components/esp_psram /Users/tennis/esp/v5.2/esp-idf/components/esp_ringbuf /Users/tennis/esp/v5.2/esp-idf/components/esp_rom /Users/tennis/esp/v5.2/esp-idf/components/esp_system /Users/tennis/esp/v5.2/esp-idf/components/esp_timer /Users/tennis/esp/v5.2/esp-idf/components/esp_wifi /Users/tennis/esp/v5.2/esp-idf/components/espcoredump /Users/tennis/src/esp32-s3-eye-micro-speech-example/managed_components/espressif__esp-nn /Users/tennis/src/esp32-s3-eye-micro-speech-example/managed_components/espressif__esp-tflite-micro /Users/tennis/esp/v5.2/esp-idf/components/esptool_py /Users/tennis/esp/v5.2/esp-idf/components/fatfs /Users/tennis/esp/v5.2/esp-idf/components/freertos /Users/tennis/esp/v5.2/esp-idf/components/hal /Users/tennis/esp/v5.2/esp-idf/components/heap /Users/tennis/esp/v5.2/esp-idf/components/http_parser /Users/tennis/esp/v5.2/esp-idf/components/idf_test /Users/tennis/esp/v5.2/esp-idf/components/ieee802154 /Users/tennis/esp/v5.2/esp-idf/components/json /Users/tennis/esp/v5.2/esp-idf/components/log /Users/tennis/esp/v5.2/esp-idf/components/lwip /Users/tennis/src/esp32-s3-eye-micro-speech-example/main /Users/tennis/esp/v5.2/esp-idf/components/mbedtls /Users/tennis/esp/v5.2/esp-idf/components/mqtt /Users/tennis/esp/v5.2/esp-idf/components/newlib /Users/tennis/esp/v5.2/esp-idf/components/nvs_flash /Users/tennis/esp/v5.2/esp-idf/components/nvs_sec_provider /Users/tennis/esp/v5.2/esp-idf/components/openthread /Users/tennis/esp/v5.2/esp-idf/components/partition_table /Users/tennis/esp/v5.2/esp-idf/components/perfmon /Users/tennis/esp/v5.2/esp-idf/components/protobuf-c /Users/tennis/esp/v5.2/esp-idf/components/protocomm /Users/tennis/esp/v5.2/esp-idf/components/pthread /Users/tennis/esp/v5.2/esp-idf/components/sdmmc /Users/tennis/esp/v5.2/esp-idf/components/soc /Users/tennis/esp/v5.2/esp-idf/components/spi_flash /Users/tennis/esp/v5.2/esp-idf/components/spiffs /Users/tennis/esp/v5.2/esp-idf/components/tcp_transport /Users/tennis/esp/v5.2/esp-idf/components/touch_element /Users/tennis/esp/v5.2/esp-idf/components/ulp /Users/tennis/esp/v5.2/esp-idf/components/unity /Users/tennis/esp/v5.2/esp-idf/components/usb /Users/tennis/esp/v5.2/esp-idf/components/vfs /Users/tennis/esp/v5.2/esp-idf/components/wear_levelling /Users/tennis/esp/v5.2/esp-idf/components/wifi_provisioning /Users/tennis/esp/v5.2/esp-idf/components/wpa_supplicant /Users/tennis/esp/v5.2/esp-idf/components/xtensa
-- Configuring done (4.4s)
-- Generating done (0.3s)
-- Build files have been written to: /Users/tennis/src/esp32-s3-eye-micro-speech-example/build
[0/1] Re-running CMake...
-- Building ESP-IDF components for target esp32s3
-- Project sdkconfig file /Users/tennis/src/esp32-s3-eye-micro-speech-example/sdkconfig
-- Compiler supported targets: xtensa-esp-elf
-- Adding linker script /Users/tennis/esp/v5.2/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Bootloader project name: "bootloader" version: 1
-- Adding linker script /Users/tennis/esp/v5.2/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script /Users/tennis/esp/v5.2/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script /Users/tennis/esp/v5.2/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script /Users/tennis/esp/v5.2/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
[... cut ...]
I (272) spi_flash: flash io: qio
W (276) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
W (289) i2s(legacy): legacy i2s driver is deprecated, please migrate to use driver/i2s_std.h, driver/i2s_pdm.h or driver/i2s_tdm.h
I (301) sleep: Configure to isolate all GPIO pins in sleep state
I (308) sleep: Enable automatic switching of GPIO sleep configuration
I (315) main_task: Started on CPU0
I (325) main_task: Calling app_main()
```

gamename, 2024-04-06:
@vikramdattu FYI

gamename, 2024-04-06:
Here is a line from the above log.  Is this the real issue?
```W (289) i2s(legacy): legacy i2s driver is deprecated, please migrate to use driver/i2s_std.h, driver/i2s_pdm.h or driver/i2s_tdm.h
```

vikramdattu, 2024-04-08:
> Here is a line from the above log. Is this the real issue?
> 
> ```
> W (289) i2s(legacy): legacy i2s driver is deprecated, please migrate to use driver/i2s_std.h, driver/i2s_pdm.h or driver/i2s_tdm.h
> ```

This should not be an issue. Did you try the example on IDF `release/v5.1`?
I will test on 5.2 and update on the same.

gamename, 2024-04-08:
Hi @vikramdattu 

> This should not be an issue. Did you try the example on IDF `release/v5.1`? 

No. I'll try that too.

gamename, 2024-04-08:
Same results with `v5.1.3`:

```Executing action: all (aliases: build)
Running cmake in directory /Users/tennis/src/cat-doorbell-v3-esp32-s3-eye-micro-speech-example/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/Users/tennis/.espressif/python_env/idf5.1_py3.12_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /Users/tennis/src/cat-doorbell-v3-esp32-s3-eye-micro-speech-example"...
-- IDF_TARGET is not set, guessed 'esp32s3' from sdkconfig '/Users/tennis/src/cat-doorbell-v3-esp32-s3-eye-micro-speech-example/sdkconfig'
-- Found Git: /usr/bin/git (found version "2.39.3 (Apple Git-146)") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/tennis/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/tennis/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/tennis/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Dependency "idf (5.2.0)" version changed, solving dependencies.
...Updating lock file at /Users/tennis/src/cat-doorbell-v3-esp32-s3-eye-micro-speech-example/dependencies.lock
Processing 3 dependencies:
[1/3] espressif/esp-nn (1.0.2)
[2/3] espressif/esp-tflite-micro (1.2.0)
[3/3] idf (5.1.3)
-- Project sdkconfig file /Users/tennis/src/cat-doorbell-v3-esp32-s3-eye-micro-speech-example/sdkconfig
Loading defaults file /Users/tennis/src/cat-doorbell-v3-esp32-s3-eye-micro-speech-example/sdkconfig.defaults...
/Users/tennis/src/cat-doorbell-v3-esp32-s3-eye-micro-speech-example/sdkconfig.defaults:16 CONFIG_FLASHMODE_QIO was replaced with CONFIG_ESPTOOLPY_FLASHMODE_QIO
/Users/tennis/src/cat-doorbell-v3-esp32-s3-eye-micro-speech-example/sdkconfig.defaults:18 CONFIG_INT_WDT was replaced with CONFIG_ESP_INT_WDT
/Users/tennis/src/cat-doorbell-v3-esp32-s3-eye-micro-speech-example/sdkconfig.defaults:19 CONFIG_TASK_WDT was replaced with CONFIG_ESP_TASK_WDT_INIT
/var/folders/x2/4lbk1sy91jv_j1811qfzb02h0000gn/T/kconfgen_tmpcdqqkwv_:18 line was updated to CONFIG_ESP_INT_WDT=n
/var/folders/x2/4lbk1sy91jv_j1811qfzb02h0000gn/T/kconfgen_tmpcdqqkwv_:19 line was updated to CONFIG_ESP_TASK_WDT_INIT=n
Compiler supported targets: xtensa-esp32s3-elf

-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: /Users/tennis/.espressif/python_env/idf5.1_py3.12_env/bin/python (found version "3.12.2") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "micro_speech" version: c3aad2d
-- Adding linker script /Users/tennis/src/cat-doorbell-v3-esp32-s3-eye-micro-speech-example/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /Users/tennis/esp/v5.1.3/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script /Users/tennis/esp/v5.1.3/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script /Users/tennis/esp/v5.1.3/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script /Users/tennis/esp/v5.1.3/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script /Users/tennis/esp/v5.1.3/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script /Users/tennis/esp/v5.1.3/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script /Users/tennis/esp/v5.1.3/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__esp-nn espressif__esp-tflite-micro esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport touch_element ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /Users/tennis/esp/v5.1.3/components/app_trace /Users/tennis/esp/v5.1.3/components/app_update /Users/tennis/esp/v5.1.3/components/bootloader /Users/tennis/esp/v5.1.3/components/bootloader_support /Users/tennis/esp/v5.1.3/components/bt /Users/tennis/esp/v5.1.3/components/cmock /Users/tennis/esp/v5.1.3/components/console /Users/tennis/esp/v5.1.3/components/cxx /Users/tennis/esp/v5.1.3/components/driver /Users/tennis/esp/v5.1.3/components/efuse /Users/tennis/esp/v5.1.3/components/esp-tls /Users/tennis/esp/v5.1.3/components/esp_adc /Users/tennis/esp/v5.1. [... cut ...]
I (273) spi_flash: detected chip: gd
I (277) spi_flash: flash io: qio
W (281) i2s(legacy): legacy i2s driver is deprecated, please migrate to use driver/i2s_std.h, driver/i2s_pdm.h or driver/i2s_tdm.h
I (293) sleep: Configure to isolate all GPIO pins in sleep state
I (300) sleep: Enable automatic switching of GPIO sleep configuration
I (307) app_start: Starting scheduler on CPU0
I (312) app_start: Starting scheduler on CPU1
I (312) main_task: Started on CPU0
I (322) main_task: Calling app_main()
```

vikramdattu, 2024-04-14:
Hi @gamename I tried the example right now on esp-s3-eye (ESP32-S3-EYE-MB_V2.1) board and seems to be working fine for me.
I tried with IDF `release/v5.1`.
Will try with 5.2 as well.

```
I (358) main_task: Calling app_main()
I (378) feature_provider: InitializeMicroFeatures successful
I (448) TF_LITE_AUDIO_PROVIDER: Audio Recording started
Detected      no, score: 0.92
Detected     yes, score: 1.00
Detected unknown, score: 0.91
Detected unknown, score: 0.91
Detected      no, score: 0.93
Detected unknown, score: 0.81
Detected     yes, score: 1.00
Detected     yes, score: 1.00
Detected unknown, score: 0.84
Detected unknown, score: 1.00
Detected      no, score: 0.83
Detected unknown, score: 0.81
Detected     yes, score: 0.97
Detected     yes, score: 0.92
Detected      no, score: 0.97
Detected unknown, score: 0.90
```

gamename, 2024-04-14:
I found the problem! It was a copy of the code I was using, not the original one in the repo.  Somehow, I broke it when making the copy. Thanks for the help. Closing.

closed 2024-04-14 by gamename