espressif/esp-tflite-micro #80: tflite-micro is breaking complex.h

pschatzmann, 2024-04-16:
Not sure if [this issue](https://github.com/espressif/arduino-esp32/issues/9516) is better addressed here.

After all it does not make any sense to have the complex.h in src since you can be sure that it is available for all ESP32

maalos, 2024-09-18:
have you found an easy solution yet? i just got this problem too

vikramdattu, 2024-09-18:
I am not sure, how the `complex.h` from `esp-tflite-micro` is coming into picture for other projects, but most likely it's getting included from CMakeLists.txt. If you are building using lib_builder, removing this header includes from `INCLUDE_DIRS` should remove this issue.

[do_not_include_signal_headers.patch](https://github.com/user-attachments/files/17042892/do_not_include_signal_headers.patch)

pschatzmann, 2024-09-18:
Since a regular Arduino user has nothing to do with lib_builder, I do not see how this would help.