espressif/esp-tflite-micro #81: ESP-NN kernel does not allow for hybrid models

andresovela, 2024-05-28:
Running a model with a fully connected layer with different input / filter types fails due to the following assert in the ESP-NN kernel:

https://github.com/espressif/esp-tflite-micro/blob/7fd66e40112590056aeae4937c4506759775eea1/tensorflow/lite/micro/kernels/esp_nn/fully_connected.cc#L69-L70

In the normal kernel this condition is more relaxed:

https://github.com/espressif/esp-tflite-micro/blob/7fd66e40112590056aeae4937c4506759775eea1/tensorflow/lite/micro/kernels/fully_connected.cc#L59-L67

vikramdattu, 2024-05-29:
@andresovela thanks for bringing it to our attention. I will update the port files by this weekend and release the code.

vikramdattu, 2024-06-05:
@andresovela should be fixed with latest change: https://github.com/espressif/esp-tflite-micro/commit/30f24e6ace11c17b57a0feb3de201084435848ff.
The latest commit is tagged with `v1.3.1` and is pushed to component manager as well.

vikramdattu, 2026-11-09:
This was fixed in v1.3.1, as noted above. Closing.

closed 2026-11-09 by vikramdattu