espressif/esp-tflite-micro #82: Cannot build the hello world example on Windows 10

mingyr, 2024-06-01:
Hi, I have some problem in building the hello world example accompanying with the esp-tflite-micro repo after cloned it into my local hard disk. Here are some information and I appreciate you engineers can take some time looking into it and give some suggestions. Thanks a lot. 

`D:\esp-tflite-micro\examples\hello_world>idf.py set-target esp32s3
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory 'D:\esp-tflite-micro\examples\hello_world\build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32s3, new sdkconfig will be created.
Running cmake in directory D:\esp-tflite-micro\examples\hello_world\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=D:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=1 D:\esp-tflite-micro\examples\hello_world"...
-- Found Git: D:/Espressif/tools/idf-git/2.43.0/cmd/git.exe (found version "2.43.0.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: D:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: D:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Dependencies lock doesn't exist, solving dependencies.
.Using component placed at D:\esp-tflite-micro for dependency espressif/esp-tflite-micro(*), specified in D:/esp-tflite-micro/examples/hello_world/main\idf_component.yml
CMake Error at D:/Espressif/frameworks/esp-idf-v5.2.1/tools/cmake/build.cmake:544 (message):
  ERROR: Because no versions of espressif/esp-tflite-micro match >*

   and espressif/esp-tflite-micro (*) depends on espressif/esp-nn (^1.0.0-rc1), espressif/esp-tflite-micro (>=*) requires espressif/esp-nn (^1.0.0-rc1).

  So, because no versions of espressif/esp-nn match ^1.0.0-rc1

   and project depends on espressif/esp-tflite-micro (*), version solving failed.

Call Stack (most recent call first):
  D:/Espressif/frameworks/esp-idf-v5.2.1/tools/cmake/project.cmake:605 (idf_build_process)
  CMakeLists.txt:3 (project)

...-- Configuring incomplete, errors occurred!
See also "D:/esp-tflite-micro/examples/hello_world/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the D:\esp-tflite-micro\examples\hello_world\build\log\idf_py_stderr_output_25124 and D:\esp-tflite-micro\examples\hello_world\build\log\idf_py_stdout_output_25124

`

The following are some build system information: 

`D:\esp-tflite-micro\examples\hello_world>idf.py --version
ESP-IDF v5.2.1-dirty

D:\esp-tflite-micro\examples\hello_world>git log origin/master -n 1
commit 7fd66e40112590056aeae4937c4506759775eea1 (HEAD -> master, tag: v1.3.0, origin/master, origin/HEAD)
Author: Vikram Dattu <vikram.dattu@espressif.com>
Date:   Wed May 8 18:51:55 2024 +0530

    CI: Add builds for IDF v5.3

D:\esp-tflite-micro\examples\hello_world>ver

Microsoft Windows [Version 10.0.22631.3593]

D:\esp-tflite-micro\examples\hello_world>cmake --version
cmake version 3.24.0

CMake suite maintained and supported by Kitware (kitware.com/cmake).

D:\esp-tflite-micro\examples\hello_world>`

Feel free to contact me for any extra information helpful to diagnose the problem.

vikramdattu, 2024-06-05:
Hi @mingyr it is expected that `esp-nn` is automatically pulled in by component manager. This seems to be not happening in your case.
For component manager to download new component, you will need internet connection active and also the component manager installed. Which gets installed and added to the path with IDF's installation process.

If your problem persists, can you please make a directory named `components` in your project directory and clone [esp-nn](https://github.com/espressif/esp-nn) with folder name `esp-nn` inside this `components` directory.
This should override, component manager issue and get your build through.

mingyr, 2024-06-07:
Dear @vikramdattu, appreciate your response. I would think I have done as you mentioned in your response, however, the problem is STILL there. I would think I have no problem in accessing the Internet, please check whether it is the directory structure as you indicated:

D:\esp-tflite-micro\examples\hello_world>tree
Folder PATH listing
Volume serial number is 88E6-C63B
D:.
├───build
│   ├───CMakeFiles
│   │   ├───3.24.0
│   │   │   ├───CompilerIdASM
│   │   │   ├───CompilerIdC
│   │   │   │   └───tmp
│   │   │   └───CompilerIdCXX
│   │   │       └───tmp
│   │   ├───CMakeTmp
│   │   ├───git-data
│   │   └───pkgRedirects
│   └───log
├───components
│   └───esp-nn
│       ├───.github
│       │   └───workflows
│       ├───include
│       ├───src
│       │   ├───activation_functions
│       │   ├───basic_math
│       │   ├───common
│       │   ├───convolution
│       │   ├───fully_connected
│       │   ├───pooling
│       │   └───softmax
│       ├───tests
│       │   ├───include
│       │   └───src
│       └───test_app
│           └───main
└───main

D:\esp-tflite-micro\examples\hello_world>

vikramdattu, 2024-06-07:
@mingyr the directory structure looks correct.
What happens when you edit `esp-tflite-micro/idf_component.yml` contents to relax the esp-nn dependency?
```
dependencies:
  idf:
    version: ">=4.2"
  espressif/esp-nn:
    version: "*"
```

Also, can you please do one clean build? (`rm -r build/ sdkconfig sdkconfig.old dependencies.lock managed_components/`) and attach the cmake logs files for me to analyse? (`hello_world\build\log\idf_py_stdout_output_<>`)

mingyr, 2024-06-07:
I did as instructed above, seems the problem is still pending, please check the output from the commandline and attached files:

D:\esp-tflite-micro\examples\hello_world>type ..\..\idf_component.yml
description: TensorFlow Lite Micro component for ESP-IDF
url: https://github.com/espressif/tflite-micro-esp-examples
repository: https://github.com/espressif/tflite-micro-esp-examples.git
issues: https://github.com/espressif/tflite-micro-esp-examples/issues
documentation: https://github.com/espressif/tflite-micro-esp-examples#readme

dependencies:
  idf:
    version: ">=4.2"
  espressif/esp-nn:
#     version: "^1.0.0-rc1"
    version: "*"
    pre_release: true
files:
  exclude:
    - scripts
    - .gitlab-ci.yml
    - .github

D:\esp-tflite-micro\examples\hello_world>idf.py set-target esp32s3
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory 'D:\esp-tflite-micro\examples\hello_world\build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32s3, new sdkconfig will be created.
Running cmake in directory D:\esp-tflite-micro\examples\hello_world\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=D:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=1 D:\esp-tflite-micro\examples\hello_world"...
-- Found Git: D:/Espressif/tools/idf-git/2.43.0/cmd/git.exe (found version "2.43.0.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: D:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: D:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Dependencies lock doesn't exist, solving dependencies.
.Using component placed at D:\esp-tflite-micro for dependency espressif/esp-tflite-micro(*), specified in D:/esp-tflite-micro/examples/hello_world/main\idf_component.yml
CMake Error at D:/Espressif/frameworks/esp-idf-v5.2.1/tools/cmake/build.cmake:544 (message):
  ERROR: Because no versions of espressif/esp-tflite-micro match >*

   and espressif/esp-tflite-micro (*) depends on espressif/esp-nn (*), espressif/esp-tflite-micro (>=*) requires espressif/esp-nn (*).

  So, because no versions of espressif/esp-nn match *

   and project depends on espressif/esp-tflite-micro (*), version solving failed.

Call Stack (most recent call first):
  D:/Espressif/frameworks/esp-idf-v5.2.1/tools/cmake/project.cmake:605 (idf_build_process)
  CMakeLists.txt:3 (project)

The following are files grabbed from the mentioned log directory. However, to upload them here, I just append the .txt file extension which required by the system and without any touch of the file contents:

[idf_py_stdout_output_5100.txt](https://github.com/user-attachments/files/15721574/idf_py_stdout_output_5100.txt)
[idf_py_stderr_output_5100.txt](https://github.com/user-attachments/files/15721587/idf_py_stderr_output_5100.txt)

kumekay, 2024-06-07:
Hello @mingyr,
Could you please check if the issue still exist with the idf-component-manager from master branch? 

https://github.com/espressif/idf-component-manager?tab=readme-ov-file#installing-a-development-version-of-the-component-manager

This version, while compatible with ESP-IDF 5.2 may won't satisfy the constrain,
so you should either to remove line for `idf-component-manager` from `[HOME DIRECTORY]\.espressif\espidf.constraints.v5.2.txt` or try it with the master branch of ESP-IDF

mingyr, 2024-06-08:
Dear @kumekay , thanks for your suggestions. However, after I tried all aboves, seems the problem is still there. 

> This version, while compatible with ESP-IDF 5.2 may won't satisfy the constrain,
> so you should either to remove line for idf-component-manager from [HOME DIRECTORY]\.espressif\espidf.constraints.v5.2.txt or try it with the master branch of ESP-IDF

I first tried to give a try of this one, because it sounds just removing an item can solve the problem. However, the directory [HOME DIRECTORY]\.espressif seems NO exist in my system. The below is the command I tried to locate it:

`D:\esp-tflite-micro\examples\hello_world>dir %userprofile% | findstr .espressif`

I would believe the above command is correct, since I tried on other hidden folders such as .conda, it correctly pins it out:
```
D:\esp-tflite-micro\examples\hello_world>dir %userprofile% | findstr .conda
2023/09/18  16:00    <DIR>          .conda
2023/09/25  14:32                36 .condarc
2023/09/18  15:45    <DIR>          miniconda3

```

Later, I found it locate in the the installation directory D:\Espressif. However, after I commented the line "idf-component-manager~=1.2" out, and retry the build process, the same error showed up threre again.

Because I just installed the idf toolset offline, I am not sure updating to the master branch will break the compatibility or not, so I didn't do it.

Then I turned to your first suggestion, I read through the thread of the link, did as instructed perticular below:

```
# remove old version of the component manager
python -m pip uninstall -y idf-component-manager
# install the development version (from the main branch)
python -m pip install git+https://github.com/espressif/idf-component-manager.git@main
```

However, after I did it, the ESP-IDF command line terminal refuses to start up correctly. It complains something, sounds like there might be some conflicts and indicates I have to install something more. (I thought I need the administrative privilege and I simply closed the window withou copy the error message, only the instruactions. So I don't remember exactly what did it complain later)

After I installed more packages and try to relaunch the idf terminal and to rebuild the project, it works FINALLY.

So up to now, the root cause seems to be with the component manager. 

BIG thanks to you. @kumekay

mingyr, 2024-06-08:
Since it can work at least temporarily, just close the issue.

closed 2024-06-08 by mingyr