espressif/esp-tflite-micro #87: hello-world example can't load model

maxgerhardt, 2024-06-29:
Running ESP-IDF 4.x + Arduino-ESP32 2.x and trying the hello-world example, the output is wrong. It seems the interpreter loads the model but then `interpreter->input(0);` and `interpreter->output(0);` returns vectors that have an extremely high zero-point and a scale of either 0 or `-nan`. This ofc makes the quantized x & y values wrong, too. Specifically, the output `y` is always 0.

```
Quantized x: -1 input zero point 1073421472 scale -nan
Quantized y: -48 output zero point 1073421456 scale 0.000000
x_value: 0.000000, y_value: -0.000000

Quantized x: -1 input zero point 1073421472 scale -nan
Quantized y: -48 output zero point 1073421456 scale 0.000000
x_value: 0.314159, y_value: -0.000000
```

This is running on a regular ESP32 dev board with no PSRAM. Example projects see https://github.com/maxgerhardt/pio-esp32-tfmicro-example, it uses PlatformIO for compilation. Any help is appreciated.

maxgerhardt, 2024-06-29:
Note: Manually fixing input and output quantization with values read from the model and https://netron.app/ still leads to wrong model output.
```cpp
  int8_t x_quantized = x * 40.69345526012819f + 128;
//..
  float y = (y_quantized - 4) * 0.008472034707665443f;
```
There must be something wrong when TFlite or Flatbuf decodes the model...

vikramdattu, 2024-07-01:
Hi @maxgerhardt did you try the example from `esp-tflite-micro` without Arduino? https://github.com/espressif/esp-tflite-micro/tree/master/examples/hello_world 
Do you find results with this problematic too?

Inmoresentum, 2024-08-13:
I am having the same issue when testing out the provided hello word example. I am also using PlatformIO with espidf framework (*no arduino*). I am wondering if this is an issue with the tflite micro interpreter or the model itself?

vikramdattu, 2026-11-09:
Closing, as there has been no activity here for a long time. esp-tflite-micro has changed a lot since (examples, ESP-IDF support and the upstream TFLite Micro sync). If you still see this with the latest release, please open a new issue with the ESP-IDF version, the board and the full log.

closed 2026-11-09 by vikramdattu