espressif/esp-tflite-micro #89: In person_detection example on ESP32-S3 Korvo 2, Camera captured Failed.

hardikofficial963, 2024-07-26:
I am trying to build person_detection example on ESP32 Korvo 2, I configured the Pins as defined in datasheet:

[image]

Then I build the project, Build successful, but giving me  "Camera captured failed" error.

[image]

I used "idf.py set-target esp32s3"

Please tell me how can I resolve it.

Thanks

vikramdattu, 2024-07-26:
Can you please check in the initial logs if the camera init is indeed successful? If not, the image capture will keep failing.

hardikofficial963, 2024-07-26:
It is failing here

person_detection\main\image_provider.cc

```
// Get an image from the camera module
TfLiteStatus GetImage(int image_width, int image_height, int channels, int8_t* image_data) {
#if ESP_CAMERA_SUPPORTED
  ESP_LOGE(TAG, "Trying to init camera");
  camera_fb_t* fb = esp_camera_fb_get();
  if (!fb) {
    ESP_LOGE(TAG, "Camera capture failed");
    return kTfLiteError;
  }
```

vikramdattu, 2026-11-09:
Closing, as there has been no activity here for a long time. esp-tflite-micro has changed a lot since (examples, ESP-IDF support and the upstream TFLite Micro sync). If you still see this with the latest release, please open a new issue with the ESP-IDF version, the board and the full log.

closed 2026-11-09 by vikramdattu