espressif/esp-tflite-micro #91: esp_nn.h File During Compilation with -D ESP_NN

pravee625, 2024-08-01:
When I include the -D ESP_NN flag in my platformio.ini, I encounter the following error during compilation:

Compiling .pio\build\esp-wrover-kit\libde0\tfmicro\tensorflow\lite\micro\kernels\esp_nn\add.cc.o
Compiling .pio\build\esp-wrover-kit\libde0\tfmicro\tensorflow\lite\micro\kernels\esp_nn\conv.cc.o
lib/tfmicro/tensorflow/lite/micro/kernels/esp_nn/add.cc:34:10: fatal error: esp_nn.h: No such file or directory

****************************************************************
* Looking for esp_nn.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_nn.h"
* Web  > https://registry.platformio.org/search?q=header:esp_nn.h
*
****************************************************************

 #include <esp_nn.h>
          ^~~~~~~~~~
compilation terminated.
Compiling .pio\build\esp-wrover-kit\libde0\tfmicro\tensorflow\lite\micro\kernels\esp_nn\depthwise_conv.cc.o
*** [.pio\build\esp-wrover-kit\libde0\tfmicro\tensorflow\lite\micro\kernels\esp_nn\add.cc.o] Error 1
lib/tfmicro/tensorflow/lite/micro/kernels/esp_nn/conv.cc:30:10: fatal error: esp_nn.h: No such file or directory

****************************************************************
* Looking for esp_nn.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_nn.h"
* Web  > https://registry.platformio.org/search?q=header:esp_nn.h
*
****************************************************************

 #include <esp_nn.h>
          ^~~~~~~~~~
compilation terminated.
*** [.pio\build\esp-wrover-kit\libde0\tfmicro\tensorflow\lite\micro\kernels\esp_nn\conv.cc.o] Error 1
lib/tfmicro/tensorflow/lite/micro/kernels/esp_nn/depthwise_conv.cc:30:10: fatal error: esp_nn.h: No such file or directory

****************************************************************
* Looking for esp_nn.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_nn.h"
* Web  > https://registry.platformio.org/search?q=header:esp_nn.h
*
****************************************************************

 #include <esp_nn.h>
          ^~~~~~~~~~
compilation terminated.
*** [.pio\build\esp-wrover-kit\libde0\tfmicro\tensorflow\lite\micro\kernels\esp_nn\depthwise_conv.cc.o] Error 1
========================================================= [FAILED] Took 31.34 seconds =========================================================

 *  The terminal process "C:\Users\prave\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

When I remove the -D ESP_NN flag from my platformio.ini, the compilation is successful.

Is the esp_nn.h file not present in the esp-tflite-micro library?
How can I properly include the ESP-NN library while compiling in PlatformIO?

I am attaching my platformio.ini file for  your reference
[image]

vikramdattu, 2024-08-07:
@pravee625 is this issue still valid after https://github.com/espressif/esp-tflite-micro/issues/93 is addressed?

pravee625, 2024-08-07:
@vikramdattu Yes, I am still unable to compile with -D ESP_NN if declared in my platformio.ini file 

if ESP_NN is declared in platformio.ini 
then esp_nn.h file is required 
https://github.com/espressif/esp-tflite-micro/blob/e63e91b37c52e8085a6af83048aafc8b0ddaf1a2/tensorflow/lite/micro/kernels/esp_nn/add.cc#L34
and I am unable to find that file in the current repo...

vikramdattu, 2024-08-07:
Hi @pravee625 `esp-nn` is separate component and with `esp-idf`'s build system, it gets pulled in transparently as a dependency to `esp-tflite-micro`. This happens via dependency specified in `idf_component.yml`: https://github.com/espressif/esp-tflite-micro/blob/e63e91b37c52e8085a6af83048aafc8b0ddaf1a2/idf_component.yml#L10-L12

<br>
For platform.io, can you please follow [this](https://docs.platformio.org/en/latest/frameworks/espidf.html#esp-idf-components) piece of documentation? 
`esp-nn` can be cloned from [here](https://github.com/espressif/esp-nn) and you can then add it manually as per above documentation.

pravee625, 2024-08-08:
Thanks for the help @vikramdattu. My code is working fine, and esp_nn is successfully integrated. You can close this issue as well as the other issue: https://github.com/espressif/esp-tflite-micro/issues/90. The previous issue was also caused by esp_nn optimizations not being included. Now, my inference time has decreased from 900ms to 200ms.

vikramdattu, 2024-08-08:
Hi @pravee625 thanks for the confirmation. Closing this issue!

closed 2024-08-08 by vikramdattu