espressif/esp-tflite-micro #92: Use as Wakeword engine

garudaonekh, 2024-08-03:
Hi,
Is it feasible to use this as wakeword engine for esp32c3 instead of Wakenet of esp32-SR. 
I am concerned about resources consumption and size and other compatible issue.
AFAIK, espressif Wakenet doesn't support self training of wakeword as of now.

Thanks;

vikramdattu, 2024-08-07:
Hi @garudaonekh what's your expected model size (in parameters) and the memory you can spare. Though wake-word models are lighter on resources, it is worthwhile knowing your requirements, given, `esp32c3` does not come with AI instructions and no SPI-RAM.

garudaonekh, 2024-08-08:
> Hi @garudaonekh what's your expected model size (in parameters) and the memory you can spare. Though wake-word models are lighter on resources, it is worthwhile knowing your requirements, given, `esp32c3` does not come with AI instructions and no SPI-RAM.

Sorry, I mean ESP32S3. My end goal is to train my own wakeword in my language. and to get this git to work as wakeword engine

thirstyone, 2024-10-01:
same question, but this time esp32c3 with INMP441...

vikramdattu, 2026-11-09:
Yes, the `micro_speech` example is a small keyword-spotting model, and you can train your own words with the upstream TensorFlow Lite Micro training scripts, then replace the model. For a production wake word, ESP-SR's WakeNet is built and tuned for that job.

Closing, as this is a question rather than an issue.

closed 2026-11-09 by vikramdattu