espressif/esp-tflite-micro #93: Cannot build the micro_speech example

pravee625, 2024-08-07:
I am currently using vs-code with esp_idf extension installed"version 5.1.4", I am trying to build the micro_speech example and I am getting the following errors 

after cloning the master repo.
I went to the below dir and executed idf.py set-target esp32
PS C:\Users\prave\Desktop\esp-tflite-micro-master\esp-tflite-micro-master\examples\micro_speech>
I am getting the below error.

PS C:\Users\prave\Desktop\esp-tflite-micro-master\esp-tflite-micro-master\examples\micro_speech> idf.py set-target esp32
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory 'C:\Users\prave\Desktop\esp-tflite-micro-master\esp-tflite-micro-master\examples\micro_speech\build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32, new sdkconfig will be created.
Running cmake in directory C:\Users\prave\Desktop\esp-tflite-micro-master\esp-tflite-micro-master\examples\micro_speech\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\Users\prave\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=1 C:\Users\prave\Desktop\esp-tflite-micro-master\esp-tflite-micro-master\examples\micro_speech"...
-- Found Git: C:/Users/prave/.espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1")   
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/prave/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/prave/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/prave/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
Processing 3 dependencies:
[1/3] espressif/esp-nn (1.0.2)
[2/3] espressif/esp-tflite-micro (*)
[3/3] idf (5.1.4)
CMake Error at C:/Users/prave/esp/v5.1.4/esp-idf/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'espressif__esp-tflite-micro'.
Call Stack (most recent call first):
  C:/Users/prave/esp/v5.1.4/esp-idf/tools/cmake/build.cmake:308 (__build_resolve_and_add_req)
  C:/Users/prave/esp/v5.1.4/esp-idf/tools/cmake/build.cmake:595 (__build_expand_requirements)
  C:/Users/prave/esp/v5.1.4/esp-idf/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:4 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/prave/Desktop/esp-tflite-micro-master/esp-tflite-micro-master/examples/micro_speech/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the C:\Users\prave\Desktop\esp-tflite-micro-master\esp-tflite-micro-master\examples\micro_speech\build\log\idf_py_stderr_output_2452 and C:\Users\prave\Desktop\esp-tflite-micro-master\esp-tflite-micro-master\examples\micro_speech\build\log\idf_py_stdout_output_2452   

I have attached the dependencies.lock file in micro_speech.
[image]

I have attached my dir structure for you to look over. 
[image]

What could be the problem here?  I have not changed anything from the original repo...

vikramdattu, 2024-08-07:
Hi @pravee625 does the resolution works fine when you rename the folder `esp-tflite-micro-master` to just `esp-tflite-micro`?

pravee625, 2024-08-07:
Hi @vikramdattu ,Yes, renaming the folder from esp-tflite-micro-master to esp-tflite-micro resolved the issue. 

Thanks for the tip!

vikramdattu, 2024-08-07:
@pravee625 that's great to here. I will close the issue, since this happens because of the component name being used with `-master` suffix because of wrong directory name.
Feel free to reopen or open a new issue if the problem re-appears.

closed 2024-08-07 by vikramdattu