espressif/esp-tflite-micro #94: Using a large model on a ESP32 S3 N16R8

nicklasb, 2024-08-25:
Hi, 

I am inferring a large model (~2 MB) on the ESP32 S3, and it takes about 60 seconds, while taking about 50 ms on my PC. 
As the tensor arena seem to have to be about 5 MB to satisfy TF lite, and the RGB image is larger than SRAM (it's a YOLO model, it prefers RGB), obviously everything ends up happening in PSRAM, which slows things down significantly. 
However I don't think it does so by a factor of thousand even though the ESP32 S3 is obviously slower as well.

What can I do? Can I override the memory allocator to put just some stuff in SRAM, I have about 300K available that isn't being used?
I saw that the S4 will basicall have SRAM as a cache for a faster PSRAM more or less, is there something similar that could be done in the meanwhile? Or something else?

vikramdattu, 2024-08-26:
Hi @nicklasb ESP32-S3 have cache option and you can explore the same. If you have that much internal RAM underutilised, do increase the Data cache and I cache sizes from `menuconfig` to their max. It should give you a good boost in performance.

As far as moving some of the allocations to internal goes, the current tflite structure is not flexible enough to allow that. You may move some of the critical kernels (from esp-nn) to IRAM to make them always persist in RAM to boost it even further. Please explore `esp_att.h` for the same. Simply add `IRAM_ATTR` in front of the function and it will be placed in the IRAM.

5MB tensor Arena requirement is indeed high and I am not sure if this should be the case. I will give the Yolo model a try and do some experiments myself.

nicklasb, 2024-08-26:
Hi, thanks for you answer!

> Hi @nicklasb ESP32-S3 have cache option and you can explore the same. If you have that much internal RAM underutilised, do increase the Data cache and I cache sizes from `menuconfig` to their max. It should give you a good boost in performance.

I am afraid that they have not done much difference in my case, also, their maximum values aren't very high, I will revisit them and see if have missed something. 

> As far as moving some of the allocations to internal goes, the current tflite structure is not flexible enough to allow that. 

It sort of is IMO, but only up to a point, the MicroAllocator can be initialized with a non-persistent area that I think could help to some degree. However, I have not been able to make that work, then again C++ semantics is not my home turf (yet) and they seem to use all the tricks. There are some PR:s that touch this over at TFLite, if Espressif put some of their might behind that I think it would make a difference.

> You may move some of the critical kernels (from esp-nn) to IRAM to make them always persist in RAM to boost it even further. Please explore `esp_att.h` for the same. Simply add `IRAM_ATTR` in front of the function and it will be placed in the IRAM.

Ok, I will take a look.

> 5MB tensor Arena requirement is indeed high and I am not sure if this should be the case. I will give the Yolo model a try and do some experiments myself.

It is, but perhaps it is not that strange, the model is > 2 MB. Either way, the size doesn't matter as long it fits well within PSRAM. It is more about that the frequent stuff needs a faster memory.

nicklasb, 2024-08-26:
@vikramdattu 

> Please explore `esp_att.h` for the same. Simply add `IRAM_ATTR` in front of the function and it will be placed in the IRAM.
I am afraid that made no discernable difference. 

What I am going to do now is clone this library instead of using it as a stand-alone compinent, and focus on if I can override the memory management in some way by using MicroAllocator. Maybe I can help out in some way.

nicklasb, 2024-08-26:
@vikramdattu 
On a side note, I ran the model without the nn function optimization, and that made the inference 3.5 times slower. 
So good work there. :-)

vikramdattu, 2024-08-27:
Did you set `-O2` from menuconfig options or from `esp-nn/CMakeLists.txt`? Can you please share the observations about strange results? Do you get the bit mismatch?

nicklasb, 2024-08-27:
Hi, I set it from menuconfig and basically the box coordinates ended upp smaller, I had some of them going negative. 
I don't have a working codebase currently (writing a custom MicroAllocator), I can get you more specifics tonight (CET).

mirecta, 2025-02-08:
> Hi,
> 
> I am inferring a large model (~2 MB) on the ESP32 S3, and it takes about 60 seconds, while taking about 50 ms on my PC. As the tensor arena seem to have to be about 5 MB to satisfy TF lite, and the RGB image is larger than SRAM (it's a YOLO model, it prefers RGB), obviously everything ends up happening in PSRAM, which slows things down significantly. However I don't think it does so by a factor of thousand even though the ESP32 S3 is obviously slower as well.
> 
> What can I do? Can I override the memory allocator to put just some stuff in SRAM, I have about 300K available that isn't being used? I saw that the p4 will basically have SRAM as a cache for a faster PSRAM more or less, is there something similar that could be done in the meanwhile? Or something else?

Can u publish some tutorial. Step by step how to run large model on esp32s3 n16r8 even uf inference cost is 60s  please ?

nicklasb, 2025-02-09:
> Can u publish some tutorial. Step by step how to run large model on esp32s3 n16r8 even uf inference cost is 60s please ?

Not really, it was quite cumbersome and had a *lot* of steps, so my project is a complete mess and doesn't even build anymore, I never intended it for more than performance testing ([note that I also tried with the P4](https://github.com/espressif/esp-nn/issues/11#issuecomment-2371061908)). 

But I suppose that if you don't venture into my attempts of memory optimization and is ok with everything living in PSRAM, my steps were these:
1. Train a nano model in YOLO v5
2. Export to tflite and quantize it to int8. I think I had issues with later versions failing to provide a [full int8 quantization](https://github.com/ultralytics/ultralytics/issues/16090#event-15987226242)`, however, looking at that thread now, someone else seems to at least make that work with later YOLO versions (11). Not sure if tf-lite-micro supports all those operations though.

3. As a project, I stole as much as I could from the TF-Lite-Micro exampes (mostly from [person_detection](https://github.com/espressif/esp-tflite-micro/tree/master/examples/person_detection), as Copilot to frankenstein something together from that and then try and fill out the gaps.
4. I used Platform IO and its board_build.embed_files - functionality to embed the model into PSRAM. I also used a spiffs-file with a 15MB factory partition. 

5. One thing that took a long time was to find out what operations it used, these where the ones I got to work:
    resolver.AddConv2D();
    resolver.AddDepthwiseConv2D();
    resolver.AddAdd();
    resolver.AddConcatenation();
    resolver.AddResizeNearestNeighbor();
    resolver.AddRelu();
    resolver.AddPad();
    resolver.AddLogistic();
    resolver.AddMul();
    resolver.AddMaxPool2D();
    resolver.AddReshape();
    resolver.AddStridedSlice();
    resolver.AddQuantize();

Note that this is not a very simple thing to get going, I do not recommend it basically, but if the above helps someone..

vikramdattu, 2026-11-09:
Closing this discussion. For large models, put the tensor arena in PSRAM and keep the weights in flash. Newer ESP-NN releases are much faster on ESP32-S3 and ESP32-P4, and can split the larger layers across both cores. Please open a new issue for a specific problem on the latest release.

closed 2026-11-09 by vikramdattu