espressif/esp-tflite-micro #98: mobilenetv3_small_float32.tflite量化成int8后精度下降严重

xiaojianjun22, 2024-11-07:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

1.mobilenetv3_small_float32.tflite量化成int8后精度下降严重，基本无法用来识别
2.mobilenetv3_small_float32.tflite不量化模型大的导致esp32s3无法加载
3.mobilenetv3_small_float32.tflite量化float16后s3加载模型报错，大小凑合
[image]
4.mobilenetv3_small_float32.tflite采用混合动态量化后，大小合适，加载模型报错
/IDF/examples/camera_my_debug/esp-tflite-micro/tensorflow/lite/micro/kernels/esp_nn/conv.cc Hybrid models are not supported on TFLite Micro.

5.麻烦看看有什么方案可以在esp32s3上正常运行mobilenetv3

vikramdattu, 2024-11-11:
Hello @xiaojianjun22 I see that, for MobileNetV3, the model quantisation to int8 is problematic. I have not found a solution for this yet.
You are right about float16 model being good at accuracy. I will not, however, advocate in favour of this and as you found out tflite-micro does not support hybrid models. Can you raise this question of tflite-micro forum, so that it could be helped better?

vikramdattu, 2024-12-03:
Hello again @xiaojianjun22 even after spending considerable amount of time on getting mobilenetv3_small quantization to work, I couldn't get it working. (Couldn't get it work with QAT and bad accuracy with PTQ int8).
I think, PyTorch tools here can be used somehow to get it work: https://pytorch.org/tutorials/advanced/static_quantization_tutorial.html

I understand that this is not exactly what you are looking for but what I was able to get lot of help and good accuracy for quantised model was with MobileNetV2. I would recommend you to use the V2 if you're not fixated on V3.
I am attaching the python file here which I developed over time which you can use to do conversions and testing the models as well. (Please change quantize_tflite_model.txt to .py extension)
[quantize_tflite_model.txt](https://github.com/user-attachments/files/17994581/quantize_tflite_model.txt)
[requirements.txt](https://github.com/user-attachments/files/17994582/requirements.txt)

Please share if you get the precise solution for MobileNetV3.

closed 2026-11-09 by vikramdattu