espressif/esp-tflite-micro #103: hello_world don't work

well1ngt0nso, 2024-12-02:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

I downloaded the directory: esp-tflite-micro-master, navigated to the hello_world example via ESP-IDF 5.3 CMD, do idf.py build and I get this:

[image]
my structure is:

[image]

vikramdattu, 2025-01-31:
@well1ngt0nso can you change the directory name to `esp-tflite-micro` (from `esp-tflite-micro-master`) and give it a clean try? Please do a clean build., i.e., run `rm -rf build dependencies.lock` files before retrying.

vikramdattu, 2025-08-13:
Closing this as no reply and old issue. Please feel free to re-open if you see the problem with above suggestion.

closed 2025-08-13 by vikramdattu