espressif/esp-tflite-micro #108: ESP32-S3 with TFLite Int8 Quantized Model Returns Only Zeros

manhdatbn93, 2025-01-31:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hi Team,

When deploying a TensorFlow Lite (TFLite) model on an ESP32-S3, the model works correctly in its non-quantized (float32) version, producing consistent results on both the training PC and the ESP32-S3. However, after quantizing the model to int8 (for better size and inference efficiency), the ESP32-S3 always returns 0 as the output, despite the fact that:

1. The model still performs correctly on the PC after quantization.
2. The scale and zero-point values read from the model on ESP32-S3 match those from the PC version.
3. The input pattern array used for inference is identical on both platforms.

Here is the code in setup() on my esp32s3
```
	// Obtain pointers to the model's input and output tensors.
	input = interpreter->input(0);
	output = interpreter->output(0);
	input_scale = input->params.scale;
	input_zero_point = input->params.zero_point;
	output_scale = output->params.scale;
	output_zero_point = output->params.zero_point;
	printf("C Input Scale: %f\n", input->params.scale);
	printf("C Input Zero Point: %ld\n", input->params.zero_point);
	printf("C Output Scale: %f\n", output->params.scale);
	printf("C Output Zero Point: %ld\n", output->params.zero_point);
	printf("Tensor arena usage: %zu/%d bytes\n", 
       interpreter->arena_used_bytes(), 
       kTensorArenaSize);
```

And here is the log that I have:

> C Input Scale: 1.007899
> C Input Zero Point: -16
> C Output Scale: 0.003906
> C Output Zero Point: -128
> Tensor arena usage: 66892/122880 bytes

These values are identical to those on my PC training. However, I consistently receive an output of 0.
```
// Run inference
		if (interpreter_ptr->Invoke() != kTfLiteOk)
		{
			printf("Inference failed for trial %d\n", tr);
			continue;
		}

		// Verify output dimensions
		if (output->dims->data[1] != 2)
		{
			printf("Unexpected output shape! Expected 2 classes\n");
			return;
		}

		// Get confidence values
		int nonblink_conf = output->data.int8[0];
		int blink_conf = output->data.int8[1];

		printf("Trial %d - Non-blink score: %d, Blink score: %d\n",
			   tr,
			   nonblink_conf,
			   blink_conf);
```

Here is the log, I am using 6 patterns for testing

> Input data for trial 0: -76 -76 -76 -75 -75 -74 -74 -74 -74 -75
> Trial 0 - Non-blink score: 0, Blink score: 0
> Trial 1 - Non-blink score: 0, Blink score: 0
> Trial 2 - Non-blink score: 0, Blink score: 0
> Trial 3 - Non-blink score: 0, Blink score: 0
> Trial 4 - Non-blink score: 0, Blink score: 0
> Trial 5 - Non-blink score: 0, Blink score: 0

Could you help me figure out why the quantization model always return 0?

vikramdattu, 2025-01-31:
@manhdatbn93 please try following experiments:
- Try running the model with the esp-nn optimisations turned off
- If that also gives the outputs `0`, possible for you to share the example which we can reproduce the issue with?

manhdatbn93, 2025-01-31:
> [@manhdatbn93](https://github.com/manhdatbn93) please try following experiments:
> 
> * Try running the model with the esp-nn optimisations turned off
> * If that also gives the outputs `0`, possible for you to share the example which we can reproduce the issue with?

Hi @vikramdattu, regarding esp-nn, do you mean selecting ANSI C option in ESP-NN from menuconfig? I tried but the result is still 0

[image]

vikramdattu, 2025-01-31:
@manhdatbn93 that's right. Thanks. Can you also try with this line commented out?

https://github.com/espressif/esp-tflite-micro/blob/78e5532e682d3863a3c2985c3a74eed0a9ebaa61/CMakeLists.txt#L130
Do share a minimal example so I could help with the issue.

manhdatbn93, 2025-01-31:
> [@manhdatbn93](https://github.com/manhdatbn93) that's right. Thanks. Can you also try with this line commented out?
> 
> [esp-tflite-micro/CMakeLists.txt](https://github.com/espressif/esp-tflite-micro/blob/78e5532e682d3863a3c2985c3a74eed0a9ebaa61/CMakeLists.txt#L130)
> 
> Line 130 in [78e5532](/espressif/esp-tflite-micro/commit/78e5532e682d3863a3c2985c3a74eed0a9ebaa61)
> 
>  target_compile_options(${COMPONENT_LIB} PRIVATE -DESP_NN) 
> 
> Do share a minimal example so I could help with the issue.

Yes, here is my example. I am following the micro_speech example code but I have changed the model and using my pattern. you will see the eyeblink_quantized.tflite model in this project. Sorry if my code looks messy at that point

[micro_speech.zip](https://github.com/user-attachments/files/18613708/micro_speech.zip)

manhdatbn93, 2025-02-01:
Hi @vikramdattu , I tried to figure out what was going on in the quantized model and print out the details of that quantized model by following these code below
```
for l in interpreter.get_tensor_details():
    print(f'Name: {l["name"]}')
    print(f' Index: {l["index"]}')
    print(f' Shape: {l["shape"]}')
    print(f' DType: {l["dtype"]}')
```
Here is the log for my quantized model:

> [{'name': 'serving_default_keras_tensor:0', 'index': 0, 'shape': array([  1,   4, 500,   1]), 'shape_signature': array([ -1,   4, 500,   1]), 'dtype': <class 'numpy.int8'>, 'quantization': (1.0078994035720825, -16), 'quantization_parameters': {'scales': array([1.0078994], dtype=float32), 'zero_points': array([-16]), 'quantized_dimension': 0}, 'sparsity_parameters': {}}]
> [{'name': 'StatefulPartitionedCall_1:0', 'index': 26, 'shape': array([1, 2]), 'shape_signature': array([-1,  2]), 'dtype': <class 'numpy.int8'>, 'quantization': (0.00390625, -128), 'quantization_parameters': {'scales': array([0.00390625], dtype=float32), 'zero_points': array([-128]), 'quantized_dimension': 0}, 'sparsity_parameters': {}}]
> Name: serving_default_keras_tensor:0
>  Index: 0
>  Shape: [  1   4 500   1]
>  DType: <class 'numpy.int8'>
> Name: arith.constant
>  Index: 1
>  Shape: [2]
>  DType: <class 'numpy.int32'>
> Name: tfl.pseudo_qconst
>  Index: 2
>  Shape: [2]
>  DType: <class 'numpy.int32'>
> Name: tfl.pseudo_qconst1
>  Index: 3
>  Shape: [  2 240]
>  DType: <class 'numpy.int8'>
> Name: tfl.pseudo_qconst2
>  Index: 4
>  Shape: [16]
>  DType: <class 'numpy.int32'>
> Name: tfl.pseudo_qconst3
>  Index: 5
>  Shape: [16  1  1 16]
>  DType: <class 'numpy.int8'>
> Name: tfl.pseudo_qconst4
>  Index: 6
>  Shape: [16]
>  DType: <class 'numpy.int32'>
> Name: tfl.pseudo_qconst5
>  Index: 7
>  Shape: [ 1  1 16 16]
>  DType: <class 'numpy.int8'>
> Name: tfl.pseudo_qconst6
>  Index: 8
>  Shape: [16]
>  DType: <class 'numpy.int32'>
> Name: tfl.pseudo_qconst7
>  Index: 9
>  Shape: [ 1  4  1 16]
>  DType: <class 'numpy.int8'>
> Name: tfl.pseudo_qconst8
>  Index: 10
>  Shape: [8]
>  DType: <class 'numpy.int32'>
> Name: tfl.pseudo_qconst9
>  Index: 11
>  Shape: [ 8  1 64  1]
>  DType: <class 'numpy.int8'>
> Name: functional_1/batch_normalization_1/batchnorm/add_1;functional_1/conv2d_1/convolution;functional_1/batch_normalization_1/batchnorm/sub1
>  Index: 12
>  Shape: [  1   4 500   8]
>  DType: <class 'numpy.int8'>
> Name: functional_1/batch_normalization_1_2/batchnorm/add_1;functional_1/batch_normalization_1_2/batchnorm/mul_1;functional_1/depthwise_conv2d_1/depthwise;functional_1/batch_normalization_1_2/batchnorm/mul;functional_1/batch_normalization_1_2/batchnorm/sub
>  Index: 13
>  Shape: [  1   1 500  16]
>  DType: <class 'numpy.int8'>
> Name: tfl.dequantize
>  Index: 14
>  Shape: [  1   1 500  16]
>  DType: <class 'numpy.float32'>
> Name: functional_1/activation_1/Elu
>  Index: 15
>  Shape: [  1   1 500  16]
>  DType: <class 'numpy.float32'>
> Name: tfl.quantize
>  Index: 16
>  Shape: [  1   1 500  16]
>  DType: <class 'numpy.int8'>
> Name: functional_1/average_pooling2d_1/AvgPool
>  Index: 17
>  Shape: [  1   1 125  16]
>  DType: <class 'numpy.int8'>
> Name: functional_1/separable_conv2d_1/separable_conv2d/depthwise
>  Index: 18
>  Shape: [  1   1 125  16]
>  DType: <class 'numpy.int8'>
> Name: functional_1/batch_normalization_2_1/batchnorm/add_1;functional_1/separable_conv2d_1/separable_conv2d;functional_1/batch_normalization_2_1/batchnorm/sub     
>  Index: 19
>  Shape: [  1   1 125  16]
>  DType: <class 'numpy.int8'>
> Name: tfl.dequantize1
>  Index: 20
>  Shape: [  1   1 125  16]
>  DType: <class 'numpy.float32'>
> Name: functional_1/activation_1_2/Elu
>  Index: 21
>  Shape: [  1   1 125  16]
>  DType: <class 'numpy.float32'>
> Name: tfl.quantize1
>  Index: 22
>  Shape: [  1   1 125  16]
>  DType: <class 'numpy.int8'>
> Name: functional_1/average_pooling2d_1_2/AvgPool
>  Index: 23
>  Shape: [ 1  1 15 16]
>  DType: <class 'numpy.int8'>
> Name: functional_1/flatten_1/Reshape
>  Index: 24
>  Shape: [  1 240]
>  DType: <class 'numpy.int8'>
> Name: functional_1/dense_1/MatMul;functional_1/dense_1/BiasAdd
>  Index: 25
>  Shape: [1 2]
>  DType: <class 'numpy.int8'>
> Name: StatefulPartitionedCall_1:0
>  Index: 26
>  Shape: [1 2]
>  DType: <class 'numpy.int8'>

As you can see, some layers are still in float32 (elu-14, elu-20) or some layers are still in int32? I don't know why. Here is the code to quantize my model from float to int8
```
quantized_converter.optimizations = [tf.lite.Optimize.DEFAULT]
quantized_converter.representative_dataset = representative_dataset
quantized_converter.target_spec.supported_ops = [tf.lite.OpsSet.TFLITE_BUILTINS_INT8]
quantized_converter.inference_input_type = tf.int8  # Input type for deployment
quantized_converter.inference_output_type = tf.int8  # Output type for deployment

# --------------------------------------------------
# 6. Model Conversion & Saving
# --------------------------------------------------
try:
    quantized_model = quantized_converter.convert()
    
    # Save quantized model
    with open("output/source/eyeblink_quantized.tflite", "wb") as f:
        f.write(quantized_model)
    print("Quantized model saved successfully!")
    
except Exception as e:
    print(f"Quantization failed: {e}")
```

manhdatbn93, 2025-02-02:
Hi @vikramdattu , It seems that ELU activation is not supported in TFLite because the datatypes are still in float32. I changed ELU to ReLU6 and quantized again, but the result remains the same: it always returns 0.
You can find the trained model in zip file with float version, int8 with elu and int8 with relu6

[trained_model.zip](https://github.com/user-attachments/files/18634430/trained_model.zip)

vikramdattu, 2025-02-04:
@manhdatbn93 thanks for the details and sharing the example to try out. Let me try this out and see what's going on! I think there is some other limitations with `tflite-micro` kernels itself which needs to be addressed.
Maybe not exactly the same but we have similar issue with one of our MobileNetV2 quantised models: https://github.com/tensorflow/tflite-micro/issues/3046

vikramdattu, 2025-02-04:
> [@manhdatbn93](https://github.com/manhdatbn93) that's right. Thanks. Can you also try with this line commented out? [...]

Now this is weird! Will try this as well.

manhdatbn93, 2025-02-04:
Hi @vikramdattu , I will wait for your response. Let me know if you need further information.

vikramdattu, 2025-02-06:
@manhdatbn93 
Hello again! We have found an issue. The fully_connected interface file needs to be updated with the latest changes from `tflite-micro`.

For now, I am providing you the patch on CMakeLists.txt which removes fully_connected entirely from the compilation.

[skip_esp_nn_fully_connected.patch](https://github.com/user-attachments/files/18683509/skip_esp_nn_fully_connected.patch)

After applying the patch, please remove the `build` folder from your example and try again. You can enabled `esp_nn` from flag and enabled optimizations as well from menuconfig at this point. Do share your findings.

manhdatbn93, 2025-02-06:
Hi @vikramdattu , Thanks for your update!.
I modified the CMakeLists.txt in esp-tflite-micro, but the inference time remains high and unchanged.

[image]

manhdatbn93, 2025-02-07:
Hi @vikramdattu , I apologize for the misunderstanding. I have checked the quantized model, and it works.

vikramdattu, 2025-02-07:
Hi @manhdatbn93 good to hear that it worked for you. 

> Hi @vikramdattu , Interestingly, if I comment out this line, the inference time is reduced in my case.

This issue still exists for you? Or was it some build config miss?

manhdatbn93, 2025-02-08:
Hi @vikramdattu , I'd like to confirm whether disabling the ESP-NN optimization is sufficient by selecting the ANSI option in menuconfig, or if I also need to comment out the line **"target_compile_options(${COMPONENT_LIB} PRIVATE -DESP_NN)"** in the [esp-tflite-micro/CMakeLists.txt](https://github.com/espressif/esp-tflite-micro/blob/78e5532e682d3863a3c2985c3a74eed0a9ebaa61/CMakeLists.txt#L130)

vikramdattu, 2025-02-08:
Hi @manhdatbn93, disabling from menuconfig disables optimizations, however, ansi C versions are plugged in from esp-nn.
Even those can be removed with commenting out that line from CMakeLists.txt.

manhdatbn93, 2025-02-08:
> Hi [@manhdatbn93](https://github.com/manhdatbn93), disabling from menuconfig disables optimizations, however, ansi C versions are plugged in from esp-nn. Even those can be removed with commenting out that line from CMakeLists.txt.

Hi @vikramdattu , I have updated my comment. Could you please take a look my previous comment.

Abhinav-Dhamne, 2025-05-07:
Hi @vikramdattu this thread has been a life-saver, was facing a similar issue .
Irrespective of the input my output was constant (1/(no of output classes)). Applied your patch , and it worked flawlessly.
Any idea when this patch would be pushed up to master ?
Thank you  !

vikramdattu, 2025-05-07:
Hi @Abhinav-Dhamne this is already merged and available with v1.1.1: https://components.espressif.com/components/espressif/esp-nn/versions/1.1.1.

Please do a clean build of your example, and make sure to use the latest version (v1.1.1) of esp-nn if not already.