espressif/esp-tflite-micro #115: Applying model into ESP32S3 Problem

leanhplus2612, 2025-08-09:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hi espressif,
I am researching micro_speech example and trying to replace the google dataset into my own data, in addition, I also want to extend the time for detecting the speech with INMP441 micro.
I had read a [google colab link](https://colab.research.google.com/github/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/examples/micro_speech/train/train_micro_speech_model.ipynb#scrollTo=XohZOTjR8ZyE) that train for yes/no word by accident. But after training and replace the model in micro speech example, It is not work. Moreover, I also follow an wake word model in this link: [(https://github.com/atomic14/voice-controlled-robot/tree/main)]. However, I just work with the lower tensorflow version.

My question is that what is the tensorflow version which is supported by esp32s3 and esp32 in general? Do you have any method for resolve my problem above? 

I am really looking forward to your positive response! 
Thank you very much in advance!

Best Regards,
Ant Leo

maltsev-dev, 2025-09-03:
Hi @leanhplus2612

Due to the large number of tightly coupled fragile components, it is very difficult to determine who is to blame from your story.

As I understand it, you use ready-made scripts from tutorials for training the model on your data. Probably, your data is already consistent with the original dataset in format, balance, markup, etc.

If after training the model gives a good matrix, just like the quantized model gives good results on test data.
It seems to me that the problem of data fluctuations may be in the excessive _sensitivity_ of the model to input data due to noise.

First of all, (if the previous stages are passed successfully) on the MCU I would try to normalize the raw data and set a suitable threshold, making the entry into the class more strict, conservative filtering so as not to pass garbage data to the inference, etc.

Find the culprit and punish him.
Good luck to you!

vikramdattu, 2026-11-09:
Closing, as there has been no activity here for a long time. esp-tflite-micro has changed a lot since (examples, ESP-IDF support and the upstream TFLite Micro sync). If you still see this with the latest release, please open a new issue with the ESP-IDF version, the board and the full log.

closed 2026-11-09 by vikramdattu